Koenkk/zigbee2mqtt #11687: ZSDR-850 Smoke Detector from Klik Aan Klik Uit (Trust)

guidokieboom, 2022-03-03:
### Link

https://www.trust.com/en/product/71197-zigbee-smoke-detector-zsdr-850#

### Database entry

{"id":12,"type":"EndDevice","ieeeAddr":"0x000d6f00162c242d","nwkAddr":41663,"manufId":4619,"manufName":"Trust","powerSource":"Battery","modelId":"SmokeSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,1282],"outClusterList":[25],"clusters":{"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff0810d3","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":16,"dateCode":"2020.10.14","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1646292224377,"defaultSendRequestWhen":"immediate"}

### Comments

Smoke detector paired successfully. However, it is seen as a HEIMAN HS3SA, where I consider this more a HEIMAN HS1SA. I've seen similar issues in https://github.com/Koenkk/zigbee2mqtt/issues/9919. I have the latest Z2M (1.24.0) installed. I'm using a ConbeeII zigbee router (not using ZHA, nor the Conbee/Deconz interface; just directly the Z2M). Multiple screenshots see below. Request is to add this ZSDR-850 device to the devices list.

After successful pairing and device rename:
[image]

State info:
[image]

Debug log around a device self-test:
[image]

Picture of the ZSDR-850 device:
[image]

Katamave, 2022-03-14:
I have the same problem.

leoGr3006, 2022-03-29:
I'm a starter and like to connect this device (ZSDR-850) to my Aeotec Smartthings hub but can't find the handler (GitHub and included drivers)
Thanks for your help.

closed 2022-05-06 by github-actions[bot]