Koenkk/zigbee2mqtt #11690: IKEA LED2003G10 is supported but frontend reports as unsupported device

24jonny, 2022-03-03:
### What happened?

Reset bulb and discovered via zigbee2mqtt through Homeassistant frontend. 
Logs report: 
- Successfully interviewed, device has successfully been paired
- Device with Zigbee model 'TRADFRIbulbE27WSglobeopal1055lm' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

### What did you expect to happen?

According to supported devices list [device](https://www.zigbee2mqtt.io/devices/LED2003G10.html) should be supported.
Model name LED2003G10 is printed on bulb.
So I expect it to just work like it awesomely usually does (Thanks for this awesome service! <3).
Tried resetting buld several times with no change and other devices join flawlessly so I don't think anything is wrong with my setup.

I believe trying to manually support it if it should already be supported won't do any good?

### How to reproduce it (minimal and precise)

- Allow new devices to join via Homeassistant frontend
- Reset LED2003G10 (switching on and off)

### Zigbee2MQTT version

1.18.1

### Adapter firmware version

20190223

### Adapter

Wireless Zigbee CC2531

Koenkk, 2022-03-03:
You are using an old z2m version, updating to the latest will fix it.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2022-03-03 by Koenkk

pblanar, 2022-04-23:
I'm experiencing the same problem running version 1.25.0-1 but the one distinction appears to be that while it has the same model number (LED2003G10) this is a 1100 lm bulb as opposed to the 1055 lm one that appears to be in the compatible devices (can also be seen from the different Zigbee model from the error code below. 

I get the following error: Device '0x003c84fffe37516e' with Zigbee model 'TRADFRIbulbE26WSglobeopal1100lm' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Koenkk, 2022-04-23:
`TRADFRIbulbE26WSglobeopal1100lm` is already supported in the latest dev and next release on 1 May. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)