Koenkk/zigbee2mqtt #11697: [New device support]: TuYa TS TS0505B  _TZ3210_jmiuubkz

nautiqu3, 2022-03-03:
### Link

https://www.lazada.co.th/products/nestmotion-zigbee-rgb-cw-color-bulb-by-tuya-16-google-assistant-amazon-alexa-i2837699627-s10355212036.html?spm=a2o4m.searchlist.list.2.67f8f478oErVzV&search=1

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0x0c4314fffe0db207","nwkAddr":17455,"manufId":4098,"manufName":"_TZ3210_jmiuubkz","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_jmiuubkz","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1646331762711,"defaultSendRequestWhen":"immediate"}

### Comments

I have the dev pull and I can see TS0505B in the tuya.js
But non of the manufacturer _TZ3210_jmiuubkz doesnt seem to be supported. 

Is it possible for someone to help me add that?

nautiqu3, 2022-03-04:
Got it to work myself by adding the manufacturerName to the tuya.js file.

closed 2022-03-04 by nautiqu3

Koenkk, 2022-03-05:
Added it. 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)