Koenkk/zigbee2mqtt #11699: Invalid readings on DATEK (EVA) HAN meter

odechr, 2022-03-04:
### What happened?

[image]

Doesnt seem like the EVA Datek Power metering is measuring the correct values.
https://www.zigbee2mqtt.io/devices/HSE2905E.html

In the updatelog in Zigbee2Mqtt log we get this:

Error Failed to configure 'maleravleser', attempt 1 (Error: Read 0x086bd7fffeb6cc6f/1 haElectricalMeasurement(["acFrequencyDivisor","acFrequencyMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:25:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/datek.js:44:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))

Using the [SONOFF ZigBee 3.0 USB Dongle Plus](https://sonoff.tech/product/diy-smart-switch/sonoff-dongle-plus/

Let me know if we have provided enough information.

### Zigbee2MQTT version

1.24.0-1

### Adapter firmware version

Unknown

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus

odechr, 2022-03-09:
We did get a new unit, but still we can't get all the values.
Missing kwh and V/A on the phases. 
[image]

gregersn, 2022-04-11:
I am also seeing issues with this dongle now. I also have a Sonoff USB-dongle, and the same HAN meter, and it reports to me 2394 Volt and 11500 Ampere usage, as 2780 Watts.

odechr, 2022-04-11:
I have been helping the person who made the HAN meter work with Z2M. He found the problem after i sent logs.
Works for me with the fix (not released).
Should be released soon, if its not already released.