Koenkk/zigbee2mqtt #11700: "forgetting" configuration

MoonshineSG, 2022-03-04:
### What happened?

the following are setup in configuration (via the Add-Ons-> Z2M -> Configuration) but after reboot, in the WebUI Settings are shown as 'true'

```
homeassistant:
  legacy_entity_attributes: false
  legacy_triggers: false
advanced:
  legacy_availability_payload: false

```

### What did you expect to happen?

expected to follow the configuration setup. If setup via WebUI, same thing happens. 

### How to reproduce it (minimal and precise)

Change in Add-Ons
Restart Z2M
Check in Settings

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2

Koenkk, 2022-03-05:
Could you check with the edge version of the addon? This should allow to configure this via the frontend. You can uninstall the current version and install the edge (no data loss)

MoonshineSG, 2022-03-05:
I tried, via the UI on the AddOns, Once the z2m guy, as well as via the FileEditor... 

What do you mean by "edge"?

MoonshineSG, 2022-03-06:
yes, edge seams to be ok. settings are there after a few reboots

Koenkk, 2022-03-06:
great, plan is to merge this to a release in April (https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/pull/270)

closed 2022-03-07 by MoonshineSG