Koenkk/zigbee2mqtt #11701: FUT038Z incorrectly detected as Tuya TS0503B

jkempson, 2022-03-04:
### What happened?

I tried to pair a FUT038Z with z2m (https://www.zigbee2mqtt.io/devices/FUT038Z.html) but it is showing up a Tuya TS0503B (https://www.zigbee2mqtt.io/devices/TS0503B.html#tuya-ts0503b). Although I can control the device I need color_temp which isn't happening because the device being detected doesn't support this.

### What did you expect to happen?

FUT038Z to be paired correctly with z2m

### How to reproduce it (minimal and precise)

Follow pairing process for FUT038Z

### Zigbee2MQTT version

1.24.0-dev commit: 3535b2c

### Adapter firmware version

20210708

### Adapter

zzh

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8185995/log.txt)

Koenkk, 2022-03-05:
Can you provide the data/database.db entry of this device?

jkempson, 2022-03-05:
Sure!
`{"id":56,"type":"Router","ieeeAddr":"0x84fd27fffe39f71e","nwkAddr":17382,"manufId":4098,"manufName":"_TZ3210_778drfdt","powerSource":"Mains (single phase)","modelId":"TS0503B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0503B","manufacturerName":"_TZ3210_778drfdt","stackVersion":0,"dateCode":"","zclVersion":3}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1646399410454,"defaultSendRequestWhen":"immediate"}
`

Koenkk, 2022-03-05:
Thanks, just checked but FUT038Z is an RGBW light, are you sure FUT038Z is listed on the device?

jkempson, 2022-03-05:
Right, I'm an idiot. Really sorry to waste your time here but I've actually got a FUT03**7**Z. When I'd googled FUT037Z the top result was for the FUT038Z and I never noticed the digit difference.

So I'm guessing the FUT037Z isn't supported?

I'm very confused in general right now. I have an led strip with contacts marked R,G,B and +12V but the leds are actually just cool/warm white. So I thought I'd need an RGB controller? I can kinda change the temperature of the leds using the color wheel so the converter js you provided made perfect sense although doesn't seem to work using HA.

jkempson, 2022-03-05:
This is what the leds I have look like
[image]

Koenkk, 2022-03-06:
Does it work with the external converter from https://github.com/Koenkk/zigbee2mqtt/issues/11701#issuecomment-1059779453 ?

jkempson, 2022-03-06:
So yes it does work with the converter but 2 big issues. It's reversed and not calibrated correctly. If I use the color wheel in HA and set to blue I get a warm orange light, if I set to orange I get blue light. The color temp slider kinda works (although also reversed) but the cool end of the scale is setting a value in the middle of the color wheel rather than a blue color if that makes sense? Happy to provide screenshots if it would help? PS I've triple checked wiring on the controller and is correct.

jkempson, 2022-03-06:
Update. So I've swapped wires around on the controller by trial and error and now have correct color control using the RGB wheel (orange/blue). The slider doesn't seem to work at all though weirdly

jkempson, 2022-03-06:
I'm gonna close this. I've realised by controlling individual color channels I can achieve the result I was looking for with a bit of node-red logic.

closed 2022-03-06 by jkempson