Koenkk/zigbee2mqtt #11703: [New device support]: Zigbee model 'TS011F' and manufacturer name '_TZ3000_amdymr7l' (Blitzwolff SHP-13)

Nenodema, 2022-03-04:
### Link

https://www.blitzwolf.com/ZigBee-3.0-Smart-Socket-EU-p-518.html

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0xa4c138bf6ccf49dd","nwkAddr":35956,"manufId":4417,"manufName":"_TZ3000_amdymr7l","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_amdymr7l","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1646401884765,"defaultSendRequestWhen":"immediate"}

### Comments

This is my second SHP-13, the first is working perfectly but the second one is not working.

```
Zigbee2MQTT:warn  2022-03-04 14:51:33: Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_amdymr7l'
Zigbee2MQTT:warn  2022-03-04 14:51:33: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

Could you please also support this device? Thanks!

Koenkk, 2022-03-05:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-05 by Koenkk

Nenodema, 2022-03-05:
Thanks Koen!!

danigd82, 2022-03-07:
I have the same problem, when for the stable version? Thank you very much!!

bboelaert, 2022-03-08:
Same question : I modified the js file in the docker version and it works. When will this change be incorporated in the stable and docker version?

JanHACS, 2022-03-10:
I have the same problem with home assistant and zigbee2mqtt, it would be very nice to have support for this device.

han9988, 2022-03-11:
Just received these today with the same problem. Took the opportunity and tried to add the device as in https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html described.
After all, it shows up in the list. From there one functionality at a time. Steep learning curve. ;-)
When will this be availble in the stable version, asking for a friend.

JanHACS, 2022-03-12:
> Just received these today with the same problem. Took the opportunity and tried to add the device as in https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html described. [...]

I would like to have more programming and English level, but for me that option is out of my control unfortunately, just hoping for a solution from @Koenkk .

pantalooon, 2022-03-14:
Same problem for me. Received 2 of the plugs a few days ago. Unfortunately I do not know how to switch the TrueNAS Scale app to the dev train.

> Same question : I modified the js file in the docker version and it works. When will this change be incorporated in the stable and docker version?

Can you post what's in the js file?

han9988, 2022-03-15:
No need to edit docker container. I did the following:
 In the zigbee2mqtt folder -> configuration.yaml change the line "external_converters": [],
to this:
  "external_converters": [
    "TS011F.js"
  ],

 In the zigbee2mqtt folder add a new file TS011F.js and paste this:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS011F'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'BW-SHP13', // Vendor model number, look on the device for a model number
    vendor: 'Blitzwolf', // Vendor of the device (only used for documentation and startup logging)
    description: 'Blitzwolf Power Plug', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acVoltageMultiplier: 1, acVoltageDivisor: 1, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
            acPowerDivisor: 1,
        });
        try {
            await reporting.currentSummDelivered(endpoint);
        } catch (error) {/* fails for some https://github.com/Koenkk/zigbee2mqtt/issues/11179 */}
    },
    options: [exposes.options.measurement_poll_interval()],
    // This device doesn't support reporting correctly.
    // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage')],
    onEvent: tuya.onEventMeasurementPoll,
};

module.exports = definition;

Restart Zigbee2mqtt
Worked for me, you may do this on your own risk.

edwinzelf, 2022-04-05:
Hi,
I tried the custom external_converter, and it worked like a charm. 
After the April 1st update of z2m (1.25.0-1), it seems the plug is now recognized as an Tuya "TS011F_plug_1", and the custom external_converter is not needed.

I have a couple of actual Tuya TS011F_plug_1 (_TZ3000_typdpbpg). The issue I have with them is power measurement drops to 0 every cpl of minutes. And the even sometimes go to "off" for no reason.

Back to the newer Blitzwolf SHP-13 (_TZ3000_amdymr7l). Since I updated to 1.25.0-1, I am seeing the same power drop to 0 every cpl of minutes. This did not occur with the custom external_converter.

What am I missing?

[image]
[image]
[image]

nohn, 2022-04-05:
> I have a couple of actual Tuya TS011F_plug_1 (_TZ3000_typdpbpg). The issue I have with them is power measurement drops to 0 every cpl of minutes. And the even sometimes go to "off" for no reason. [...]

Looks like the problem described in #11800.

edwinzelf, 2022-04-05:
@nohn looks like it, but it's still strange it worked fine with the external_converter before release 1.25.0-1.

tomasz-s, 2022-04-05:
I didn't try the custom_converter but I used the z2m edge version. Drops were there too.

KaportsevIA, 2022-04-23:
[_TZ3000_amdymr7l.txt](https://github.com/Koenkk/zigbee2mqtt/files/8547351/_TZ3000_amdymr7l.txt)
[image]
[image]
It was possible to obtain energy monitoring data on the outlet. Of the minuses - there is no choice of the mode of operation of the LED indicator

barmazu, 2022-04-25:
Can confirm. Bought one plug (_TZ3000_amdymr7l) 3 days ago to replace TP-Link HS110 (RIP) for washing machine monitoring. It drops to 0 every now and then, doesn't matter if load is constant or not. What's more it has trouble to report low (<5W) loads, simply defaulting to constant 0W. And - last but not least - I think, but not 100% sure, if input voltage exceeds 250V the plug turns off by itself. Garbage.

niolang, 2022-06-02:
Facing the same issue here, and a bit reluctant to switch to edge version just for this one plug ^^
I am new to this, so any classic delay on when this should go to stable?
Thanks

nohn, 2022-06-02:
AFAIK the changes are already included in "latest" (docker) / "master" (source). What you see broken now is really the plug (or it's firmware) that's broken. Use them for monitoring relatively constant loads, for everything else they're crap.

niolang, 2022-06-02:
Oh...
my version was dating back to 2021, because i had an old (wrong?) repo address...
thanks for the heads up!

stathismes, 2022-10-13:
Using KaportsevIA's code [above](https://github.com/Koenkk/zigbee2mqtt/issues/11703#issuecomment-1107477127), and some code from [Koenkk's tyja.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js) I created a file "BW-SHP13.js" and put in inside "/config/zigbee2mqtt/", which helped me get the appropriate readings for BW-SHP13 ++ plus "Indicator Mode" (which I confirm works!)

I hope this helps you.
Many thanks to @KaportsevIA 

[BW-SHP13.js](https://github.com/stathismes/zigbee2mqtt_custom_device_converters/blob/c00227290c8745be81e64182ad1cd640327de770/BW-SHP13.js)

[image]

[image]

Upload the file in HA zigbee2mqtt folder:
[image]

Main settings (here put the file name):
[image]

manny199, 2022-10-25:
> Using KaportsevIA's code [above](https://github.com/Koenkk/zigbee2mqtt/issues/11703#issuecomment-1107477127), and some code from [Koenkk's [...]

Thank you. I inserted new BW-SHP13.js in zigbee2mqtt folder and add it in external converters but power is always 0 for values under 10 watt.
I Have zigbee2mqtt 1.28.0 version

Termi01, 2022-10-30:
@stathismes 
I tried your solution, but I get errors upon loading. This solution only works in the edge version maybe? I am using the regular Z2M addon.

manny199, 2022-10-30:
@Termi01 
Confirm, same problem, error on loading z2m.
This solution not work for me that I have 1.28 version

stathismes, 2022-11-18:
Hmmm that's weird.
Maybe some devices are with a different board?

I'm also using Zigbee2MQTT (full version, not edge)
Current version: 1.28.2-1

I had to update to 1.28.2 because one of the previous ones were generating problems to me too.