Koenkk/zigbee2mqtt #11706: Unable to get power readings from SmartThings STS-OUT-US-2

solomongumbie, 2022-03-04:
### What happened?

I've added a SmartThings STS-OUT-US-2 outlet to Zigbee2mqtt and I'm able to toggle the relay in it, but I'm just seeing null values for volts, amps, and watts. I've tried reconfiguring, and removing the device then adding back again but I'm not having any luck with that. I did get an error message saying the following:
`Error Failed to configure 'Living room TV power', attempt 4 (Error: Read 0x24fd5b00010923bf/1 haElectricalMeasurement(["acFrequencyDivisor","acFrequencyMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:25:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/smartthings.js:160:13) at Configure.configure (/app/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))`

I do hope that makes more sense to you than it does to me.

### What did you expect to happen?

To get values for the energy related sensors on the device

### How to reproduce it (minimal and precise)

The only way I know to do it is to add a SmartThings STS-OUT-US-2 yourself and see if you can get values.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 Dongle-P

### Debug log

Debug Received MQTT message on 'zigbee2mqtt/Shower bulb/set' with data '{"state": "OFF"}'
Debug Publishing 'set' 'state' to 'Shower bulb'
Info MQTT publish: topic 'zigbee2mqtt/Shower bulb', payload '{"energy":0,"linkquality":69,"power":0,"state":"OFF"}'
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Living room TV power","transaction":"qor1c-1"}'
Info Configuring 'Living room TV power'
Debug Received Zigbee message from 'Living room TV power', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acCurrentMultiplier":1,"acVoltageDivisor":10,"acVoltageMultiplier":1}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Living room TV power', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acCurrentDivisor":1000,"acPowerDivisor":10,"acPowerMultiplier":1}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Living room TV power', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Living room TV power', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Living room TV power', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
Error Failed to configure 'Living room TV power', attempt 3 (Error: Read 0x24fd5b00010923bf/1 haElectricalMeasurement(["acFrequencyDivisor","acFrequencyMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:25:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/smartthings.js:160:13) at Configure.configure (/app/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Living room TV power"},"error":"Failed to configure (Read 0x24fd5b00010923bf/1 haElectricalMeasurement([\"acFrequencyDivisor\",\"acFrequencyMultiplier\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))","status":"error","transaction":"qor1c-1"}'

Koenkk, 2022-03-05:
Could you check if this issue has been fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['outletv4'],
    model: 'STS-OUT-US-2',
    vendor: 'SmartThings',
    description: 'Zigbee smart plug with power meter CUSTOM',
    fromZigbee: [fz.on_off, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint);
        // await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.activePower(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.rmsVoltage(endpoint, {change: 10});
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)
- press the yellow reconfigure button
- does it report power measurments now?

solomongumbie, 2022-03-05:
Ha, perfect that fixed it. Much thanks I can now see the power, current and voltage. I've made sure to PayPal a small token of thanks. Love the project.

Koenkk, 2022-03-06:
great and thanks!

pushed the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-06 by Koenkk