Koenkk/zigbee2mqtt #11707: [New device support]:

gargarse, 2022-03-04:
### Link

https://www.beok-controls.com/room-thermostat/underfloor-heating-thermostat/zigbee-underfloor-heating-thermostat.html

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c138c2a86d3a20","nwkAddr":60586,"manufId":4417,"manufName":"_TZE200_2ekuz3dz","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001dM?)i,M?)i<M?)iLM?)i\\M?)ilM?)i|M?)i?M?)i","65506":48,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_2ekuz3dz","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1646426382647,"defaultSendRequestWhen":"immediate"}

### Comments

Beok wall mounted thermostas. Tuya device. similar to Moes BHT-002.

===  tuya.dump.txt

1646426908655 0xa4c138c2a86d3a20 1f00 00 18 02 00 00 00 e5
1646426923026 0xa4c138c2a86d3a20 2000 00 18 02 00 00 00 e4
1646426932920 0xa4c138c2a86d3a20 2100 00 03 04 00
1646426932986 0xa4c138c2a86d3a20 2200 00 01 01 00
1646426935387 0xa4c138c2a86d3a20 2300 00 01 01 01
1646426938952 0xa4c138c2a86d3a20 2400 00 18 02 00 00 00 e5
1646426944923 0xa4c138c2a86d3a20 2500 00 03 04 00
1646426954886 0xa4c138c2a86d3a20 2600 00 18 02 00 00 00 e5
1646426970809 0xa4c138c2a86d3a20 2700 00 18 02 00 00 00 e5
1646426986753 0xa4c138c2a86d3a20 2800 00 18 02 00 00 00 e5
1646427002670 0xa4c138c2a86d3a20 2900 00 18 02 00 00 00 e5
1646427022097 0xa4c138c2a86d3a20 500 00 02 04 00
1646427022166 0xa4c138c2a86d3a20 600 00 03 04 00
1646427022225 0xa4c138c2a86d3a20 700 00 28 01 00
1646427022286 0xa4c138c2a86d3a20 800 00 07 01 01
1646427022345 0xa4c138c2a86d3a20 900 00 2d 05 00
1646427022412 0xa4c138c2a86d3a20 a00 00 1b 02 ff ff ff ec
1646427022473 0xa4c138c2a86d3a20 b00 00 13 02 00 00 00 3c
1646427022535 0xa4c138c2a86d3a20 c00 00 2b 04 00
1646427022595 0xa4c138c2a86d3a20 d00 00 0a 01 00
1646427022669 0xa4c138c2a86d3a20 e00 00 27 01 00
1646427022730 0xa4c138c2a86d3a20 f00 00 68 04 03
1646427022790 0xa4c138c2a86d3a20 1000 00 1f 04 00
1646427022854 0xa4c138c2a86d3a20 1100 00 1e 00 06 00 00 c8 08 00 00 a0 0b 1e 00 a0 0c 1e 00 a0 11 00 00 dc 16 00 00 a0 08 00 00 dc 17 00 00 a0
1646427022913 0xa4c138c2a86d3a20 1200 00 67 01 00
1646427022974 0xa4c138c2a86d3a20 1300 00 65 02 00 00 00 0a
1646427023033 0xa4c138c2a86d3a20 1400 00 66 02 00 00 00 23
1646427023094 0xa4c138c2a86d3a20 1500 00 10 02 00 00 00 dc
1646427023420 0xa4c138c2a86d3a20 1800 00 03 04 00
1646427029501 0xa4c138c2a86d3a20 1900 00 18 02 00 00 00 e5
1646427045426 0xa4c138c2a86d3a20 1a00 00 18 02 00 00 00 e5
1646427061350 0xa4c138c2a86d3a20 1b00 00 18 02 00 00 00 e6
1646427077275 0xa4c138c2a86d3a20 1c00 00 18 02 00 00 00 e6
1646427093212 0xa4c138c2a86d3a20 1d00 00 18 02 00 00 00 e6
1646427109124 0xa4c138c2a86d3a20 1e00 00 18 02 00 00 00 e6
1646427125047 0xa4c138c2a86d3a20 1f00 00 18 02 00 00 00 e6
1646427140970 0xa4c138c2a86d3a20 2000 00 18 02 00 00 00 e6
1646427156897 0xa4c138c2a86d3a20 2100 00 18 02 00 00 00 e6
1646427172821 0xa4c138c2a86d3a20 2200 00 18 02 00 00 00 e6
1646427188759 0xa4c138c2a86d3a20 2300 00 18 02 00 00 00 e6
1646427204667 0xa4c138c2a86d3a20 2400 00 18 02 00 00 00 e6
1646427220588 0xa4c138c2a86d3a20 2500 00 18 02 00 00 00 e6
1646427236512 0xa4c138c2a86d3a20 2600 00 18 02 00 00 00 e6
1646427252435 0xa4c138c2a86d3a20 2700 00 18 02 00 00 00 e6
1646427268358 0xa4c138c2a86d3a20 2800 00 18 02 00 00 00 e6
1646427284281 0xa4c138c2a86d3a20 2900 00 18 02 00 00 00 e6
1646427300213 0xa4c138c2a86d3a20 2a00 00 18 02 00 00 00 e6
1646427316128 0xa4c138c2a86d3a20 2b00 00 18 02 00 00 00 e6
1646427332048 0xa4c138c2a86d3a20 2c00 00 18 02 00 00 00 e6
1646427347972 0xa4c138c2a86d3a20 2d00 00 18 02 00 00 00 e6
1646427363900 0xa4c138c2a86d3a20 2e00 00 18 02 00 00 00 e6
1646427379816 0xa4c138c2a86d3a20 2f00 00 18 02 00 00 00 e6
1646427395753 0xa4c138c2a86d3a20 3000 00 18 02 00 00 00 e6
1646427411661 0xa4c138c2a86d3a20 3100 00 18 02 00 00 00 e6
1646427427584 0xa4c138c2a86d3a20 3200 00 18 02 00 00 00 e5
1646427443507 0xa4c138c2a86d3a20 3300 00 18 02 00 00 00 e5
1646427459428 0xa4c138c2a86d3a20 3400 00 18 02 00 00 00 e5
1646427475350 0xa4c138c2a86d3a20 3500 00 18 02 00 00 00 e5
1646427491273 0xa4c138c2a86d3a20 3600 00 18 02 00 00 00 e5
1646427507207 0xa4c138c2a86d3a20 3700 00 18 02 00 00 00 e5
1646427523117 0xa4c138c2a86d3a20 3800 00 18 02 00 00 00 e5
1646427539040 0xa4c138c2a86d3a20 3900 00 18 02 00 00 00 e5
1646427554975 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427555161 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427555377 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427555802 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427555969 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427556164 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427556366 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427556565 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427556777 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427556967 0xa4c138c2a86d3a20 3a00 00 18 02 00 00 00 e5
1646427570893 0xa4c138c2a86d3a20 3b00 00 18 02 00 00 00 e5
1646427586815 0xa4c138c2a86d3a20 3c00 00 18 02 00 00 00 e5
1646427602747 0xa4c138c2a86d3a20 3e00 00 18 02 00 00 00 e5
1646427618658 0xa4c138c2a86d3a20 4000 00 18 02 00 00 00 e5
1646427634580 0xa4c138c2a86d3a20 4100 00 18 02 00 00 00 e4
1646427650497 0xa4c138c2a86d3a20 4200 00 18 02 00 00 00 e4
1646427666422 0xa4c138c2a86d3a20 4300 00 18 02 00 00 00 e4
1646427682350 0xa4c138c2a86d3a20 4400 00 18 02 00 00 00 e4
1646427698267 0xa4c138c2a86d3a20 4500 00 18 02 00 00 00 e4
1646427714210 0xa4c138c2a86d3a20 4600 00 18 02 00 00 00 e4
1646427730115 0xa4c138c2a86d3a20 4700 00 18 02 00 00 00 e4
1646427746038 0xa4c138c2a86d3a20 4800 00 18 02 00 00 00 e4
1646427761961 0xa4c138c2a86d3a20 4900 00 18 02 00 00 00 e4
1646427777885 0xa4c138c2a86d3a20 4a00 00 18 02 00 00 00 e4

### External converter

```shell
I use this external converter:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_2ekuz3dz'
        },
    ],
    model: 'TGR85-ZB',
    vendor: 'Beok',
    description: 'Beok Thermostat Zigbee',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

PaoloTK, 2022-03-08:
Would like this to be supported as well. On the outside it looks like the Moes BHT-002-GCLZB but the pairing method is different and the device is reported to Z2M as TuYa TS0601.

gargarse, 2022-03-08:
> Would like this to be supported as well. On the outside it looks like the Moes BHT-002-GCLZB but the pairing method is different and the device is reported to Z2M as TuYa TS0601.

Moes BHT-002-GCLZB  also report to Z2M as Tuya device type TS0601.

See in: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/moes.js

`{
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_ztvwu4nk'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_u9bfwha0'}],
        model: 'BHT-002-GCLZB',
        vendor: 'Moes',
        description: 'Moes BHT series Thermostat',
        fromZigbee: [fz.moes_thermostat],
        toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.moes_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit],
        exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 30, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
                .withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program']).withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
        onEvent: tuya.onEventSetLocalTime,
    },`

nicoxygen, 2022-04-08:
i'm so interested for integration

nicoxygen, 2022-05-13:
any idea for integration !

itsolon, 2022-05-26:
Hi i enqueue in this row today i mounted this device in my house it was successfully recognized by zha and could connect without effort but it shows no entities at all.. thats the reason why im hoping someone has any idea for a quirk

itsolon, 2022-06-26:
Im still interested in solution

itsolon, 2022-07-02:
okay boys**girls winter is coming.. 
still no progress.. what do you need to bring this device to work with zigbee2mqtt
its already supported by zha,, this means it should be possible to get this thing work with z2m

ianclegg, 2022-07-25:
looks like there is work in progress on this, its a generic device, the manufacturerName `_TZE200_2ekuz3dz` is identical
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3840

closed 2022-09-03 by github-actions[bot]

sheaur, 2022-11-02:
Wrong time synchronization and no periodical time synchronization on thermostat
hello, my device, the problem has not been solved: _TZE200_2ekuz3dz, z2m 1.28.1 zStack3x0 20220219
ieeeAddr: 0xa4c13851bf0f0033

PaoloTK, 2022-11-14:
Same issue here, the device time doesn't rollback at 24 hours and keeps going past that (currently at 29:31).