Koenkk/zigbee2mqtt #11713: [Feature request]: Add intermediate states for locks

scootec, 2022-03-05:
### Is your feature request related to a problem? Please describe

When setting a lock state via zigbee2mqtt (using the homebridge-z2m plugin specifically), zigbee2mqtt will return the current lock state while waiting for the lock to return it's final state once it has changed. This causes the lock device in homekit to _jitter_, since homekit optimistically sets the final state, then receives a message of the old state (causing homekit to show the old state again), then receives a message of the new state (causing homekit to show the final state). 

For example:
```
debug 2022-03-05 19:11:34: Received MQTT message on 'zigbee2mqtt/0x000d6f000b64f449/set' with data '{"state":"UNLOCK"}'
debug 2022-03-05 19:11:34: Publishing 'set' 'state' to 'Front Door Lock'
debug 2022-03-05 19:11:41: Received Zigbee message from 'Front Door Lock', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":2,"opereventsrc":1,"pin":{"data":[],"type":"Buffer"},"userid":65535,"zigbeelocaltime":324768334}' from endpoint 1 with groupID 0
info  2022-03-05 19:11:41: MQTT publish: topic 'zigbee2mqtt/Front Door Lock', payload '{"action":"unlock","action_source":1,"action_source_name":"rf","action_user":65535,"battery":95,"linkquality":138,"lock_state":"locked","state":"LOCK"}'
info  2022-03-05 19:11:41: MQTT publish: topic 'zigbee2mqtt/Front Door Lock/action', payload 'unlock'
debug 2022-03-05 19:11:47: Received Zigbee message from 'Front Door Lock', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
info  2022-03-05 19:11:47: MQTT publish: topic 'zigbee2mqtt/Front Door Lock', payload '{"battery":95,"linkquality":138,"lock_state":"unlocked","state":"UNLOCK"}'
```
When requesting the unlock, homekit optimistically sets the state to unlocked, then returns to locked (as a result of message 4 above), then finally settles on unlocked when the final state comes through.

### Describe the solution you'd like

Locks in Zigbee2MQTT can currently only have the following states:
`'not_fully_locked', 'locked', 'unlocked'`

If we could add the following states that were used as optimistic states while we wait for the final state:
`'locking','unlocking'`
It would allow for MQTT plugins/apps to recognize that we are waiting for the lock to change state as a result of a command. 

[Home Assistant appears to also support these intermediate states](https://github.com/home-assistant/core/blob/cdb463ea55dadf51f78d2f351cc1311459bb6007/homeassistant/components/lock/reproduce_state.py#L9), so adding this could extend functionality there as well (at a minimum).

### Describe alternatives you've considered

Adding a delay in homebridge-z2m to wait for the final state would solve the _jitter_ of state in homekit. Solves my particular issue, but potentially missing out on additional functionality in Home Assistant and other apps.

### Additional context

Happy to attempt a PR if you think this would add value!

sjorge, 2022-03-05:
You can solve this by setting the device option `filtered_optimistic` and adding `state` and `lock_state` to the list.
I do something similar with a few other attributes, e.g. with color, color_temp, color_mode, and position to avoid those issues with my node-red homekit integration for bulbs and covers. (I have reporting  configured for all my bulbs)

scootec, 2022-03-05:
Thanks @sjorge !

I gave that configuration a shot, but didn't seem to help in this case. See below debug log after those changes were made:
```
debug 2022-03-05 22:18:46: Received MQTT message on 'zigbee2mqtt/0x000d6f000b64f449/set' with data '{"state":"LOCK"}'
debug 2022-03-05 22:18:46: Publishing 'set' 'state' to 'Front Door Lock'
debug 2022-03-05 22:18:51: Received Zigbee message from 'Front Door Lock', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":1,"opereventsrc":1,"pin":{"data":[],"type":"Buffer"},"userid":65535,"zigbeelocaltime":324779564}' from endpoint 1 with groupID 0
info  2022-03-05 22:18:51: MQTT publish: topic 'zigbee2mqtt/Front Door Lock', payload '{"action":"lock","action_source":1,"action_source_name":"rf","action_user":65535,"battery":95,"linkquality":138,"lock_state":"unlocked","state":"UNLOCK"}'
info  2022-03-05 22:18:51: MQTT publish: topic 'zigbee2mqtt/Front Door Lock/action', payload 'lock'
debug 2022-03-05 22:18:57: Received Zigbee message from 'Front Door Lock', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
info  2022-03-05 22:18:57: MQTT publish: topic 'zigbee2mqtt/Front Door Lock', payload '{"battery":95,"linkquality":138,"lock_state":"locked","state":"LOCK"}'
```
And here is the raw configuration from the web ui:
```
"0x000d6f000b64f449": {
    "filtered_optimistic": [
        "lock",
        "lock_state"
    ],
    "friendly_name": "Front Door Lock",
    "optimistic": true
},
```
I am wondering whether the optimistic publish is my issue, since the logs almost make it seem that we are receiving a message from the device that is triggering the publish, as opposed to an optimistic publish.

Thankyou for your help!

sjorge, 2022-03-05:
Ah yes, because it gets the update it republishes the entire state which has the other values unchanged!
You could try to set `debounce` to a slightly larger value and hope the messages get merged.

closed 2022-04-12 by github-actions[bot]