Koenkk/zigbee2mqtt #11715: Ugly OTA message: No image available for imageType '5634'

McGiverGim, 2022-03-05:
### What happened?

When asking for updates, an ugly error appears. I don't know if this is expected, but seems more like a no controlled error. The device is a TS011F.

[image]

### What did you expect to happen?

No get this kind of error. There are several possibilities:
- The device does not must appear in the OTA page to ask for OTAs
- The device must simply not found an update
- It must find an update, but something in the index is wrong

### How to reproduce it (minimal and precise)

Simply go to OTA page and press to ckeck for updates.

### Zigbee2MQTT version

1.24.0-1

### Adapter firmware version

20211217

### Adapter

Egony E72

### Debug log

```
debug 2022-03-05 16:10:11: Received Zigbee message from 'enchufe_comedor_lampara', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":67,"imageType":5634,"manufacturerCode":4098}' from endpoint 1 with groupID 0
debug 2022-03-05 16:10:11: Device 'enchufe_comedor_lampara' requested OTA
debug 2022-03-05 16:10:11: ZigbeeOTA: downloaded main index
info  2022-03-05 16:10:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'enchufe_comedor_lampara' (No image available for imageType '5634')","meta":{"device":"enchufe_comedor_lampara","status":"check_failed"},"type":"ota_update"}'
info  2022-03-05 16:10:12: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"enchufe_comedor_lampara"},"error":"Failed to check if update available for 'enchufe_comedor_lampara' (No image available for imageType '5634')","status":"error","transaction":"zwak1-3"}'
error 2022-03-05 16:10:12: Failed to check if update available for 'enchufe_comedor_lampara' (No image available for imageType '5634')
debug 2022-03-05 16:10:12: AssertionError [ERR_ASSERTION]: No image available for imageType '5634'
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:113:5)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:193:39)
```

inrepublica, 2022-03-06:
Same here:

**What happened?**

When asking for updates, an ugly error appears. I don't know if this is expected, but seems more like a no controlled error. The device is a TS011F_plug_3.
[https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3](url)

**How to reproduce it (minimal and precise)**

Simply go to OTA page and press to ckeck for updates.

**Zigbee2MQTT version**
1.24.0-1

**Adapter firmware version**
20210708

**Adapter**
zStack3x0

**Debug log**
```
error Failed to check if update available for 'interrupteur_chauffage_lit_cecilia' (No image available for imageType '5634')
error Failed to check if update available for 'interrupteur_chauffage_lit_micka' (No image available for imageType '5634')
```

Ivano62, 2022-03-07:
I've got same errors with Tuya - TS011F_plug_1 (TS011F - _TZ3000_cphmq0q7), Tuya - TS011F_plug_3 (TS011F - _TZ3000_cphmq0q7) and MOES BRT-100-TRV (Tuya TS0601 - _TZE200_b6wax7g0)

Ivano62, 2022-03-08:
On 1.24.0-dev (commit: 42f229b) the same error "No image available for imageType '5634'" for Tuya - TS011F_plug_1 (TS011F - _TZ3000_cphmq0q7), Tuya - TS011F_plug_3 (TS011F - _TZ3000_cphmq0q7) and MOES BRT-100-TRV (Tuya TS0601 - _TZE200_b6wax7g0)

Koenkk, 2022-03-08:
That is expected, but the message below should be gone now:

```
debug 2022-03-05 16:10:12: AssertionError [ERR_ASSERTION]: No image available for imageType '5634'
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:113:5)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:193:39)
```

McGiverGim, 2022-03-09:
Yes, the log message now is gone. Maybe it can be done better with the message in the UI, but at least the assert is gone.

closed 2022-04-17 by github-actions[bot]

Timo2488, 2022-04-17:
Have the same issue and i don´t understand why!? 

My mother have the same plug and  no OTA Update problems with Zigbee2MQTT / [TS011F_plug_3]

McFarTech, 2022-09-20:
Any update on how to update the firmware on these TS011F_plug_3 via OTA?

Timo2488, 2022-09-21:
> Any update on how to update the firmware on these TS011F_plug_3 via OTA? 

Hi! I think, isn't coming a new update. For this issue, I have find a information by the Zigbee2MQTT website...

[image]
https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html

BR

Alfagek, 2023-01-05:
Hi Koen,

Got the firmware update error again that the topic starter in mentioning. With z2m version 1.29.0. 

Get warnings in the HA log:

“ Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'
Template variable warning: 'dict object' has no attribute 'installed_version' when rendering '{{ value_json['update']['installed_version'] }}'”

Latest version in z2m: -1
Installed version: -1

Did try repairing and also removed and add the devices (2x) but no solution. 

Regards,
Marco

h27rolf, 2023-01-13:
Hello @Koenkk , same here with version 1.29.1
In my case, Climate device is https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee

CultUr, 2023-01-26:
Same here with the Climate device:
Zigbee Modell: TS0601
Zigbee Hersteller: _TZE200_hue3yfsn
https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html

seflm, 2023-02-08:
Same here with the TV02 TRV valve on version 1.30.1-1.
https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html

oberdannunes, 2023-02-14:
Same problem here with SONOFF ZBMINI Extreme.

xraver, 2023-02-25:
> Same problem here with SONOFF ZBMINI Extreme. https://www.zigbee2mqtt.io/devices/ZBMINIL2.html#sonoff-zbminil2

Same here
Failed to check if update available for 'Luce Veranda' (No image available for imageType '4')

mdzhigarov, 2023-03-09:
Same issue with 1.30.2 and BRT-100-TRV.
 `2023-03-10 00:39:38MQTT publish: topic 'zigbee2mqtt/Guest Bathroom TRV', payload '{"battery":91,"boost_heating":"OFF","boost_heating_countdown":0,"boost_heating_countdown_time_set":900,"child_lock":"UNLOCK","current_heating_setpoint":19,"eco_mode":"OFF","eco_temperature":19,"linkquality":156,"local_temperature":21,"local_temperature_calibration":0,"max_temperature":24,"min_temperature":15,"position":0,"preset":"manual","programming_mode":"06:00/20°C 11:30/21°C 13:30/22°C 17:30/23°C 06:00/24°C 12:00/23°C 14:30/22°C 17:30/21°C 06:00/19°C 12:30/20°C 14:30/21°C 18:30/20°C","running_state":"idle","system_mode":"heat","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"valve_state":"CLOSED","window":"CLOSED","window_detection":"OFF`

serlinGi, 2023-04-07:
I can confirm that the same error for me on two models of thermostatic valve: 
[BRT-100-TRV] (https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html#moes-brt-100-trv) 
and 
another valve another valve sold as model "tuya-TV05" which Z2M recognized and paired as
[TV02-Zigbee] (https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee)
[image]

surfingbytes, 2023-04-09:
Same issue with Zigbee Model TS0601 Zigbee Manufacturer _TZE200_hue3yfsn, recognized as https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee, and the documentation even says "This device supports OTA updates"

BartGysens, 2023-05-08:
Same issue for this one: Tuya TV02
https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee

How can we fix this?

jonathh21, 2023-08-10:
Same issue for me for BRT-100-TRV :(

Lstick, 2023-08-16:
Same issue for TONGOU TO-Q-SY2-163JZT
https://www.zigbee2mqtt.io/devices/TO-Q-SY2-163JZT.html
Zigbee2MQTT version: 1.32.2-1

`Zigbee2MQTT:debug 2023-08-16 11:40:30: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Kitchen switch","transaction":"w52qs-1"}'
Zigbee2MQTT:info  2023-08-16 11:40:30: Checking if update available for 'Kitchen switch'
Zigbee2MQTT:info  2023-08-16 11:40:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Kitchen switch'","meta":{"device":"Kitchen switch","status":"checking_if_available"},"type":"ota_update"}'
Zigbee2MQTT:debug 2023-08-16 11:40:30: Check if update available for '0xe0798dfffe51e846' (TS011F)
Zigbee2MQTT:debug 2023-08-16 11:40:30: Using endpoint '1'
Zigbee2MQTT:debug 2023-08-16 11:40:30: Received Zigbee message from 'Kitchen switch', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":67,"imageType":5634,"manufacturerCode":4098}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-08-16 11:40:30: Got OTA request '{"fieldControl":0,"manufacturerCode":4098,"imageType":5634,"fileVersion":67}'
Zigbee2MQTT:debug 2023-08-16 11:40:30: ZigbeeOTA: downloaded main index
Zigbee2MQTT:info  2023-08-16 11:40:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Kitchen switch' (No image available for imageType '5634')","meta":{"device":"Kitchen switch","status":"check_failed"},"type":"ota_update"}'
Zigbee2MQTT:info  2023-08-16 11:40:30: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Kitchen switch"},"error":"Failed to check if update available for 'Kitchen switch' (No image available for imageType '5634')","status":"error","transaction":"w52qs-1"}'
Zigbee2MQTT:error 2023-08-16 11:40:30: Failed to check if update available for 'Kitchen switch' (No image available for imageType '5634')
Zigbee2MQTT:debug 2023-08-16 11:40:30: Error: No image available for imageType '5634'
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:114:15)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Object.isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:273:18)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:264:29)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)
Zigbee2MQTT:debug 2023-08-16 11:40:33: Received Zigbee message from 'Kitchen switch', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":67,"imageType":5634,"manufacturerCode":4098}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-08-16 11:40:33: Device 'Kitchen switch' requested OTA
Zigbee2MQTT:debug 2023-08-16 11:40:36: Received Zigbee message from 'Kitchen switch', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":67,"imageType":5634,"manufacturerCode":4098}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-08-16 11:40:36: Device 'Kitchen switch' requested OTA
Zigbee2MQTT:debug 2023-08-16 11:40:39: Received Zigbee message from 'Kitchen switch', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":67,"imageType":5634,"manufacturerCode":4098}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-08-16 11:40:39: Device 'Kitchen switch' requested OTA
Zigbee2MQTT:debug 2023-08-16 11:40:42: Received Zigbee message from 'Kitchen switch', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":67,"imageType":5634,"manufacturerCode":4098}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-08-16 11:40:42: Device 'Kitchen switch' requested OTA`

smarteshome2020, 2023-08-27:
Hi, i'm now using the SPH15 (TS011F_pluy_3) too and have the same Error. 

"Failed to check if update available for 'S15-frei' (No image available for imageType '5634" Is there a solution or is it normal? 

Thanks for the  feedback.

tracetechnical, 2023-10-25:
Same issue here. *bump*

Sn00kiT, 2023-11-08:
Same Here for the TV02. No Firmware downloaf possible. The Thermostat is Not working correctly either. The Temperature Always Drops to a lower Temperature than set in Automode.

rodrigofragadf, 2023-12-27:
The same problem here, I have two TS011F_plug_1, I updated one yesterday and it worked normally but when I tried to update the other today it presented this problem. Zigbee2mqtt 1.34.0

rhmunich, 2023-12-27:
> The same problem here, I have two TS011F_plug_1, I updated one yesterday and it worked normally but when I tried to update the other today it presented this problem. Zigbee2mqtt 1.34.0

I can confirm this issue. I updated a **TS011F_plug_1** yesterday but as of today I am receiving the error message:
Update of 'device name' failed (No image available for imageType '54179')

How can this be fixed?

BenCos17, 2023-12-27:
just showed up for me also with a TS011F plug
[image]

Gleras, 2023-12-28:
confirm TS011F plug. it throwing this error from yesterday

Markourai, 2023-12-29:
Today it worked for me

Gleras, 2023-12-29:
yes. today it worked. temporary failure

BenCos17, 2023-12-29:
Yep it worked fine for me also just now when I checked

Rustam2000G, 2024-02-07:
Same problem for BlitzWolf BW-SHP15. 
Zigbee Manufacturer _TZ3000_u5u4cakc. 
Model
 TS011F_plug_3.

sleveille4, 2024-10-26:
same "no new image available" for sonoff TRVZB. but in my case, only ONE (so far) of them (i have 6), they all were fw 1.1.5 and 1.2.1 just got released, 3 updated without problem, a 4th in progress but one throw the error.....