Koenkk/zigbee2mqtt #11722: TS0201 (_TZ2000_a476raq2) battery reported as NULL

yaruslavm, 2022-03-05:
### What happened?

2 devices , new out of the box
[https://www.aliexpress.com/item/4001276768541.html?spm=a2g0o.order_list.0.0.21ef1802yNK701](url)
paired without issues but not reporting some items

Zigbee2MQTT:info  2022-03-05 18:55:55: MQTT publish: topic 'zigbee2mqtt/Dining_Temp', payload '{"battery":null,"humidity":66.7,"last_seen":"2022-03-05T16:09:41.724Z","linkquality":null,"temperature":18.1,"voltage":null}'
Zigbee2MQTT:info  2022-03-05 18:55:55: MQTT publish: topic 'zigbee2mqtt/Living_Temp', payload '{"battery":null,"humidity":64.9,"last_seen":"2022-03-05T15:29:54.742Z","linkquality":null,"temperature":18.4,"voltage":null}'

repairing doesn't work, happens on both.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

sonof zStack3x0

b16rdp, 2022-03-08:
Thank you yaruslavm for reporting this issue.  Just to let you know you are not alone.  I have about 10 of these on various Z2M gateways.  I have had to update one gateway to the new SONOFF Zigbee 3.0 USB Dongle Plus (as the CC2531 was crashing with memory errors, every few days).  I am using a development firmware as I was having issues with a direct swap from CC2531 to SONOFF.  Basically I worked out you can swap the dongles but you can then only pair new devices next to the coordinator.  To do it properly you have to change the Panid, generate new keys and repair everything.

Anyway long story short, the TS0201 devices on the SONOFF dongle does not expose the battery data.  On my old CC2531 dongles all present battery data as wall as temp data.

I can take a TS0201, pair on the new SONOFF dongle and I only get temp, humidity and linkquality.   If I take the same sensor and pair with the CC2531 I get the battery data also.

I am using 1.24.0 on all Z2M gateways.

Most grateful if anyone has found a work around.

Liionboy, 2022-03-11:
Hello, any news regarding this problem? why on ZHA with Sonoff Dongle all work great?

shirou93, 2022-03-23:
I have sonoff dongle and 2 devices. One get battery info and a other one say battery and voltage null. Half a year ago it worked properly on both.

closed 2022-04-30 by github-actions[bot]