Koenkk/zigbee2mqtt #11723: Humidity sensor rounding - TS0601 / _TZE_bjawzodf

lintaba, 2022-03-05:
### What happened?

It was discussed earlier in #11432 that humidity sensor reports bad value.
Not sure if it happens with other devices too, but definetly with the one in the title.

fix is easy and quick (fromZigbee.js@tuya_temperature_humidity_sensor needs a ‘/10’ for humidity too), however I’m not sure if it affects other devices than this one.
It might be better to handle it as a new case for tuya device sensorsets.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

/

Koenkk, 2022-03-06:
@sbur83 could you check if the humidity is correctly reported for the `_TZE200_bq5c8xfe`? ([mr](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3643))

sbur83, 2022-04-06:
> 

Sorry I only just seen this was active. I'm not a Dev and not sure what I done here, I followed the herdsman instructions and done my best. I don't believe this sensor supports humidity; the battery sensor does not report state, this might be possible to expose; it reports tamper status which I have not tested.

closed 2022-05-16 by github-actions[bot]