Koenkk/zigbee2mqtt #11728: [New device support]: Immax Neo Smart Keypad

paffeke, 2022-03-06:
### Link

https://www.tipa.eu/en/smart-keyboard-immax-neo-07505l-zigbee-tuya/d-227860/

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0x847127fffe0571c4","nwkAddr":966,"manufId":4098,"manufName":"_TZE200_n9clpsht","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_n9clpsht","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1646527477163,"defaultSendRequestWhen":"immediate"}

### Comments

I do not know how to make this device works. Tried a simple external converter from Tuya, but i don't know how to procede (sadly i'm not a developer).
JS converter copied and pasted from [https://github.com/Koenkk/zigbee-herdsman-converters/issues/2623](here)
Maybe someone of you could make it works.
Thanks!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_n9clpsht'
        },
    ],
    model: '07505L', // Vendor model number, look on the device for a model number
    vendor: 'Immax', // Vendor of the device (only used for documentation and startup logging)
    description: 'Immax Neo Smart Keypad', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
                fz.ignore_basic_report,
                {
                  cluster: "manuSpecificTuya",
                  type: [
                        "commandSetDataResponse",
                        "commandGetData",
                        "commandActiveStatusReport",
                  ],
                  convert: (model, msg, publish, options, meta) => {
                        const dp = msg.data.dp;
                        const armModeImmax = {
                                26: 'disarm',
                                27: 'arm_away',
                                28: 'arm_home',
                                29: 'sos',
                        };

                        return {action: armModeImmax[dp]};
                  },
                },
        ],
    exposes: [e.action(['disarm', 'arm_home', 'arm_away', 'sos'])],
    toZigbee: [],
};
```

paffeke, 2022-04-06:
> Hello, Any news for support of this device?

Sadly not :(

cdr93, 2022-04-14:
Hello,

I'm also having issues with this keyboard.

Could someone please advise if any information is missing to enable integration into zigbee2mqtt?

Cordially,

geniesek, 2022-05-03:
Hello, 

Same for me. :)

PC

paffeke, 2022-05-06:
Still no news about this feature :(

geniesek, 2022-05-12:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_n9clpsht'
        },
    ],
    model: '07505L', // Vendor model number, look on the device for a model number
    vendor: 'Immax', // Vendor of the device (only used for documentation and startup logging)
    description: 'Immax Neo Smart Keypad', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
		fz.ignore_basic_report,
		{
		  cluster: "manuSpecificTuya",
		  type: [
			"commandSetDataResponse",
			"commandGetData",
			"commandActiveStatusReport",
			"commandDataResponse"
		  ],
		  convert: (model, msg, publish, options, meta) => {
			const dp = msg.data.dpValues[0].dp;
			const armModeImmax = {
				26: 'disarm',
				27: 'arm_away',
				28: 'arm_home',
				29: 'sos',
			};

			return {action: armModeImmax[dp]};
		  },
		},
	],
    exposes: [e.action(['disarm', 'arm_home', 'arm_away', 'sos'])],
    toZigbee: [],
};

module.exports = definition;
`

closed 2022-06-19 by github-actions[bot]

littlegrizmin, 2022-07-06:
Has anyone managed to get it working?

pimseb, 2022-11-20:
Hello. I need a zigbee keypad. This one is not so expensive and its design is great.
Is it supporter into zigbee2mqtt at the moment ?
Thank you

pierrepinon, 2022-11-25:
I bought this keyboard today, and convert works good thx @geniesek

pimseb, 2022-11-30:
I've bought it but I'm not able to pair it in zigbee2mqtt
```Nov 30 18:28:13 raspberrypi npm[22668]: Zigbee2MQTT:info  2022-11-30 18:28:13: MQTT publish: topic 'zigbee2mqtt2/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x847127fffe057216","ieee_address":"0x847127fffe057216","status":"successful","supported":false},"type":"device_interview"}'
Nov 30 18:28:13 raspberrypi npm[22668]: Zigbee2MQTT:info  2022-11-30 18:28:13: MQTT publish: topic 'zigbee2mqtt2/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x847127fffe057216","supported":false},"type":"pairing"}'
Nov 30 18:28:16 raspberrypi npm[22668]: Zigbee2MQTT:warn  2022-11-30 18:28:16: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_n9clpsht'
Nov 30 18:28:17 raspberrypi npm[22668]: Zigbee2MQTT:warn  2022-11-30 18:28:17: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```
What am I supposed to do with the code of @geniesek ? Or should I write a new version of zigbee2mqtt ?
Thank you

geniesek, 2022-12-04:
Yes, this is the exact thing you need to do... just register external converter. This step makes you zigbee2mqtt to understand basic information from the keypad. 
Thank you for explaining for newbies. :)

pimseb, 2022-12-04:
I'm quite satisfied with it. I just wanted to have a device that send arm_away or disarm through zigbee2mqtt.
All the automation is handled by home assistant (alarmo plugin). I also have another keypad (zwave zipato) in another part of my house. For my usage it's what I was looking for

marijandomi, 2023-01-17:
I bought this keyboard today. I like it and it's not expensive, but unfortunately, it has four actions... (arm_away, disarm, arm_home, sos). 10 codes can be programmed, but each code only triggers the above 4 actions... it would be great if it could, for example, show which code triggered the arm_away action. I bought it to open the garage door with one code and the sliding fence with another code. but unfortunately, I can't use it like that. I can only use code + arm_away to open the garage door, and code + disarm opens the sliding fence.

Xavoo, 2023-06-26:
Hello ! 

After a few hours trying to understand how to use this keypad, I finally understood, and i noticed a mistake might be present into this code 

I finally get this keypad to work by replacing "commandDataResponse" by "commandDataReport"

I thought it might be useful for others :)

geniesek, 2023-06-27:
> And I add that "24" allows to get the "tamper switch" info, the little switch that indicates when the keypad is unmounted

Great

captshadab, 2023-09-13:
> Yes, this is the exact thing you need to do... just register external converter. This step makes you zigbee2mqtt to understand basic information from the keypad. Thank you for explaining for newbies. :)
> 
> You can register external converter via Settings / External converters: just register the js file and submit. I suggest to check the effect checking the configuration.yaml just to be sure. Final result will be visible, while you will try to use it.
> 
> This keypad PROs:
> 
> * easy to use, some functionality inbuilt
> * in some angle of view: seems a bit safer, but block more systematic activities (controlling zones etc.)
> 
> CONs:
> 
> * I expected I will get the code from keyboard or user identifier to be able to let the logic to decide next steps. Instead only command is sent.
> * some more work needed to be able to "write" to keypad, who wants to program it? reasons: sync two keypads in different entrances, automatic management from central unit etc.
> 
> Comment: When buying this keypad, I expected to get something like LOCK {pin} or UNLOCK {pin} . I also expected the central unit to decide action based on incoming information, manage user credentials, lock/unlock activities driven from central unit etc.

Hi. Can you explain me one more time what do I have to enter and exactly where in yaml file.

Slightly confused and do not want to break my up and running Hass.

I am planning to buy this keypad as this is the only cheapest option available in matket which can work with alarmo. Any new updates since last time you posted your comment

NikolamViser, 2024-06-14:
Hello,

I bought this keypad, but I'm unable to pair it. I tried also external converter as explained above, but no luck.
I put the keypad in pairing mode, it starts the interview, but it always finishes with:
Device ... with Zigbee model 'TS0601' and manufacturer name '_TZE200_moycceze' is NOT supported.

I don't know what else to try. Can anyone point me in a right direction?

NikolamViser, 2024-06-15:
I solved it with the help of this thread [#20273](https://github.com/Koenkk/zigbee2mqtt/issues/20273)
I realized I have different model than the one currently supported by Z2M, so I had to use different external converter.