Koenkk/zigbee2mqtt #11729: Aqara Presence Detector FP1 (RTCZCGQ11LM) is NOT supported

piggydamn, 2022-03-06:
### What happened?

my adapter is cc2652p, and i Updated the firmware to 20211114, but the device is not supported.

### What did you expect to happen?

Join the device to z2m, to work properly with the new firmware.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211114

### Adapter

cc2652p

### Debug log

warn  2022-03-06 10:18:59: `permit_join` set to  `true` in configuration.yaml.
warn  2022-03-06 10:18:59: Allowing new devices to join.
warn  2022-03-06 10:18:59: Set `permit_join` to `false` once you joined all devices.
warn  2022-03-06 10:19:00: Using experimental new availability feature
warn  2022-03-06 10:21:21: Device '0x54ef441000473044' with Zigbee model 'lumi.motion.ac01' and manufacturer name 'aqara' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
warn  2022-03-06 10:21:22: Received message from unsupported device with Zigbee model 'lumi.motion.ac01' and manufacturer name 'aqara'
warn  2022-03-06 10:21:22: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

McGiverGim, 2022-03-06:
Have you tried to update to version 1.24? It was added very recently https://github.com/Koenkk/zigbee2mqtt/issues/11019

piggydamn, 2022-03-06:
> Have you tried to update to version 1.24? It was added very recently #11019

the latest version for cc2652p is [20211217]，isnt it？

McGiverGim, 2022-03-06:
I talk about zigbee2mqtt version, not the adapter.

piggydamn, 2022-03-06:
> 

it works，thx man

closed 2022-03-06 by Koenkk

arthur-morgan-1, 2022-03-09:
Does it support zones in Zigbee2mqtt?

Otnow, 2022-03-09:
> Does it support zones in Zigbee2mqtt?

At the moment, no.

arthur-morgan-1, 2022-03-10:
Is it supported in any level on deconz?

Otnow, 2022-03-10:
> Is it supported in any level on deconz?

[Looks like no](https://github.com/search?q=org%3Adresden-elektronik+RTCZCGQ11LM).

piggydamn, 2022-03-19:
> Does it support zones in Zigbee2mqtt?

not yet