Koenkk/zigbee2mqtt #11735: Devices intermittently don't respond

Sku1ly, 2022-03-06:
### What happened?

Sometimes, and it isn't consistent, devices fail to respond to commands through the Z2M GUI and HA. I understand this could happen with signal issues but I have a large network and I'm running on a channel that doesn't overlap with 2.4ghz.

Debug log is attached.

Thanks for your help.

### What did you expect to happen?

The blind to close when command sent

### How to reproduce it (minimal and precise)

It is intermittent but when it occurs once, it usually continues until the next day.

### Zigbee2MQTT version

1.2.4.0

### Adapter firmware version

0x26660700

### Adapter

ConBee2

### Debug log

Zigbee2MQTT:debug 2022-03-06 14:34:44: Received MQTT message on 'zigbee2mqtt/Office Blind/set' with data 'CLOSE'
Zigbee2MQTT:debug 2022-03-06 14:34:44: Publishing 'set' 'state' to 'Office Blind'
2022-03-06T14:34:44.120Z zigbee-herdsman:controller:endpoint Command 0xb4e3f9fffe61e546/1 closuresWindowCovering.downClose({}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-03-06T14:34:44.121Z zigbee-herdsman:controller:endpoint Sending to 0xb4e3f9fffe61e546/1 when active
2022-03-06T14:34:44.327Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 89
2022-03-06T14:34:44.329Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-03-06T14:34:44.329Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-03-06T14:34:44.405Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-03-06T14:34:44.405Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-03-06T14:34:44.406Z zigbee-herdsman:deconz:driver query aps data indication
2022-03-06T14:34:44.411Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 90
2022-03-06T14:34:44.413Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 90 srcAddr: 0xf14 destAddr: 0x0 profile id: 0x104 cluster id: 0xb04 lqi: 255
2022-03-06T14:34:44.413Z zigbee-herdsman:deconz:frameParser response payload: 8,97,10,11,5,41,207,8
2022-03-06T14:34:44.414Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":97,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1291,"dataType":41,"attrData":2255}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":3860,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'

Stooovie, 2022-03-06:
Same here, started today morning. 100% solid for three years prior. Z2M 1.24.0-1, HA 2022.3.1, Slaesh CC2652RB stick. Missed clicks on all my Aqara buttons, huge delays. I´m not aware of any updates in the last few days except Z2M update but that was working for four days fine.

Sku1ly, 2022-03-07:
No I'm in the UK and it isn't enabled over here.

It's been ongoing on and off since August 2021 but I'm only just getting fed up enough with it. I have a Sonoff 3.0 stick I can try but looking at your issue it doesn't seem to be stick related.

sanasol, 2022-03-10:
Same here, Tuya Buttons sometimes doesn't trigger mqtt events.

Log about click, but there is no mqtt publish after it.

```
Debug Received Zigbee message from 'Double 1', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 2 with groupID 0
```

No obvious reason and can't repeat it.

Installed it less than month ago, so I can't say it worked ok before it. Missclicks started from first days of usage probably.

estebanz01, 2022-03-21:
I'm seeing this same behaviour, but more constant after upgrading to 1.24.0. Here's the error on the logs:
```javascript
Mar 20 18:18:26 raspberrypi npm[27744]: UnhandledPromiseRejection: 
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
The promise rejected with the reason "APS TIMEOUT".
```

Sku1ly, 2022-03-24:
It's getting worse now I am able to consistently reproduce this issue with specific devices.

Zigbee2MQTT:debug 2022-03-24 11:12:31: Received Zigbee message from 'Coordinator', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 30, ignoring since it is from coordinator
2022-03-24T11:12:31.036Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-03-24T11:12:31.166Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2022-03-24T11:12:31.166Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-03-24T11:12:31.167Z zigbee-herdsman:deconz:driver query aps data confirm
2022-03-24T11:12:31.167Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 201
2022-03-24T11:12:31.169Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x1e request id: 177 confirm status: 0
2022-03-24T11:12:31.169Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2022-03-24 11:12:31: MQTT publish: topic 'zigbee2mqtt/Landing 1', payload '{"brightness":254,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-03-24 11:12:31: MQTT publish: topic 'zigbee2mqtt/Landing', payload '{"brightness":254,"color_mode":"color_temp","color_temp":333,"state":"ON"}'
Zigbee2MQTT:info  2022-03-24 11:12:31: MQTT publish: topic 'zigbee2mqtt/Landing 2', payload '{"brightness":254,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-03-24 11:12:31: MQTT publish: topic 'zigbee2mqtt/Landing 3', payload '{"brightness":254,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-03-24 11:12:31: MQTT publish: topic 'zigbee2mqtt/Landing 4', payload '{"brightness":254,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-03-24 11:12:31: MQTT publish: topic 'zigbee2mqtt/Landing 5', payload '{"brightness":254,"color_mode":"color_temp","color_temp":333,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}'

estebanz01, 2022-03-24:
4 days running on 1.23.0 without problems, so it's definitely something on the latest version. I'm a bit busy this weekend, but I can give it a try to identify which commit is triggering that error.

estebanz01, 2022-04-24:
I'm still doing some testing and I got caught with the 1.25.0 version and lots of work stuff 😅 but I'm still trying to identify which commit triggers it.

estebanz01, 2022-05-19:
it seems it got fixed on latest version. I did the full upgrade and it's working as expected. Can anyone else with the same issue, confirm ? I couldn't find the commit that broke everything.

closed 2022-06-27 by github-actions[bot]

Kill0p, 2022-08-08:
I was testing how long Zigbee2MQTT can be online without any interruptions, I have no devices connected right now just using conbee II and still I have got this error.

UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "APS TIMEOUT"

Verions
Zigbee2MQTT version 1.27.0 (commit #6d25fb2d)
zigbee-herdsman (0.14.46)

urban-thinking, 2022-10-17:
Got same error ... as well with no devices connected.

Zigbee2MQTT version 1.28.0 (commit #03ba647)
zigbee-herdsman (0.14.62)