Koenkk/zigbee2mqtt #11736: Huge delays and intermittent operation, started today

Stooovie, 2022-03-06:
### What happened?

No issues for 3 years, since today, huge delays (like 5-10s), missed Zigbee button presses (but all devices do work eventually). 

### How to reproduce it (minimal and precise)

All Zigbee operation is affected.

### Zigbee2MQTT version

 1.24.0-1

### Adapter firmware version

20210120

### Adapter

Slaeh.sh CC2652RB

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8193248/log.txt)

Sku1ly, 2022-03-06:
Huge delay? Mine just took 4 hours to respond! :(

Stooovie, 2022-03-06:
On the missed button presses, I only see this in the log:
`Debug Received Zigbee message from 'Xiaomi Button Bed', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0`

the actual payload is and MQTT PUBLISH is totally missing

Stooovie, 2022-03-06:
Also the map now looks like this, which is super weird and not at all how it looked before
[image]
.

Stooovie, 2022-03-06:
It looks like the Slaeh.sh stick has failed, I have swapped in the old CC2531 and it all just works again. Thanks!

Stooovie, 2022-03-07:
No, not hardware problem. The CC2531 is starting to flake out too. It's unlikely that two Zigbee sticks would fail at the same time. Any tips? Thanks!

Stooovie, 2022-03-07:
To add to my own comments for posterity - is it possible that **Amazon Sidewalk** that got turned ON by an Echo device reset is the culprit? It does use the same frequency of 900MHz (BLE does too)... Too early to tell but after turning it off, it seems my network is stable again.

Stooovie, 2022-03-10:
The problem is gone, fine for three days... I can't make any sense of it.

closed 2022-04-17 by github-actions[bot]