Koenkk/zigbee2mqtt #11738: HA: The 'send_if_off' option is deprecated, please remove it from your configuration

machielw, 2022-03-06:
### What happened?

Popp 701721 radiator valve gives this error.
Home assistant had Some Breaking changes in the MQTT integration. 

### What did you expect to happen?

. . .

### How to reproduce it (minimal and precise)

. . .

### Zigbee2MQTT version

 1.24.0-dev commit: f7c6207

### Adapter firmware version

. . .

### Adapter

Conbee 2

### Debug log

Logger: homeassistant.components.mqtt.climate
Source: helpers/config_validation.py:817 
Integration: MQTT (documentation, issues) 
First occurred: 22:54:57 (10 occurrences) 
Last logged: 22:54:57

The 'send_if_off' option is deprecated, please remove it from your configuration
The 'hold_modes' option is deprecated, please remove it from your configuration

gopheridze, 2022-03-07:
I've got same errors with Tuya TS0601

2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'send_if_off' option is deprecated, please remove it from your configuration
2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'hold_modes' option is deprecated, please remove it from your configuration
2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'away_mode_command_topic' option is deprecated, please remove it from your configuration
2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'away_mode_state_template' option is deprecated, please remove it from your configuration
2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'away_mode_state_topic' option is deprecated, please remove it from your configuration
2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'hold_command_topic' option is deprecated, please remove it from your configuration
2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'hold_state_template' option is deprecated, please remove it from your configuration
2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'hold_state_topic' option is deprecated, please remove it from your configuration
2022-03-07 04:10:00 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'hold_modes' option is deprecated, please remove it from your configuration

the config topic in MQTT
`{
  "action_template":"{% set values = {None:None,'idle':'off','heat':'heating','cool':'cooling','fan_only':'fan'} %}{{ values[value_json.running_state] }}",
  "action_topic":"zigbee2mqtt/LR Thermostat",
  "availability":[
        {
          "topic":"zigbee2mqtt/bridge/state"
        },
        {
          "topic":"zigbee2mqtt/LR Thermostat/availability"
        }
  ],
  "availability_mode":"all",
  "away_mode_command_topic":"zigbee2mqtt/LR Thermostat/set/away_mode",
  "away_mode_state_template":"{{ value_json.away_mode }}",
  "away_mode_state_topic":"zigbee2mqtt/LR Thermostat",
  "current_temperature_template":"{{ value_json.local_temperature }}",
  "current_temperature_topic":"zigbee2mqtt/LR Thermostat",
  "device":{
      "identifiers":[
        "zigbee2mqtt_0x842e14fffeefe7e0"
      ],
      "manufacturer":"TuYa",
      "model":"Radiator valve with thermostat (TS0601_thermostat)",
      "name":"LR Thermostat"
  },
  "hold_command_topic":"zigbee2mqtt/LR Thermostat/set/preset",
  "hold_modes":[
      "schedule",
      "manual",
      "boost",
      "complex",
      "comfort",
      "eco"
  ],
  "hold_state_template":"{{ value_json.preset }}",
  "hold_state_topic":"zigbee2mqtt/LR Thermostat",
  "json_attributes_topic":"zigbee2mqtt/LR Thermostat",
  "max_temp":"35",
  "min_temp":"5",
  "mode_command_topic":"zigbee2mqtt/LR Thermostat/set/system_mode",
  "mode_state_template":"{{ value_json.system_mode }}",
  "mode_state_topic":"zigbee2mqtt/LR Thermostat",
  "modes":[
      "heat",
      "auto",
      "off"
  ],
  "name":"LR Thermostat",
  "temp_step":0.5,
  "temperature_command_topic":"zigbee2mqtt/LR Thermostat/set/current_heating_setpoint",
  "temperature_state_template":"{{ value_json.current_heating_setpoint }}",
  "temperature_state_topic":"zigbee2mqtt/LR Thermostat",
  "temperature_unit":"C",
  "unique_id":"0x842e14fffeefe7e0_climate_zigbee2mqtt"
}`

jfs75, 2022-03-07:
Same issue for me: 

2022-03-07 07:09:39 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'send_if_off' option is deprecated, please remove it from your configuration
2022-03-07 07:09:39 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'hold_modes' option is deprecated, please remove it from your configuration
2022-03-07 07:09:39 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'send_if_off' option is deprecated, please remove it from your configuration
2022-03-07 07:09:39 WARNING (MainThread) [homeassistant.components.mqtt.climate] The 'hold_modes' option is deprecated, please remove it from your configuration

Ivano62, 2022-03-07:
I've got same errors with MOES BRT-100-TRV (Tuya TS0601 - _TZE200_b6wax7g0), despite having removed from the configuration the parameters indicated as deprecated

mrwogu, 2022-03-27:
Same here [TS0601_thermostat](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#tuya-ts0601_thermostat)

Maurizion, 2022-03-28:
Same Issue here [TS0601_thermostat](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#tuya-ts0601_thermostat)

Enzokot, 2022-04-28:
Keep alive

csszokolai, 2022-05-08:
Hello, same problem here. BHT-002

11harveyj, 2022-05-10:
Same issue here.. TS0601_thermostat

```
The 'away_mode_state_topic' option is deprecated, please remove it from your configuration
The 'hold_command_topic' option is deprecated, please remove it from your configuration
The 'hold_state_template' option is deprecated, please remove it from your configuration
The 'hold_state_topic' option is deprecated, please remove it from your configuration
The 'hold_modes' option is deprecated, please remove it from your configuration
```

MJeka, 2022-05-29:
+ I confirm

aegjoyce, 2022-06-12:
I’ve also got this on a Moes BHT-002. Not sure where to start with fixing this, should it be in zigbee-herdsman-converters?

aegjoyce, 2022-06-12:
So I think this the issue:

These thermostats often use the withPreset statement (see https://github.com/Koenkk/zigbee-herdsman-converters/blob/e8428204244eea288617a0eb972e7b961e109b9d/devices/moes.js) which then triggers all of the above deprecated exposes from the below (https://github.com/Koenkk/zigbee2mqtt/blob/a252914ecd1325cc1099f03af61872eb5cf0a405/lib/extension/homeassistant.ts#L285-L292)

The correct thing to do would be to remove these lines from homeassistant.ts given none of these exposes will be supported any more - however I wonder if this would then remove the ability to select between Hold and Program modes (using the Moes BHT-002 as an example)

Coollision, 2022-06-14:
Looks like those modes also have been removed from the documentation

https://www.home-assistant.io/integrations/climate.mqtt/

Koenkk, 2022-06-15:
The `hold_mode*` and `away_mode*` errors should be fixed now. I don't know about the `send_if_off`, I don't see this mentioned in the z2m codebase (but maybe fixing the hold/away also fixes that error). Could you check if these errors are gone now?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

aegjoyce, 2022-06-15:
@Koenkk can confirm latest changes on dev branch fix the issue for me, no more errors logged. Will this get rolled into next month's stable release? Many thanks for your efforts

csszokolai, 2022-06-15:
> The `hold_mode*` and `away_mode*` errors should be fixed now. I don't know about the `send_if_off`, I don't see this mentioned in the z2m codebase (but maybe fixing the hold/away also fixes that [...]

Hello @Koenkk, i also confirm, no more errors logged. Thank you for your work again!

Koenkk, 2022-06-16:
Great, thanks for confirming.