Koenkk/zigbee2mqtt #11739: Zigbee2MQTT after updaet from 1.21 to 1.24 z2m dont start with systemctl.

alfakenzo, 2022-03-07:
### What happened?

Hi,
after update from 1.21 to 1.24 z2m dont start with systemctl. With npm start in /opt/zigbee2mqtt it start without errors.
I got in the logs this error.

error 2022-03-07 06:19:25: Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/>
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 72. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1057:18)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)

Dont know how to solve this problem tried to with  npm cache clean --force and removed the node_moduls and then nmp install.
BUt got always the same error. DOnt know how to get it working. Maybe you can help me guys.
Big thx

### Zigbee2MQTT version

1.24

### Adapter firmware version

fronted dont start

### Adapter

Conbee 2

### Debug log

debug 2022-03-07 06:19:25: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2022-03-07 06:19:25: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-03-07.06-19-23' filename: log.txt
debug 2022-03-07 06:19:25: Removing old log directory '/opt/zigbee2mqtt/data/log/2022-03-07.06-18-37'
info  2022-03-07 06:19:25: Starting Zigbee2MQTT version 1.24.0 (commit #7a2ddf24)
info  2022-03-07 06:19:25: Starting zigbee-herdsman (0.14.20)
debug 2022-03-07 06:19:25: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath">
error 2022-03-07 06:19:25: Error while starting zigbee-herdsman
error 2022-03-07 06:19:25: Failed to start zigbee
error 2022-03-07 06:19:25: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2022-03-07 06:19:25: Exiting...
error 2022-03-07 06:19:25: Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/>
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 72. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1057:18)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)

alfakenzo, 2022-03-07:
hi Koenkk it still the same.

Koenkk, 2022-03-08:
@alfakenzo try upgrading your node.js version.

alfakenzo, 2022-03-08:
@Koenkk after update it seems to run! THX

closed 2022-03-10 by Koenkk