Koenkk/zigbee2mqtt #11744: Light group is shown as working but nothing happens

malantp, 2022-03-07:
### What happened?

When I try to toggle a group of lights, everything seems to work as it should be (frontend, logs and in HA), but the lights are not turning on or off in real life. I can toggle the lights one by one with no issue, but not in a group.

### What did you expect to happen?

I want my lights to turn on and off in real life when I toggle the group.

### How to reproduce it (minimal and precise)

Create a new group, add some lights in it, publish an MQTT ON state on that group (via HA in my case)
[image]

Thank you for your time!

### Zigbee2MQTT version

1.24.0 commit: 7a2ddf2

### Adapter firmware version

2103

### Adapter

zigate

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8197086/log.txt)

closed 2022-04-14 by github-actions[bot]

maxime1992, 2023-03-22:
Opened https://github.com/Koenkk/zigbee2mqtt/issues/17115 which is a duplicate of this issue just because that one was closed without any answer and I'm having exactly the same issue. Can we please close the duplicate I've created for visibility and reopen this one?

maxime1992, 2023-03-22:
@malantp have you found a fix that you could share please?

malantp, 2023-03-22:
@maxime1992 Yes, since I had some issue with groups and OTA as well, I decided to change my zigate adapter for a Sonoff Dongle Plus and everything is working fine now. Turns out zigate is not fully supported for now.

maxime1992, 2023-03-23:
Interesting, thanks. I've got a [SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P](https://www.amazon.fr/gp/product/B09KXTCMSC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) but still doesn't work for me :thinking:

jafd, 2023-03-25:
I have an Electrollama zzh and some Trådfri bulbs in one room which I hoped to group together and use as one. I distinctly remember that some previous version of z2m had them working just fine. Right now, though, the symptoms are exactly as described in the issue: you recall a scene, you can see in the web UI that the state has changed as programmed, you can see the corresponding messages being published in the log, except that nothing is happening in the actual physical world.

Which is also frustrating because it leads to HomeAssistant thinking the bulbs are in a different state than they actually are.

maxime1992, 2023-03-26:
@jafd maybe put a :+1: on the issue I've opened (#17115) and you're comment there. As it's an open issue if we want to be heard I think it'll be best to follow up there

jafd, 2023-03-26:
I have a followup on this. What helped was deleting the group and re-creating it again. But! I cannot bind a remote to the group. That is, I bind it just fine (supposedly), but when I press buttons, nothing happens. Toggling the state of the group in HA, or recalling a scene in z2m works.

I'm not sure where groups actually live, but it _looks_ like it's partly on the coordinator to be aware of the groups, and maybe it had lost some state information it was supposed to hold without z2m being aware of it.