Koenkk/zigbee2mqtt #11747: No converter available for HUE bulb

UhtredTheBold, 2022-03-07:
### What happened?

When attempting to change the power on behaviour for one of the newer HUE bulbs - Hue white A60 bulb E27 1050lm with Bluetooth (929002469202) & Hue white A60 bulb B22 1055lm with Bluetooth (9290024693) I see the following error

`No converter available for 'hue_power_on_behavior' ("OFF")`

### What did you expect to happen?

I expected a confirmation that the power on behaviour had been changed

### How to reproduce it (minimal and precise)

Send a MQTT command to [zigbee2mqtt/FRIENDLY_NAME/set](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.md#zigbee2mqttfriendly_nameset)

### Zigbee2MQTT version

1.24.0 commit: 7a2ddf2

### Adapter firmware version

20211217

### Adapter

zzh!

### Debug log

Zigbee2MQTT:error 2022-03-07 15:12:54: No converter available for 'hue_power_on_behavior' ("OFF")

UhtredTheBold, 2022-04-07:
I still have this problem with 1.25

mrskycriper, 2022-04-24:
Moreover, power on behaviour on any of my hue lights doesn't even work half of the time. I've tested all documented modes and lamps mostly ignored them

Koenkk, 2022-05-25:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-25 by Koenkk