Koenkk/zigbee2mqtt #11751: [Feature request]: Woox R7060 Garden irrigation control Battery Level

RenePhilipse, 2022-03-07:
### Is your feature request related to a problem? Please describe

The Woox R7060 Garden irrigation control is supported in zigbee2mqtt, but only for switching it on or off. The device is reporting a battery level, but is is not viewable via zigbee2mqtt.

### Describe the solution you'd like

I like to see the status of the battery of my Woox garden irrigation control in zigbee2mqtt and, even better, in Home Assistant.

### Describe alternatives you've considered

I already tried some things myself, but I do not know where to start exactly. Already posted a discussion (#9826) to help get me started, but it got no responses.

### Additional context

See discussion topic #9826.

Koenkk, 2022-03-07:
Added this.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

RenePhilipse, 2022-03-07:
Wow! That’s fast! Thanks and I will have a look at it.

RenePhilipse, 2022-03-14:
Hi @Koenkk I've checked the solution using the DEV branch, and I can confirm that the Battery Level is indeed reported and visible now! Great! Thank you very much for your quick support! Kind Regards, Rene.

closed 2022-03-14 by Koenkk

Remko76, 2022-04-07:
I am using the Dev branch as well, but do not get the battery level. It only shows "N/A" or "NULL". Is there anything I am missing, or do I need to unlink the device from the network and repair it?

[image]

RenePhilipse, 2022-04-07:
@Remko76 
It is already in the latest production branch at the moment. It works fine at my side. Maybe remove and re-add the device?
[image]

RenePhilipse, 2022-04-07:
It can take some time to get the values filled in for the first time. Has to do with the fact that it are battery operated devices that do not communicate that often to save power. Maybe wait for 24 hours to be sure?

Remko76, 2022-04-07:
I did get strange warnings in the log, I do not know if that means anything though (long time HASS user, but pretty new to Zigbee, actually my first Zigbee device...)

```
Warning zigbee-herdsman-converters:WooxR7060: Unrecognized DP #107 with data {"dp":107,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Warning zigbee-herdsman-converters:WooxR7060: Unrecognized DP #106 with data {"dp":106,"datatype":3,"data":{"type":"Buffer","data":[48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48]}}
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"oq2ab-4"}'
```

RenePhilipse, 2022-04-07:
That technical log is also to technical for me, but maybe @Koenkk can tell if that could be an issue.

helepeut, 2022-04-08:
Hi, same for me here, I was waiting for the battery control for the woox irrigation in the production release, but nothing since the beginning of the week :(

floow77, 2022-04-18:
@Koenkk hello, do you know when this fix will be in production release ? Thanks

Koenkk, 2022-04-18:
This is in 1.25.0, but if it doesn't work lets continue in https://github.com/Koenkk/zigbee2mqtt/issues/12163

floow77, 2022-04-19:
Yes it doesn't work in 1.25.0.1.
If I understand well, it'necessary to have a tuya gateway ?

floow77, 2022-04-19:
Ok but this gateway is necessary to retrieve %batterie or using the edge branch is enough ?

jjaviercs, 2022-08-03:
Updated zigbee2mqtt to version 1.27 to test... I had backup but I lost all devices and problem persist, not solve.

acaonweb, 2022-08-23:
#13649