Koenkk/zigbee2mqtt #11754: [New device support]: Gu10 LED Lights Bulb Spotlight 4W RGBW / Smart Life

AlexEamo, 2022-03-07:
### Link

https://www.aliexpress.com/item/1005003754490865.html?srcSns=sns_More&spreadType=socialShare&bizType=ProductDetail&social_params=20498306106&aff_fcid=a26c8607252a41e7ab4f0db65a0d0d9e-1646686257986-03317-_mP2QCxK&tt=MG&aff_fsk=_mP2QCxK&aff_platform=default&sk=_mP2QCxK&aff_trace_key=a26c8607252a41e7ab4f0db65a0d0d9e-1646686257986-03317-_mP2QCxK&shareId=20498306106&businessType=ProductDetail&platform=AE&terminal_id=16cd1ca52b9140b989c3e516ae8110c0&afSmartRedirect=y

### Database entry

{"id":27,"type":"Router","ieeeAddr":"0xa4c1384147ae4f2f","nwkAddr":32310,"manufId":4417,"manufName":"_TZ3210_it1u8ahz","powerSource":"Mains (single phase)"}

### Comments

I tried following the support on the new device page but got stuck at the adding new converters step as I'm running Home Assistant OS.

I added the external_converters reference in the add-on configuration option but the js file I'd added didn't appear to be loaded.

I then added external_converters file reference to configuration.yaml in the zigbee2mqtt folder using terminal which seemed to identify that I'd added the external_converters references (I called the file gu10.js) but got errors on trying to run the addon. This may be that the library files are stored somewhere different when installed through Home Assistant OS but I can't find them to make this work.

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[21:02:22] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[21:02:24] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.2-dev start
> node index.js
(node:358) UnhandledPromiseRejectionWarning: ReferenceError: extend is not defined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:358) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:358) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

### Supported color modes

rgbcct

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

AlexEamo, 2022-03-08:
This may replicate issue #11176 which was closed as stale.

AlexEamo, 2022-03-09:
Managed to resolve it with the following converter (Written for Home Assistant OS)

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0505B'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'GU10', // Vendor model number, look on the device for a model number
    vendor: 'Aubess', // Vendor of the device (only used for documentation and startup logging)
    description: '4W RGBW', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
};

module.exports = definition;

```

closed 2022-03-09 by AlexEamo

ferbulous, 2022-03-16:
Thanks for this, just purchased a similar bulb

psychowood, 2023-02-08:
(Please ignore the above mention to the other issue, it wasn't intended, my bad).

I've got three of this bulbs, all showing the same behaviour: even if they expose the datapoint `color_power_on_behavior`, I believe it is not working as intended (either that, or I misinterpreted it).

- `previous` does not seems too to do anything. Turning the light on (I mean via a physical switch) restores the default cool white - while keeping the previusly set brightness - but reports the previous color/temperature (i.e. you see 'red' but the light is actually white)
- `cutomized`, on the other hand, keeps the color/temperature in the z2m `exposes` panel

What should be the expected behavior?

Btw: I see that it does not support OTA, it does not even appear in the OTA panel. Is there a way to check if this is because it is "officially" not updateable, or just because no one have found the correct link? I could get a Tuya bridge if it's worth trying.