Koenkk/zigbee2mqtt #11757: Can not receive data from motion sensors which are connecting to router device

trongcongbk, 2022-03-08:
### What happened?

I am deploying zigbee2mqtt for my house, it including 10 smart switches, 10 motion sensors (PIR type) and 10 temperature sensor.
When all device are connected directly to my coordinator, my system work well. But zigbee coverage at some areas are not good, so I must add more 2 routers (I am using Sonoff zigbee adapter and flashing it with your router firmware (dev version). 

At this time, all device are connecting to coordinator and 2 routers. But with all motion sensors connecting to router, I cannot receive their messages at my coordinator, and I see these motion sensors still send message to router.

With other motion sensors connecting directly to coordinator, it work normally.

[image]

### What did you expect to happen?

Could you please help me to solve the problem.

### Zigbee2MQTT version

1.23.0-dev commit: aace02cc

### Adapter firmware version

20220103

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[Z10 10_1.zip](https://github.com/Koenkk/zigbee2mqtt/files/8204242/Z10.10_1.zip)

[image]
[image]
[image]
[image]

 Network key: C3:82:85:43:F2:64:49:27:D2:04:7F:30:22:F2:3D:1A

cwilko, 2022-03-12:
I`m having the same issue with SONOFF S26R2ZB as the router. I`m not getting any messages from end devices which connect to the router. Would like to know how to debug this situation.

trongcongbk, 2022-03-29:
[](url
[When shutting down Zigbee Gateway.pdf](https://github.com/Koenkk/zigbee2mqtt/files/8367555/When.shutting.down.Zigbee.Gateway.pdf)
)

Hello everyone,

I add more information about analyzing futher this problem. I see that the main problem is:
- When shutting down the coordinator, the PIR sensor rejoins zigbee network again via a router device.
- After failure to try connect the coordinator, the PIR sensor still send IAS zone status message, but the destination is router device instead of the coordinator.
- When power on the coordinator again, the PIR sensor still send IAS zone status message to router device. So that the coordinator never receive data from PIR sensor device any more. I had tried to send a "read attribute request" command from coordinator (based on dev console at frontend gui) and still get response from PIR sersor.
- I also try with other zigbee gateway (Tuya) and dont see same problem as Zigbee2MQTT gateway.

So our question is:
- Why PIR sensor change the destination of IAS zone status report message
- How to force PIR sensor to alway keep the destinaton (coordinator 0x0000) of IAS zone status report message.

Please help me to solve the problem.

closed 2022-05-06 by github-actions[bot]