Koenkk/zigbee2mqtt #11758: Wiser TRv resets to 20c after 2 hours

madbobmcjim, 2022-03-08:
### What happened?

I'm running a couple of wiser TRVs, and they reset their target temperature to 20c after two hours. 
This seems to be on a 2 hour cadence, rather than being 2 hours after the temperature was previously set.

The TRV is this one: [https://www.zigbee2mqtt.io/devices/WV704R0A0902.html](https://www.zigbee2mqtt.io/devices/WV704R0A0902.html)

Right now I've got Home Assistant looking for the mqtt message with a temperature change and responding to set it back, but I'd rather not need to have this workaround in place.

I've tried running the latest dev branch (as of a few days ago) in case there were fixes in there.

This issue appears to be affecting others as well, it's discussed at the end of the (now closed) Z2M github issue that has the device added, and also on the HA forums. Since the initial issue is closed, I thought I'd try raising a new one :-)

- https://github.com/Koenkk/zigbee2mqtt/issues/2521
- https://community.home-assistant.io/t/drayton-wiser-schneider-electric-itrv-issues/268728/29

### What did you expect to happen?

The TRV to maintain it's heating set point until changed.

### How to reproduce it (minimal and precise)

Change the setpoint to something other than 20C, then wait for a couple of hours.

### Zigbee2MQTT version

1.24.0-dev commit: 3535b2c

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Debug log

Debug log from TRV reporting home, I'll update when I capture the temperature change

```
debug Received Zigbee message from 'StudyTRV', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID null
debug Received Zigbee message from 'StudyTRV', type 'attributeReport', cluster 'wiserDeviceInfo', data '{"deviceInfo":"ADC,1413,1382,1,218,-32768,2700,0,500,100,212,223,225"}' from endpoint 1 with groupID null
info MQTT publish: topic 'zigbee2mqtt/StudyTRV', payload '{"ADC":"1413,1382,1,218,-32768,2700,0,500,100,212,223,225","ALG":"d,1,270,217,59,0,500,1,500,100,100","MOT":"CloseValve","battery":88,"boost":"Up","keypad_lockout":"unlock","linkquality":255,"local_temperature":21.8,"occupied_heating_setpoint":27,"pi_heating_demand":100,"running_state":null,"voltage":2900}'
debug Received Zigbee message from 'StudyTRV', type 'attributeReport', cluster 'wiserDeviceInfo', data '{"deviceInfo":"ALG,d,1,270,217,59,0,500,1,500,100,100"}' from endpoint 1 with groupID null
info MQTT publish: topic 'zigbee2mqtt/StudyTRV', payload '{"ADC":"1413,1382,1,218,-32768,2700,0,500,100,212,223,225","ALG":"d,1,270,217,59,0,500,1,500,100,100","MOT":"CloseValve","battery":88,"boost":"Up","keypad_lockout":"unlock","linkquality":255,"local_temperature":21.7,"occupied_heating_setpoint":27,"pi_heating_demand":100,"running_state":null,"voltage":2900}'
debug Received MQTT message on 'zigbee2mqtt/StudyTRV/set/occupied_heating_setpoint' with data '27'
debug Publishing 'set' 'occupied_heating_setpoint' to 'StudyTRV'
debug Received Zigbee message from 'StudyTRV', type 'attributeReport', cluster 'haDiagnostic', data '{"lastMessageLqi":112,"lastMessageRssi":-72}' from endpoint 1 with groupID null
debug Received Zigbee message from 'StudyTRV', type 'attributeReport', cluster 'haDiagnostic', data '{"57345":46097}' from endpoint 1 with groupID null
debug Received MQTT message on 'zigbee2mqtt/StudyTRV/set/occupied_heating_setpoint' with data '27'
debug Publishing 'set' 'occupied_heating_setpoint' to 'StudyTRV'
```

madbobmcjim, 2022-04-12:
Bump, I suppose. If there's any further information I can provide to aid this one let me know. I'd like to get more of these TRVs as they're easily available here, but only if I can get them working well :-)

closed 2022-05-21 by github-actions[bot]

tszyszko, 2022-10-23:
How reliable is this workaround? Does the actuator move before the setpoint gets updated ? I'm seeing the same issue on ZHA and wondering whether to try to implement something similar as a quirk

madbobmcjim, 2022-10-24:
The actuator moves and then moves back, so it's not ideal as it's using more battery than it should.

One thing to note, I had in installed them over the summer and just reinstalled one to see if things were any different, and set the temperature via Z2M instead of HA, and that persisted for a few days, so more experimentation needed I think.

10100011, 2022-12-08:
FWIW, I've put together a hack which seems effective in preventing the 20°C issue recurring.

In short:

1. Ensure that the TRV is paired to Z2M (or presumably ZHA) correctly, with multiple bindings reported
2. Disable the TRV in HA at around the 2-hour mark after pairing; re-enabling it some time afterwards.

This has so far held for five days, before I reset the TRV to continue experimentation, at which point it recurred. The root cause, however, is still unclear.

Longer explanation here: [https://community.home-assistant.io/t/drayton-wiser-schneider-electric-itrv-issues/268728/48?u=10100011](https://community.home-assistant.io/t/drayton-wiser-schneider-electric-itrv-issues/268728/48?u=10100011)

brokx, 2023-12-19:
This is not fixed

10100011, 2023-12-29:
> This is not fixed

This will almost certainly never be fixed. It's a Drayton / Schneider bug-or-feature, not a problem with Z2M.

To stop it happening, get a Wiser Hub: https://community.home-assistant.io/t/drayton-wiser-schneider-electric-itrv-issues/268728/62