Koenkk/zigbee2mqtt #11760: Problem connecting XBee 3 to coordinator CC2652P

taojoue, 2022-03-08:
### What happened?

[](https://user-images.githubusercontent.com/49972956/157068490-1f85a6ba-b740-41fd-80c0-9e49c672524d.png)[](https://user-images.githubusercontent.com/49972956/157067852-cad0a89c-6934-4ac3-ad32-85f81829128a.png)[](https://user-images.githubusercontent.com/49972956/157068318-ba3411f9-a3fd-42e2-8482-6788038126e0.png)Hello, I've seen this issue brought up before, but I could not resolve it for myself, so I'm asking here if someone has an idea !

I have a XBee 3 module as a router configured via the XCTU software :

ID : PAN ID --> 0
SC : Scan Channels --> B
ZC : ZigBee Stack Profile --> 12
AP : API Enable --> 1
AO : API Output Mode --> 3
EE : Encryption Enable --> 1
EO : Encryption Options --> 2
On my zigbee2mqtt logs (attached as "log.txt") I can see the device announcing itself and try to be interviewed by the coordinator, but it fails every time and I can't find why.

Here are screenshots of the logs and of the frontend :
The logs :
[image]

The frontend :
[image]

Device status front end :
[image]

I think @Koenkk you would easily see what's wrong and why it's not working, but I can't figure it out.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8206851/log.txt)

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210708

### Adapter

USB Key CC2652P 

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8206894/log.txt)

taojoue, 2022-03-15:
Hello @Koenkk,
I tried again today and was not able to make it work... the same problem happened again.

Could you please take a look at the issue ?

Thank you very much !

taojoue, 2022-03-18:
Hello @Koenkk,

Thank you very much for your answer!

I've added the herdsman debug log to see what happened, 
[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/8306273/logs.txt)
but still couldn't figure it out...

They are attached to this comment, if you'd like to take a look, this would be very appreciated!

Thank you!

[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/8306276/logs.txt)

Koenkk, 2022-03-18:
The device does not respond to the "active endpoints request", probably you need to update the XBee 3 firmware to respond to such requests. (0x0005 https://www.digi.com/resources/documentation/Digidocs/90002002/Reference/r_send_ZDO_cmds.htm?TocPath=API%20Operation%7C_____4)

taojoue, 2022-03-21:
Hello @Koenkk, 
Thank you very much for your help, I've been able to make the XBee 3 connect to the coordinator !

closed 2022-03-21 by taojoue