Koenkk/zigbee2mqtt #11761: [New device support]: Aubess Tuya Smart Plug (ZT2S)

KrX3D, 2022-03-08:
### Link

https://www.aliexpress.com/item/1005003749369777.html?spm=a2g0o.order_list.0.0.21ef180229sYcF

### Database entry

/

### Comments

Hello,

i got a zigbee smart plug which i could not connect. i tried the dev branch but still no luck. it shows up like this:

[image]

my version (switched back from dev branch) :

[image]

i tried to create an external converter but im not sure how to do it right even with the how to.

i opened the device up and the chip says ZT2S

[image]
[image]
[image]
[image]
[image]
[image]

hopefully someonce can help me.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'ZT2S',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_c88teujp'
        },
    ],
    model: 'ZT2S-Zigbee',
    vendor: 'Aubess',
    description: 'EU Plug',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

KrX3D, 2022-03-09:
ok nevermind, i got some problems with connecting devices to my network. now it works.

Zigbee Modell
TS011F
Zigbee Hersteller
_TZ3000_typdpbpg

closed 2022-03-09 by KrX3D

joggs, 2023-02-20:
So you still have to use the converter to get it working? or did it work without converter when you fixed the joining for your network?