Koenkk/zigbee2mqtt #11762: [New device support]: Niko Connected Single switch

barthelemyguillaume, 2022-03-08:
### Link

https://www.niko.eu/en/article/552-72111

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0xcc86ecfffee62ac5","nwkAddr":4907,"manufId":4703,"manufName":"NIKO NV","powerSource":"Mains (single phase)","modelId":"Single connectable switch,10A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2821,64512,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Single connectable switch,10A","manufacturerName":"NIKO NV","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20210224","swBuildId":"313-687-00"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20210224","swBuildId":"313-687-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1646470779190,"defaultSendRequestWhen":"immediate"}

### Comments

71 / 5 000
Résultats de traduction
I tried the method of adding a new product but without success...

sjorge, 2022-03-13:
Any idea where those can be bought? I might pick one up and try and add support if not too expensive and easy to get.

barthelemyguillaume, 2022-03-13:
https://www.dmlights.be/fr/niko_home_control_interrupteur_simple_connecte__zigbee~1CUUT

barthelemyguillaume, 2022-03-13:
Thank you for your help ;)

sjorge, 2022-03-17:
Mine should arrive somewhere next week, no promises but given the Niko Plug was pretty easy to add, I hope this will be too.

Not sure if I will have lots of time next weekend, but the weekend after that is pretty free.

sjorge, 2022-03-21:
Sadly the parcel didn't arrive on friday as planned, and now the tracker shows it in Austria :|

I'm guessing it will take a while before it gets here, annoying as it was shipped from the same country I am in.

barthelemyguillaume, 2022-03-21:
Really weird...

I think we are in the same country...but I am further south, in Wallonia ;)

sjorge, 2022-03-21:
Yeah, I'm near Gent and it went to Puurs (closest depo IIRC) -> Neuenstein (DE) -> Vienna (AT)

sjorge, 2022-03-22:
Parcel arrived, looks I ordered the wrong one without zigbee -_-

barthelemyguillaume, 2022-03-25:
Ok,

I create a file named NIKO55272111.js in the same folder than configuration.yaml using the File editor add-on of Home Assistant.

I've 3 warnings on line 18, 19 and 21.

_Line 18_
`Expected '(' instead saw '{'`

_Line 19_
```
Expected an identifier and instead saw 'const' (a reserved word).
Expected ')' to match '{' from line 18 and instead saw 'endpoint'.
Expected an identifier and instead saw "=".
Expected '}' to match '{' from line 11 and instead saw 'device'.
Expected ']' to match '[' from line 10 and instead saw ';'.
Missing semicolon
```

_Line 20_
`Missing semicolon`

_Line 21_
```
Missing semicolon
Unrecoverable syntax error. (72% scanned).
```

Then I guess I have to modify Zigbee2MQTT configuration like this 
```
external_converters:
  - NIKO55272111.js
```

Correct ?

sjorge, 2022-03-25:
More or less, you drop the .js file in the same folder where configure.yaml is and indeed add the external_coverters bit to the config.

I'm not getting the syntax errors though, perhaps double check that some symbols didn't get messed up while copying. It likes to mess with ', ", ... depending on the layout for example.

barthelemyguillaume, 2022-03-27:
It seems to work! Thank you.

If you can do the finetuning, it will be great!

sjorge, 2022-03-27:
Excellent, we’re you able to test the power_on_behavior?

sjorge, 2022-03-29:
Mine arrived, it doesn't implement much at all, unfortunately.

No power_on_behavior, no unbind + bind to a set of bulbs ... kind of disappointing. But looks like nothing else to add.

sjorge, 2022-03-29:
Just to be complete, there are 2 additional clusters on the endpoints that control the switch(es)

- 0xb05 (diagnostics)
- 0xfc00 (of interest)
- 0xfc01 (of interest)

0xb05 is the diagnostics cluster so we can ignore that one.

sjorge, 2022-03-30:
I figured out how to:
- decouple the device
- parse the 3 actions emitted when decoupled

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4061

closed 2022-05-08 by github-actions[bot]