Koenkk/zigbee2mqtt #11763: [Feature request]: Add "reason" field to sensor json message

qqqlab, 2022-03-08:
### Is your feature request related to a problem? Please describe

Currently a mqtt message for a door sensor looks like:

{"battery":100,"battery_low":false,"contact":false,"linkquality":33,"tamper":false,"voltage":3200}

The reason why this message was published not clear. Did the door sensor state changed? Or is it a periodic message without a change in sensor state? Or another reason?

### Describe the solution you'd like

Add a json field "reason", with values "sensor", "periodic", ...

{**"reason":"sensor",**"battery":100,"battery_low":false,"contact":false,"linkquality":33,"tamper":false,"voltage":3200}

From this message it is clear that the door state changed, and that it is not a periodic message without a state change.

### Describe alternatives you've considered

The reason can be inferred from the published log messages, but involves parsing text fields and recombining the log message with the sensor message. Not a very clean solution.

### Additional context

Note: In the Shelly mqtt implementation the "reason" field is an array, so multiple reasons can be given for a message. However, I have not seen a message yet where more than one reason was given.

Impact123, 2022-04-19:
Can this be reopened? This would be very useful and already exists with `StateChangeReason`. In my case I store sensor data from a TRV but I only want to persist new data when it actually came from a periodic device report, not in response to a command or similar. I can't easily differentiate what triggered the message right now. It's just one example where this could be used for.

Maybe there could be an option like `include_debug_information` similar to how `include_device_information` works. It would attach the `type`, as in `commandDataResponse`, `readResponse`, `attributeReport`, `stateChangeReason` as in `lastSeenChanged`, `publishDebounce` and the `cluster` as in `genBasic`, `haElectricalMeasurement` to the message.

The message would look similar to this when the option is enabled
```json
{
    "linkquality":143,
    "current_heating_setpoint": "19.0",
    "debug":{
        "type": "commandDataResponse",
        "stateChangeReason": "publishDebounce",
        "cluster": "genBasic"
    }
}
```

qqqlab, 2022-10-04:
Fixed this by adding these two lines
```
payload.type = data.type;
payload.cluster = data.cluster;
```
before
https://github.com/Koenkk/zigbee2mqtt/blob/03ba647dc6b5f299f8f3ab441712999fcb3a253e/lib/extension/receive.ts#L155

This is probably not the most elegant way to do this, but it works for me.

Impact123, 2022-11-05:
Unstale.

Impact123, 2022-12-06:
Unstale.

Impact123, 2023-01-07:
Unstale.

Impact123, 2023-02-07:
Unstale.

Impact123, 2023-03-11:
Unstale.

junosuarez, 2023-03-12:
@Koenkk would you accept a patch adding a setting for "include_state_change_reason" alongside `include_device_information`?

Koenkk, 2023-03-13:
@junosuarez currently there is only 1 thumbs up on the OP therefore there seems to be little interset in such feature. As an alternative you can make an extension which adds such property.

closed 2023-04-20 by github-actions[bot]