Koenkk/zigbee2mqtt #11772: Z2M entities becomes UNAVAILABLE untill Z2M restart

adizag, 2022-03-09:
### What happened?

once in awhile (about 2 weeks) the Z2M entities suddenly become unavailable to HA.
Z2M frontend is working and showing all products, allowing to control it.
HA/MQTT restart will not help, only restarting Z2M.

I can't see problems in the logs...

Any idea?

MQTT logs after restart (including successful connection from 10.0.0.102 (z2m server)):
     
```
1646826379: New connection from 10.0.0.165:61144 on port 1883.

1646826379: New client connected from 10.0.0.165:61144 as DVES_43A1AB (p2, c1, k30, u'adizag').

1646826379: New connection from 10.0.0.118:50699 on port 1883.

1646826379: New client connected from 10.0.0.118:50699 as DVES_44EF12 (p2, c1, k30, u'DVES_USER').

1646826382: New connection from 10.0.0.251:53472 on port 1883.

1646826382: New client connected from 10.0.0.251:53472 as nodered_10ddd9ee89e7d223 (p2, c1, k60).

1646826385: New connection from 10.0.0.251:42845 on port 1883.

1646826385: New client connected from 10.0.0.251:42845 as 0fAmcrDMa7jpdPLgtfjzHg (p2, c1, k60, u'adizag').

1646827100: mosquitto version 2.0.14 terminating

2022-03-09T11:58:21: mosquitto version 2.0.14 starting

2022-03-09T11:58:21: Config loaded from /mosquitto/config/mosquitto.conf.

2022-03-09T11:58:21: Opening ipv4 listen socket on port 1883.

2022-03-09T11:58:21: Opening ipv6 listen socket on port 1883.

2022-03-09T11:58:21: mosquitto version 2.0.14 running

2022-03-09T11:58:21: New connection from 10.0.0.149:55892 on port 1883.

2022-03-09T11:58:21: New client connected from 10.0.0.149:55892 as DVES_BCA302 (p2, c1, k30, u'adizag').

2022-03-09T11:58:21: New connection from 10.0.0.148:64467 on port 1883.

2022-03-09T11:58:21: New client connected from 10.0.0.148:64467 as DVES_BC4BCB (p2, c1, k30, u'adizag').

2022-03-09T11:58:21: New connection from 10.0.0.153:49521 on port 1883.

2022-03-09T11:58:21: New client connected from 10.0.0.153:49521 as DVES_BCA385 (p2, c1, k30, u'adizag').

2022-03-09T11:58:21: New connection from 10.0.0.143:51789 on port 1883.

2022-03-09T11:58:21: New client connected from 10.0.0.143:51789 as DVES_BC4C20 (p2, c1, k30, u'adizag').

2022-03-09T11:58:21: New connection from 10.0.0.161:49867 on port 1883.

2022-03-09T11:58:21: New client connected from 10.0.0.161:49867 as DVES_685F3E (p2, c1, k30, u'adizag').

2022-03-09T11:58:22: New connection from 10.0.0.251:54255 on port 1883.

2022-03-09T11:58:22: New client connected from 10.0.0.251:54255 as 0fAmcrDMa7jpdPLgtfjzHg (p2, c1, k60, u'adizag').

2022-03-09T11:58:22: New connection from 10.0.0.102:55574 on port 1883.

2022-03-09T11:58:22: New client connected from 10.0.0.102:55574 as mqttjs_fe311e99 (p2, c1, k60).

2022-03-09T11:58:31: New connection from 10.0.0.150:59727 on port 1883.
[... 143 lines cut ...]
```

### What did you expect to happen?

not going offline in HA

### How to reproduce it (minimal and precise)

don't know...

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20201127

### Adapter

zStack12

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8214531/log.txt)

budimanjojo, 2022-03-11:
I have this same issue, happened three times now.. Restarting z2m immediately solved the problem too. Are u connecting to mosquitto without authentication too @adizag?

adizag, 2022-03-11:
Yes, why would that matter?

____________________________________________
"יש המטביעים את צרותיהם - ויש המלמדים אותן לשחות..."
"אין מקום שהוא רחוק מדי"

budimanjojo, 2022-03-11:
I was speculating it's the because of `allow_anonymous true` in mosquitto, before I found the related issue. Seems like it's a bug of HASS

adizag, 2022-03-12:
in my case, It just happens, not related to restarting any of the containers.
and it is strange for me that it is HASS problem since only restarting Z2M solves it...

budimanjojo, 2022-03-13:
I have found what causes the problem and confirmed it fixes mine. You need to persists your mosquitto database, in `mosquitto.conf`:
```
persistence true
persistence_location /path/to/a/dir/mounted/with/volume
```
It's happening because mosquitto doesn't know the last state of the entity when it started, even if you have retain state enabled. If you're using docker/kubernetes, make sure that path is mounted to a volume too.

adizag, 2022-03-13:
Added as well. Let's track it...
tnx!

____________________________________________
"יש המטביעים את צרותיהם - ויש המלמדים אותן לשחות..."
"אין מקום שהוא רחוק מדי"

closed 2022-04-20 by github-actions[bot]