Koenkk/zigbee2mqtt #11775: Missing actions from IKEA E1524/E1810 remote control

pavvel11, 2022-03-09:
### What happened?

Some actions are not available when the IKEA E1524/E1810 remote control (firmware v2.3.080) is bound to some device.
When LevelCtrl and OnOff clusters are bound, following actions are NOT visible in logs: brightness_down_click, brightness_down_hold, brightness_down_release, brightness_up_click, brightness_up_hold, brightness_up_release and toggle_hold.

I think that this also causes that in Home Assistant status of entities is not being updated. When I switch on/off the light using the remote control the status in HA remains unchanged.

When the remote control is bound only with the Coordinator, and no other device, I can see all actions in logs.
Also, it seems that this issue does not appear for IKEA E2001/E2002 remote control. All actions are properly received regardless it is bound or not.

### What did you expect to happen?

All [actions](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#action-enum) are correctly captured and available.

### How to reproduce it (minimal and precise)

Pair TRADFRI remote control E1524/E1810 with firmware v2.3.080
Bind LevelCtrl and OnOff clusters to a device (eg. IKEA power supply).
Go to logs and push brightness up/down  buttons.

### Zigbee2MQTT version

1.24.0 (hassio-zigbee2mqtt v1.24.0-1)

### Adapter firmware version

 0x26580700

### Adapter

ConBee2

Koenkk, 2022-03-10:
If you bind the remote control to another device (and unbind it from the coordinator), no events will be sent anymore to the coordinator so z2m won't see these actions anymore.

pavvel11, 2022-03-10:
When I bound remote control with the coordinator, all actions are visible in logs. 
But status is still not synchronized with HA entities.
@Koenkk any idea what can be wrong?