Koenkk/zigbee2mqtt #11776: Ikea bulb stopped working

tcheronneau, 2022-03-09:
### What happened?

I own a some Ikea bulbs (different models) + different model of switches
I also have other bulbs (xiaomi mostly). 
And I also have zigbee sensors (mostly xiaomi again). 

All of the sudden (no action at all on my side) all bulbs stopped working. 
All bulbs are still ok. 
Ikea switches are still ok and other sensors as well. 

I tried restart of everything (mqtt / zigbee2mqtt / hass) 
Also tried upgrade of zigbee2mqtt (I was on version 1.21.2). 
I use CC2531 so I also tried to remove from usb. 
I even tried to reflash the firmware. 
All attempt did pretty much nothing. 

### What did you expect to happen?

Business as usual everything working. 

### How to reproduce it (minimal and precise)

Use ikea bulbs (I mostly have  TRADFRI bulb E27 WS opal 1000lm). 
While using already synced device error is : 
Data request failed with error: 'No network route' (205)
If you try to resync the device it managed to add it to coordinator but then I have another error when trying to control it : 
Data request failed with error: 'undefined' (25)

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211115

### Adapter

CC2531

### Debug log

Example after adding back device : 
```
Zigbee2MQTT:debug 2022-03-09 22:04:38: Received MQTT message on 'zigbee2mqtt/0x2c1165fffe732a1d/set' with data '{"state": "ON"}'
Zigbee2MQTT:debug 2022-03-09 22:04:38: Publishing 'set' 'state' to '0x2c1165fffe732a1d'
Zigbee2MQTT:error 2022-03-09 22:04:39: Publish 'set' 'state' to '0x2c1165fffe732a1d' failed: 'Error: Command 0x2c1165fffe732a1d/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
Zigbee2MQTT:debug 2022-03-09 22:04:39: Error: Command 0x2c1165fffe732a1d/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:info  2022-03-09 22:04:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x2c1165fffe732a1d' failed: 'Error: Command 0x2c1165fffe732a1d/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'undefined' (25))'","meta":{"friendly_name":"0x2c1165fffe732a1d"},"type":"zigbee_publish_error"}'

```

Example of existing device : 
```
Zigbee2MQTT:debug 2022-03-09 22:05:24: Received MQTT message on 'zigbee2mqtt/lampe_salle_a_manger/set' with data '{"state": "ON"}'
Zigbee2MQTT:debug 2022-03-09 22:05:24: Publishing 'set' 'state' to 'lampe_salle_a_manger'
Zigbee2MQTT:error 2022-03-09 22:05:45: Publish 'set' 'state' to 'lampe_salle_a_manger' failed: 'Error: Command 0x14b457fffe77f26e/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:debug 2022-03-09 22:05:45: Error: Command 0x14b457fffe77f26e/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:info  2022-03-09 22:05:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'lampe_salle_a_manger' failed: 'Error: Command 0x14b457fffe77f26e/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"lampe_salle_a_manger"},"type":"zigbee_publish_error"}'

```

T-IOOO, 2022-03-10:
I would recommend to update your Zigbee2mqtt!

Try to force remove a bulb that is not working, restart zigbee and then re-pair the bulb.

Chippi, 2022-03-22:
Same problem here, things just stopped working.

tcheronneau, 2022-03-22:
I tried everything on my side and nothing worked. 
I changed my zigbee adapter to a sonoff (since the other was deprecated anyway). 
I had to repair everything but I got almost everything back except this model which is not able to pair : https://www.zigbee2mqtt.io/devices/LED1732G11.html (I don't have any logs anymore since I decided to use a spare device I had to replace it).

T-IOOO, 2022-03-22:
I have experienced this maybe a year ago. Think it was the same kind of bulb too. I had to do the pairing procedure many, many times before I got it working, so don’t give up. Another thing you can try is to put the bulb in another lamp socket.. eventually you will be able to pair it

Chippi, 2022-03-24:
Update: It's working again. It didn't work for an entire day and I got the same errors as @tcheronneau. Then suddenly everything started working again /shrug.

closed 2022-05-02 by github-actions[bot]

Chippi, 2022-05-13:
This solution solved my random disconnects...
https://github.com/Koenkk/zigbee2mqtt/discussions/12467