Koenkk/zigbee2mqtt #11777: SEA802-Zigbee Local Temperatur calibration error

gomble, 2022-03-09:
### What happened?

I am getting following error:

Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:191
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 23:20:05 (16 occurrences)
Last logged: 23:45:21

Invalid value for number.thermostat_esszimmer_local_temperature_calibration: -32.0 (range -30.0 - 30.0)
Invalid value for number.thermostat_esszimmer_local_temperature_calibration: -33.0 (range -30.0 - 30.0)

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

Koenkk, 2022-03-10:
Try setting the local_temperature_calibration once via the frontend (so it comes in the -30 till 30 range)

gomble, 2022-03-10:
> Try setting the local_temperature_calibration once via the frontend (so it comes in the -30 till 30 range)

the problem is that i can only go -6 to 6 celsius if i go more then -6 or 6 then the temperature does not change.
But the slider on the frontend goes from -30 to 30 
Also some automatisations in homeassistant do not work to sync the trv with temperature sensor

Koenkk, 2022-03-11:
Changed the limits to -6 - 6 now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gomble, 2022-03-11:
Okay one more info the calibration does not work for numbers with decimal
This does not change the temperature:
[image]
But this is working:
[image]

Possible numbers are:
-6,-5,-4,-3,-2,-1,0,1,2,3,4,5,6

I will test it in the dev branch

Koenkk, 2022-03-11:
@gomble also updated the step size.

gomble, 2022-03-13:
Hey that worked 
[image]

gomble, 2022-03-13:
when this will be merged to a release ?

closed 2022-03-14 by Koenkk

ItsMisterP, 2022-09-25:
Hey @gomble do you investigate, why this aint work with decimals?
Regards
Philip

gomble, 2022-09-25:
> Hey @gomble do you investigate, why this aint work with decimals?
> 
> Regards
> 
> Philip

Yes it was fixed in the next release
But you can only calibrate from -10 to 10

ewolf29, 2022-11-14:
Are we certain that decimal step sizes are not supported by the device? Decimal number calibration does actually not make any sense in this case. 

I would volunteer to test it if it gets changed on the dev branch.