Koenkk/zigbee2mqtt #11778: TuYa ZM-35H-Q battery property

vmishurov, 2022-03-10:
### What happened?

Battery level always NULL.
Log:
 2022-03-09 23:22:29: MQTT publish: topic 'zigbee2mqtt/0xa4c138837b2aafea', payload '{"battery":null,"battery_low":false,"keep_time":30,"linkquality":43,"occupancy":false,"sensitivity":"high","tamper":false}'

I've tried uninstalling and re-pairing the device several times, but it doesn't work.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

SONOFF ZB Dongle

pavelrusiyan, 2022-04-05:
We just didn't get lucky.
I have:
Support status - Supported
But the state is not visible.
External converters didn't help.

pavelrusiyan, 2022-04-08:
Answer supporting:

Minims, 2022-04-19:
@pavelrusiyan did you create an external converted ? 

I have juste received my _TZ3000_msl6wxk9.

 - battery is not reported 
 - I am unable to configure sensitivity and keep_time.

Occupancy detection is OK.

vmishurov, 2022-05-21:
still wait solution...

Minims, 2022-05-21:
Same for me. Gonna try to find tuya ids in the next days

Minims, 2022-05-23:
> Same for me. Gonna try to find tuya ids in the next days

So there is 4 Data Point for this model : 

 - Battery Level :   "code": "4"
 - PIR Sensivity :     "code": "9"
 - Keep Time :     "code": "10"
 - PIR State :     "code": "1"

Minims, 2022-05-23:
Where can I fixed these datapoints for _TZ3000_msl6wxk9 ? Thanks

Koenkk, 2022-05-24:
Could you check if battery is reported with the following external converter

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    ZM35HQ_battery: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'ZM35HQ');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            if (dp === 4) return {battery: value};
            else {
                meta.logger.warn(`zigbee-herdsman-converters:ZM35HQ: NOT RECOGNIZED DP #${dp} with data ${JSON.stringify(dpValue)}`);
            }
        },
    }
}

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_msl6wxk9'}],
    model: 'ZM-35H-Q',
    vendor: 'TuYa',
    description: 'Motion Sensor CUSTOM',
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ZM35HQ_attr, fzLocal.ZM35HQ_battery],
    toZigbee: [tz.ZM35HQ_attr],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery(),
        exposes.enum('sensitivity', ea.ALL, ['low', 'medium', 'high']).withDescription('PIR sensor sensitivity'),
        exposes.enum('keep_time', ea.ALL, [30, 60, 120]).withDescription('PIR keep time in seconds'),
    ],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if battery is reported (wait max 24 hours)

vmishurov, 2022-05-27:
Unfortunately external converter didn't fix the problem.

[image]
[image]

vmishurov, 2022-05-27:
Question on this line of code:
`
meta.logger.warn('zigbee-herdsman-converters:ZM35HQ: NOT RECOGNIZED DP #${dp} with data ${JSON.stringify(dpValue)}`';

Should there be messages in the z2m log?

Added a line here:
`
convert: (model, msg, publish, options, meta) => {          
--->     meta.logger.warn("ZM35HQ converter");
            const dpValue = tuya.firstDpValue(msg, meta, 'ZM35HQ');
`

But I don't see these messages in the log.

Minims, 2022-05-27:
Same for me. I have double check, and it's the code 4 in Tuya api : 

```
{startRowId: "", pageNo: 1, pageSize: 10, code: "4", type: "EVENT_TYPE_DATAPOINT_PUBLISH_API",…}
code: "4"
deviceId: "****"
endTime: 1653660049540
pageNo: 1
pageSize: 10
pageStartRow: ""
projectCode: "***"
region: "EU"
sourceId: "***"
sourceType: "4"
startRowId: ""
startTime: 1653487249540
type: "EVENT_TYPE_DATAPOINT_PUBLISH_API"
```

Koenkk, 2022-05-27:
@vmishurov interesting, so it seems the device does not report anything via the tuya cluster yet. I've updated https://github.com/Koenkk/zigbee2mqtt/issues/11778#issuecomment-1136255686 (added the `configure` section). Could update your external converter with this, repair the device and see if the device starts reporting via the TuYa cluster?

Minims, 2022-05-27:
> @vmishurov interesting, so it seems the device does not report anything via the tuya cluster yet. I've updated [#11778 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/11778#issuecomment-1136255686) (added the `configure` section). Could update your external converter with this, repair the device and see if the device starts reporting via the TuYa cluster?

It works ! Thanks.

vmishurov, 2022-05-28:
Yea!
Problem solved! Thanks.

Koenkk, 2022-05-28:
Pushed the fix, thanks for confirming!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-28 by Koenkk