Koenkk/zigbee2mqtt #11797: Danfoss Ally - 'window_open_internal' not working when using 'radiator_covered'

tatec22, 2022-03-11:
### What happened?

Hi everyone,

the last couple of days I tried to figure out why my radiators are still heating when I open a window.

I narrowed it down that there might be an issue with 'radiator_covered' and 'window_open_feature/internal/external'.

Any help is appreciated. 

In the current state I can not use my setup properly. 

My radiators still remain heating after opening a window, which kinda destroys the purpose of my whole setup of sensors.

If you need any more information/logs etc. please let me know.

Kind regards
tatec22

### What did you expect to happen?

I expected that there is no difference in the window_open_feature/internal/external when using radiator_covered.

### How to reproduce it (minimal and precise)

- 'radiator_covered' is set to True
- Set 'window_open_external' to True
- Variable is set to True but 'window_open_internal' remains on 'closed'
- Radiators are still heating (load_estimate stays the same)

Otherwise:
- 'radiator_covered' is set to False
- Set 'window_open_external' to True
- Variable is set to True and 'window_open_internal' responds by changing to 'open'
- Radiators are not heating (load_estimate is set to -8000)

### Zigbee2MQTT version

1.24.0-dev commit: 52d151e

### Adapter firmware version

0x26720700

### Adapter

ConBee2

closed 2022-04-19 by github-actions[bot]

codingcatgirl, 2022-04-30:
I have this issue too and have worked around it by just sending a very low temperature when the windows are open. Of course that's not ideal but… well… no idea why it's not working. Wondering if it might be a firmware bug.

tatec22, 2022-05-08:
Well, that’s good to know I’m not alone with this issue.
I did the same thing and sent a very low temperature to the device when I opened a window.
But after closing the window and resetting the temperature back, the Ally opened its valve to nearly 100%.
Don’t know why this happened.
So this didn’t work for me either.

passibe15, 2022-09-30:
Hi,

I'm having the same issue.
It seems to be a behavior introduced with one of the later firmware versions, I have one thermostat that is on version 01.12.0008 01.12, there `window_open_internal` responds to setting `window_open_external` to true.

However, on another thermostat that I updated to 01.18.0008 01.18 setting `window_open_external` to true does not change `window_open_internal`, it just stays closed.

Is there any further progress on this?

Cheers

rsporsche, 2022-12-08:
Looks like a firmware bug to me, I'm going to try unsetting radiator_covered at the same time because I wouldn't know what to set the setpoint back to when the window closed if I just lowered the setpoint. Maybe there's a way in the automations to 'save' the original setpoint in order to set it back, but that would be more complicated and there's still the possibility that a scheduled change will change the setpoint again while in 'window open' mode.