Koenkk/zigbee2mqtt #11798: Impossible to start the process

chekdo, 2022-03-11:
### What happened?

Hello,
I try to start Zigbee2Mqtt but it doesn't work. All was ok during the installation. Here is the code when I try to start the processus.
npm start

> zigbee2mqtt@1.24.0 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info 2022-03-11 07:27:38: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-03-11.07-27-36' filename: log.txt
Zigbee2MQTT:info 2022-03-11 07:27:38: Starting Zigbee2MQTT version 1.24.0 (commit #7a2ddf24)
Zigbee2MQTT:info 2022-03-11 07:27:38: Starting zigbee-herdsman (0.14.20)
Zigbee2MQTT:error 2022-03-11 07:28:43: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-03-11 07:28:43: Failed to start zigbee
Zigbee2MQTT:error 2022-03-11 07:28:43: Check [https://www.zigbee2mqtt.io/guide/instal ... start.html](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html) for possible solutions
Zigbee2MQTT:error 2022-03-11 07:28:43: Exiting...
Zigbee2MQTT:error 2022-03-11 07:28:44: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
at ZnpAdapterManager.beginCommissioning (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:100:27)
at start (/opt/zigbee2mqtt/index.js:101:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.24.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.24.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! /home/zeus/.npm/_logs/2022-03-11T06_28_44_169Z-debug.log

### What did you expect to happen?

A success opening :)

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

Z-Stack_Home_3.x.0_20211217

### Adapter

sonoff zigbee 3.0 usb dongle

### Debug log

info  2022-03-11 07:57:57: Logging to console and directory: '/opt/zigbee2mqtt/$
info  2022-03-11 07:57:57: Starting Zigbee2MQTT version 1.24.0 (commit #7a2ddf2$
info  2022-03-11 07:57:57: Starting zigbee-herdsman (0.14.20)
error 2022-03-11 07:59:03: Error while starting zigbee-herdsman
error 2022-03-11 07:59:03: Failed to start zigbee
error 2022-03-11 07:59:03: Check https://www.zigbee2mqtt.io/guide/installation/$
error 2022-03-11 07:59:03: Exiting...
error 2022-03-11 07:59:03: Error: network commissioning timed out - most likely$
    at ZnpAdapterManager.beginCommissioning (/opt/zigbee2mqtt/node_modules/zigb$
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/s$
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/cont$
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:100:27)
    at start (/opt/zigbee2mqtt/index.js:101:5)

T-IOOO, 2022-03-25:
i had similar issue a while ago with raspberry pi zero w. This happened when i updated node to v16. If you have node v16, try downgrade to v14. I ran the ”switch to dev branch” after the downgrade and it should work

WP-0461, 2022-03-31:
i have a similar problem.
after a new installation in Raspberry of Debian and IOB with Zigbee Adapter i get the Message "you need to save an run before Paring". I saved the dialog many times and the Communication port ist correct.
But i find no way to setup a new IOB on Raspberry with Zigbee.
Therfore, i fear that my still running productive system will no longer work with zigbee after the next update.
does someone have any idea about this isue ?

Will

closed 2022-05-09 by github-actions[bot]