Koenkk/zigbee2mqtt #11799: Lidl/silvercrest  EU smartplug no button state change

RobvanSuilen, 2022-03-11:
### What happened?

I have an HG06337 smart plug from Lidl/silvercrest.

I'm using HomeAssistant, Mosuitto, and zigbee2Mqtt. and the cc2531 stick.
All; connected divices work normal. Except this plug

Problem: Although i am able to switch the plug on and off with its hardware button, no state change is detected in HA.  Only when i click the status update button in Home Assistant (blue square with white arrows). the stutus changes to the correct status.
When i switch via HA, state change is immediately detected .

### Zigbee2MQTT version

    1.24.0

### Adapter firmware version

20190608

### Adapter

cc2531 with zStack12

21stce, 2022-03-12:
Wat makes you think it is supposed to work in the first place? Does this work on the app provided by Lidl?

MattWestb, 2022-03-13:
With LIDL ZBGW is reporting fast and nice its status to the APP and also back then changing the state in the APP.

RobvanSuilen, 2022-03-13:
Thanks MattWestb, i was not able to test it with the original gateway cause i dont have it.

MattWestb, 2022-03-13:
No problem i was doing it for you so knowing that the device can sending status updates to the coordinator = no device problem and somthing is missing for getting it working OK in Z2M.

MattWestb, 2022-03-13:
@21stce In Zigbee is the standard all lights (and plugs) shall reporting its status changes to the coordinator and controlling devices (remotes) is only sending attributes to the coordinator (battery and power reports) and commands to the bounded light group.
Some lights (not tuya ones) like Philips HUE and some more is not doing it as standard and need being pulled that is making bad impact on the system performance.

RobvanSuilen, 2022-03-13:
I have looked at the mqtt logs in HA i do not see any messages when the button is pressed.

MattWestb, 2022-03-13:
Very likely then Z2M need pulling the state from the plug for sending the new state to HA then the device is is not reporting it or fixing the reporting that is not working in Z2M dor this device (its working OK in ZHA out of the box).

RobvanSuilen, 2022-03-13:
I meant, i do see mqtt messages when the plugs are controlled by HA and also when i switch the hardware buttons and then update the device status by olpressing the bleu square with the hite arriws.

I do not see messages or a state change when only pressing the hardware buttons on the smartplug itself.

RobvanSuilen, 2022-03-22:
All of a sudden these plugs did start to sent status info at a button press.I have no idea why this now does work, but it does!

closed 2022-03-22 by RobvanSuilen