Koenkk/zigbee2mqtt #11800: One of two _TZ3000_amdymr7l not reporting current Power

nohn, 2022-03-11:
### What happened?

I have two _TZ3000_amdymr7l included in latest-dev (see #11703). One of them is reporting all metrics, one of them is missing power.

### What did you expect to happen?

Both report power

### Zigbee2MQTT version

1.24.0-dev commit: 8f98a2f

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB Stick

### Debug log

```
Zigbee2MQTT:info  2022-03-11 15:25:00: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"pz0j5-2"}'
Zigbee2MQTT:debug 2022-03-11 15:25:26: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'genOnOff', data '{"20480":1,"32768":0,"moesStartUpOnOff":1,"offWaitTime":0,"onOff":1,"onTime":0,"tuyaBacklightMode":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 15:25:26: MQTT publish: topic 'zigbee2mqtt/Badezimmer Waschbecken Plug', payload '{"child_lock":"UNLOCK","current":null,"energy":0,"indicator_mode":"off/on","linkquality":0,"power":null,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":227}'
Zigbee2MQTT:debug 2022-03-11 15:25:26: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53248":1,"53249":0,"53250":1981202,"53251":161,"53252":12026,"53253":1817,"53264":1,"switchType":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:26: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53248":1,"53249":0,"53250":1981202,"53251":161,"53252":12026,"53253":1817,"53264":1,"switchType":0}'
Zigbee2MQTT:debug 2022-03-11 15:25:29: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 15:25:29: MQTT publish: topic 'zigbee2mqtt/Badezimmer Waschbecken Plug', payload '{"child_lock":"UNLOCK","current":null,"energy":0,"indicator_mode":"off/on","linkquality":0,"power":null,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":227}'
Zigbee2MQTT:debug 2022-03-11 15:25:29: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:29: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}'
Zigbee2MQTT:debug 2022-03-11 15:25:29: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:29: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}'
Zigbee2MQTT:debug 2022-03-11 15:25:29: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:29: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}'
Zigbee2MQTT:debug 2022-03-11 15:25:30: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:30: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}'
Zigbee2MQTT:debug 2022-03-11 15:25:30: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:30: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,215,10,1,208,72,2,0,6],"type":"Buffer"}'
Zigbee2MQTT:debug 2022-03-11 15:25:30: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,216,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:30: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,216,10,2,208,72,2,0,10],"type":"Buffer"}'
Zigbee2MQTT:debug 2022-03-11 15:25:31: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'genOnOff', data '{"20480":1,"32768":0,"moesStartUpOnOff":1,"offWaitTime":0,"onOff":1,"onTime":0,"tuyaBacklightMode":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 15:25:31: MQTT publish: topic 'zigbee2mqtt/Badezimmer Waschbecken Plug', payload '{"child_lock":"UNLOCK","current":null,"energy":0,"indicator_mode":"off/on","linkquality":3,"power":null,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":227}'
Zigbee2MQTT:debug 2022-03-11 15:25:31: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53248":1,"53249":0,"53250":1981202,"53251":161,"53252":12026,"53253":1817,"53264":1,"switchType":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:31: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53248":1,"53249":0,"53250":1981202,"53251":161,"53252":12026,"53253":1817,"53264":1,"switchType":0}'
Zigbee2MQTT:debug 2022-03-11 15:25:31: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 15:25:31: MQTT publish: topic 'zigbee2mqtt/Badezimmer Waschbecken Plug', payload '{"child_lock":"UNLOCK","current":null,"energy":0,"indicator_mode":"off/on","linkquality":0,"power":null,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":227}'
Zigbee2MQTT:debug 2022-03-11 15:25:32: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 15:25:32: MQTT publish: topic 'zigbee2mqtt/Badezimmer Waschbecken Plug', payload '{"child_lock":"UNLOCK","current":null,"energy":0,"indicator_mode":"off/on","linkquality":33,"power":null,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":227}'
Zigbee2MQTT:debug 2022-03-11 15:25:44: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 15:25:44: MQTT publish: topic 'zigbee2mqtt/Badezimmer Waschbecken Plug', payload '{"child_lock":"UNLOCK","current":null,"energy":0,"indicator_mode":"off/on","linkquality":33,"power":null,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":227}'
Zigbee2MQTT:debug 2022-03-11 15:25:44: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,222,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:44: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,222,10,1,208,72,2,0,6],"type":"Buffer"}'
Zigbee2MQTT:debug 2022-03-11 15:25:44: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,222,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:44: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,222,10,1,208,72,2,0,6],"type":"Buffer"}'
Zigbee2MQTT:debug 2022-03-11 15:25:45: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,222,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:debug 2022-03-11 15:25:48: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'raw', cluster '57344', data '{"data":[8,228,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-11 15:25:48: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,228,10,1,208,72,2,0,6],"type":"Buffer"}'
Zigbee2MQTT:error 2022-03-11 15:26:58: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
Zigbee2MQTT:debug 2022-03-11 15:28:39: Saving state to file /app/data/state.json

```

nohn, 2022-03-11:
Re-Binding like in #10889 does not fix it.

nohn, 2022-03-11:
OK, now it starts to get strange: Out of nothing, the plug begins to report Watts.

Feel free to close this issue (I have no idea how to reproduce it).

However: There are two more batches of (presumable) these plugs coming in. Should I test something before / while pairing them?

nohn, 2022-03-11:
Here is the database.db entry:

```
{"id":66,"type":"Router","ieeeAddr":"0xa4c1389d9b178ece","nwkAddr":42926,"manufId":4417,"manufName":"_TZ3000_amdymr7l","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_amdymr7l","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":229,"rmsCurrent":0,"activePower":659}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":1981202,"53251":161,"53252":12026,"53253":1817,"53264":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002390c2bd","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002390c2bd","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002390c2bd","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1647013971610,"defaultSendRequestWhen":"immediate"}
```

nohn, 2022-03-11:
I tested with a toaster (that was the 659 Watts by my guess).

After that (around 17:08:00), I connected it to a PoE switch that eats between 10-20 W, depending on the load of the connected devices (one accesspoint, one 60 GHz wireless bridge). However, it took until 17:12:00 till the plug started reporting 13 W:

Inbetween, reported power even went down to 0:

```Zigbee2MQTT:debug 2022-03-11 17:11:41: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:11:41: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":48,"power":659,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:11:46: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":88}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:11:46: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0.09,"energy":0,"indicator_mode":"off/on","linkquality":48,"power":659,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:12:42: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":13}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:12:42: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0.09,"energy":0,"indicator_mode":"off/on","linkquality":51,"power":13,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:13:06: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":144}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:13:06: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0.14,"energy":0,"indicator_mode":"off/on","linkquality":48,"power":13,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:16:08: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:16:08: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":48,"power":13,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:16:08: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:16:08: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":48,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:16:13: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":160}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:16:13: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0.16,"energy":0,"indicator_mode":"off/on","linkquality":48,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:16:13: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":19}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:16:13: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0.16,"energy":0,"indicator_mode":"off/on","linkquality":48,"power":19,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:16:36: Saving state to file /app/data/state.json
Zigbee2MQTT:error 2022-03-11 17:19:24: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
Zigbee2MQTT:debug 2022-03-11 17:20:31: Received Zigbee message from 'Wohnzimmer Licht Stehleuchte Vitrine Switch', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3015,"10":9437,"3":26,"4":13224,"5":70,"6":[0,1]}}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:20:31: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Licht Stehleuchte Vitrine Switch', payload '{"action":null,"battery":100,"linkquality":48,"voltage":3015}'
Zigbee2MQTT:debug 2022-03-11 17:20:36: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:debug 2022-03-11 17:20:41: Received Zigbee message from 'Github 11800 Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":19}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:20:41: MQTT publish: topic 'zigbee2mqtt/Github 11800 Plug', payload '{"child_lock":"UNLOCK","current":0.16,"energy":0,"indicator_mode":"off/on","linkquality":48,"power":19,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:21:36: Saving state to file /app/data/state.json
```

nohn, 2022-03-11:
Firmware etc. seems to be the same for both versions:

[image]

nohn, 2022-03-11:
Those ```{"53248":1,"53249":0,"53250":1981202,"53251":161,"53252":12026,"53253":1817,"53264":1,"switchType":0}'``` do not occur anymore in the log.

nohn, 2022-03-11:
Those ```Zigbee2MQTT:error 2022-03-11 17:31:37: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}``` did not occur before I integrated the plug into Z2M, so this seems related:

```zigbee2mqtt/data/log$ find . -type f
./2022-03-03.09-08-12/log.txt
./2022-01-31.14-34-44/log.txt
./2022-02-18.18-59-11/log.txt
./2022-03-10.15-40-02/log.txt
./2022-03-11.14-58-37/log.txt
./2022-02-14.10-34-52/log.txt
./2022-01-25.15-45-02/log.txt
./2022-03-11.15-36-34/log.txt
./2022-02-02.20-22-45/log.txt
./2022-03-11.15-35-34/log.txt
zigbee2mqtt/data/log$ find . -type f | xargs grep BigInt
./2022-03-11.14-58-37/log.txt:error 2022-03-11 15:00:57: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.14-58-37/log.txt:error 2022-03-11 15:08:04: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.14-58-37/log.txt:error 2022-03-11 15:13:29: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.14-58-37/log.txt:error 2022-03-11 15:20:37: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.14-58-37/log.txt:error 2022-03-11 15:26:58: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.14-58-37/log.txt:error 2022-03-11 15:33:10: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 15:40:47: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 15:47:54: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 15:54:27: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:00:26: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:06:25: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:12:52: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:18:13: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:24:51: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:30:59: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:38:50: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:45:40: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:50:54: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 16:58:43: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 17:04:02: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 17:11:31: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 17:19:24: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 17:25:24: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
./2022-03-11.15-36-34/log.txt:error 2022-03-11 17:31:37: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
```

nohn, 2022-03-11:
At 17:43:19 I turned off a PoE Device that consumed around 5 Watt. The log after that:

```Zigbee2MQTT:debug 2022-03-11 17:43:06: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":19}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:43:06: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rack Plug', payload '{"child_lock":"UNLOCK","current":0.16,"energy":0.01,"indicator_mode":"off/on","linkquality":30,"power":19,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:info  2022-03-11 17:43:28: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffeadff76', payload '{"battery":100,"battery_low":false,"linkquality":21,"occupancy":false,"tamper":false,"voltage":3000}'
Zigbee2MQTT:info  2022-03-11 17:43:28: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Licht Stehleuchte Vitrine Switch', payload '{"action":null,"battery":100,"linkquality":48,"voltage":3015}'
Zigbee2MQTT:info  2022-03-11 17:43:29: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rack Plug', payload '{"child_lock":"UNLOCK","current":0.16,"energy":0.01,"indicator_mode":"off/on","linkquality":30,"power":19,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:44:43: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":104}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:44:43: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rack Plug', payload '{"child_lock":"UNLOCK","current":0.1,"energy":0.01,"indicator_mode":"off/on","linkquality":51,"power":19,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:error 2022-03-11 17:46:12: Exception while calling fromZigbee converter: Do not know how to serialize a BigInt}
Zigbee2MQTT:debug 2022-03-11 17:46:36: Saving state to file /app/data/state.json
Zigbee2MQTT:debug 2022-03-11 17:47:21: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1]}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:47:21: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rack Plug', payload '{"child_lock":"UNLOCK","current":0.1,"energy":0.01,"indicator_mode":"off/on","linkquality":84,"power":19,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:47:28: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:47:28: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rack Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":0.01,"indicator_mode":"off/on","linkquality":84,"power":19,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:47:28: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:47:28: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rack Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":0.01,"indicator_mode":"off/on","linkquality":87,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:47:33: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":112}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:47:33: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rack Plug', payload '{"child_lock":"UNLOCK","current":0.11,"energy":0.01,"indicator_mode":"off/on","linkquality":81,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
Zigbee2MQTT:debug 2022-03-11 17:47:33: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":12}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-11 17:47:33: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rack Plug', payload '{"child_lock":"UNLOCK","current":0.11,"energy":0.01,"indicator_mode":"off/on","linkquality":84,"power":12,"power_outage_memory":"restore","state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
```

nohn, 2022-03-11:
The other plug reacts within a few seconds to changing loads.

Koenkk, 2022-03-11:
Could you check with the converter below? First force remove the device via the frontend, restart z2m and pair it:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_amdymr7l'}],
    model: 'TS011F_plug_1',
    description: 'Smart plug (with power monitoring) CUSTOM',
    vendor: 'TuYa',
    whiteLabel: [{vendor: 'LELLKI', model: 'TS011F_plug'}, {vendor: 'NEO', model: 'NAS-WR01B'},
        {vendor: 'BlitzWolf', model: 'BW-SHP15'}],
    ota: ota.zigbeeOTA,
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory,
        fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);

        // await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        // await reporting.rmsVoltage(endpoint, {change: 5});
        // await reporting.rmsCurrent(endpoint, {change: 50});
        // await reporting.activePower(endpoint, {change: 10});
        // await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        // device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage'),
        exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
            .withDescription('Plug LED indicator mode'), e.child_lock()],
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)

nohn, 2022-03-11:
Do I need to remove both or just the "broken" one?

nohn, 2022-03-12:
When ext_converter.js is referenced in configuration.yaml, Z2M goes into a restart loop:

```
(Use `node --trace-warnings ...` to show where the warning was created)
Zigbee2MQTT:info  2022-03-12 07:03:49: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-03-12 07:03:58: Stopped zigbee-herdsman
(node:6) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
Zigbee2MQTT:info  2022-03-12 07:03:58: Stopped Zigbee2MQTT
Using '/app/data' as data directory
(node:6) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Using '/app/data' as data directory
(node:7) UnhandledPromiseRejectionWarning: ReferenceError: ota is not defined
Using '/app/data' as data directory
(Use `node --trace-warnings ...` to show where the warning was created)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:7) UnhandledPromiseRejectionWarning: ReferenceError: ota is not defined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
Using '/app/data' as data directory
(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:6) UnhandledPromiseRejectionWarning: ReferenceError: ota is not defined
Using '/app/data' as data directory
(Use `node --trace-warnings ...` to show where the warning was created)
(node:6) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:6) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:6) UnhandledPromiseRejectionWarning: ReferenceError: ota is not defined
Using '/app/data' as data directory
(Use `node --trace-warnings ...` to show where the warning was created)
(node:6) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
```

Koenkk, 2022-03-12:
Found the issue, I've updated https://github.com/Koenkk/zigbee2mqtt/issues/11800#issuecomment-1065409831

nohn, 2022-03-12:
```
Zigbee2MQTT:debug 2022-03-12 20:04:15: Clearing Home Assistant config 'homeassistant/sensor/0xa4c1384672e1a90e/update_state/config'
Zigbee2MQTT:info  2022-03-12 20:04:15: MQTT publish: topic 'homeassistant/sensor/0xa4c1384672e1a90e/update_state/config', payload 'null'
Zigbee2MQTT:debug 2022-03-12 20:04:15: Clearing Home Assistant config 'homeassistant/binary_sensor/0xa4c1384672e1a90e/update_available/config'
Zigbee2MQTT:info  2022-03-12 20:04:15: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1384672e1a90e/update_available/config', payload 'null'
Zigbee2MQTT:debug 2022-03-12 20:04:16: Received MQTT message on 'homeassistant/sensor/0xa4c1384672e1a90e/update_state/config' with data ''
Zigbee2MQTT:debug 2022-03-12 20:04:16: Received MQTT message on 'homeassistant/binary_sensor/0xa4c1384672e1a90e/update_available/config' with data ''
Zigbee2MQTT:debug 2022-03-12 20:04:52: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"vap1i-2","value":true}'
Zigbee2MQTT:info  2022-03-12 20:04:52: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-03-12 20:04:52: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"vap1i-2"}'
Zigbee2MQTT:info  2022-03-12 20:05:11: Device '0xa4c1389d9b178ece' joined
Zigbee2MQTT:info  2022-03-12 20:05:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1389d9b178ece","ieee_address":"0xa4c1389d9b178ece"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-03-12 20:05:11: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece/availability', payload 'online'
Zigbee2MQTT:info  2022-03-12 20:05:11: Starting interview of '0xa4c1389d9b178ece'
Zigbee2MQTT:info  2022-03-12 20:05:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1389d9b178ece","ieee_address":"0xa4c1389d9b178ece","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:debug 2022-03-12 20:05:12: Device '0xa4c1389d9b178ece' announced itself
Zigbee2MQTT:info  2022-03-12 20:05:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1389d9b178ece","ieee_address":"0xa4c1389d9b178ece"},"type":"device_announce"}'
Zigbee2MQTT:debug 2022-03-12 20:05:13: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS011F"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:13: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-03-12 20:05:13: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_amdymr7l"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:13: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-03-12 20:05:13: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:13: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-03-12 20:05:13: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:13: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-03-12 20:05:13: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{"appVersion":74}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:13: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-03-12 20:05:13: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:13: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-03-12 20:05:13: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:13: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-03-12 20:05:14: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:14: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-03-12 20:05:14: Received Zigbee message from '0xa4c1389d9b178ece', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-12 20:05:14: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info  2022-03-12 20:05:14: Successfully interviewed '0xa4c1389d9b178ece', device has successfully been paired
Zigbee2MQTT:info  2022-03-12 20:05:14: Device '0xa4c1389d9b178ece' is supported, identified as: TuYa Smart plug (with power monitoring) CUSTOM (TS011F_plug_1)
Zigbee2MQTT:info  2022-03-12 20:05:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart plug (with power monitoring) CUSTOM","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"b [... cut ...]
Zigbee2MQTT:debug 2022-03-12 20:08:51: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,49]}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:08:51: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":null,"energy":0.49,"indicator_mode":"off/on","linkquality":45,"power":null,"power_outage_memory":null,"state":"ON","voltage":null}'
Zigbee2MQTT:debug 2022-03-12 20:08:58: Saving state to file /app/data/state.json
```

nohn, 2022-03-12:
I tried to trigger some reaction with a vacuum cleaner => Nothing.

So I reverted the change (force removed 0xa4c1389d9b178ece, remove the change in configuration.yaml, re-added 0xa4c1389d9b178ece to Z2M).

Again, 0 Watt was reported (although 15-20 W was expected).

So I started to play around a bit by adding a vacuum cleaner to the multiplug.

Immediately, the Plug started reporting realistic power consumption. When I tuned the vacuum cleaner, the plug reacted immediately, in both directions, multiple times. After turning off the vacuum cleaner, wattage return to 17 (as expected), but returned to 0 after a short time just to return tu 18 (as expected):

```
Zigbee2MQTT:info  2022-03-12 20:55:57: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":0.7,"energy":0.51,"indicator_mode":"off/on","linkquality":45,"power":0,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:55:59: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":120}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:55:59: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":0.12,"energy":0.51,"indicator_mode":"off/on","linkquality":45,"power":0,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:56:02: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":722}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:56:02: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":0.12,"energy":0.51,"indicator_mode":"off/on","linkquality":45,"power":722,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:56:04: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":5981}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:56:04: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":5.98,"energy":0.51,"indicator_mode":"off/on","linkquality":45,"power":722,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:56:07: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":1310}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:56:07: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":5.98,"energy":0.51,"indicator_mode":"off/on","linkquality":45,"power":1310,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:56:09: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":5852}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:56:09: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":5.85,"energy":0.51,"indicator_mode":"off/on","linkquality":48,"power":1310,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:56:13: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":1078}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:56:13: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":5.85,"energy":0.51,"indicator_mode":"off/on","linkquality":45,"power":1078,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:56:15: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":5732}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:56:15: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":5.73,"energy":0.51,"indicator_mode":"off/on","linkquality":45,"power":1078,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:56:18: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":1160}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-12 20:56:18: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":5.73,"energy":0.51,"indicator_mode":"off/on","linkquality":48,"power":1160,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 20:56:20: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":4353}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:info  2022-03-12 21:05:00: MQTT publish: topic 'zigbee2mqtt/0xa4c1389d9b178ece', payload '{"child_lock":"UNLOCK","current":0.14,"energy":0.52,"indicator_mode":"off/on","linkquality":42,"power":18,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:debug 2022-03-12 21:05:08: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'genBasic', data '{"65506":53,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0
```

Koenkk, 2022-03-13:
Do you mean that you see 0 being reported by the device (`'{"activePower":0}'`) when a load is connected? If yes, then this looks like an issue of the device itself.

```
Zigbee2MQTT:debug 2022-03-12 21:00:27: Received Zigbee message from '0xa4c1389d9b178ece', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
```

nohn, 2022-03-14:
So far, I was only look at ```"power":0```, but in case ```"activePower":0``` means, the device itself is broken, I guess that's the case for this device. There was a constant load of 15-20 W connected to the device throughout all the time:

```
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:22:08: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:26:38: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:31:05: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:35:33: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:40:03: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:44:30: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:48:58: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:53:28: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 05:57:55: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:06:53: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:11:20: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:15:50: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:20:18: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:24:47: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:29:15: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:33:43: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:38:12: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:38:15: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:38:18: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:42:40: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:47:08: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:47:11: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 06:51:37: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
[... cut ...]
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 08:03:13: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 08:03:16: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
./2022-03-12.20-52-45/log.txt:debug 2022-03-14 08:07:40: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
```

nohn, 2022-03-14:
As I guess there isn't anything you can do about it, feel free to close this issue. Maybe add a remark to the docs, that theses devices are b0rked (they are sold as SHP-13)

Koenkk, 2022-03-14:
It is only the case for 1 of your devices right? If yes my assumption is that you were unlucky to receive a broken device.

nohn, 2022-03-14:
I have one device that has a constant load connected to it and so can easily be identified as faulty. All other (in sum I currently have four of these devices operationa) have non-permant loads connected and are most of the time expected to be at 0 W. 

At least one of those definitely is also affected. I still have to test the other ones by putting a constant load on them for testing.

nohn, 2022-03-15:
I can confirm the second one to behave the same like the first one:

```
Zigbee2MQTT:debug 2022-03-15 17:32:31: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:14:52: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":115}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:16:22: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:16:27: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":114}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:20:51: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:20:56: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":111}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:25:19: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:25:24: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":110}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:29:31: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:49:17: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":125}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:49:22: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":111}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:52:11: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:52:16: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":111}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:56:38: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 18:56:43: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":109}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 19:01:06: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 19:01:11: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":110}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 19:03:37: Received Zigbee message from 'Kueche Dunstabzugshaube Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
```

As well as the third one to behave like the first one:

```Zigbee2MQTT:debug 2022-03-15 20:54:22: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":7}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 21:02:46: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":19}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 21:02:51: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":31}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 21:06:00: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 21:06:05: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":33}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 21:10:30: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:debug 2022-03-15 21:14:49: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-15 21:14:54: Received Zigbee message from 'Badezimmer Waschbecken Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":5}' from endpoint 1 with groupID 0
```

So with 3 of 4 and the 4th being connected to a toaster, so no long periods with a predictable load possible, I guess it's safe to assume, this generation of SHP-13 is broken.

GSzabados, 2022-03-16:
@nohn, could you try with the original converter, but changing the parameters as below:

[image]

Basically try to set the max reporting interval to 60 seconds.

nohn, 2022-03-16:
@GSzabados I'll try that. What's the meaning of "Min rep change"? Does that mean, a change is only reported if let's say power changes by more than 10 Watt?

nohn, 2022-03-16:
First results since changing to 60s:

```
Zigbee2MQTT:debug 2022-03-16 18:28:10: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure_reporting' with data '{"attribute":"activePower","cluster":"haElectricalMeasurement","id":"Wohnzimmer Rack Plug/1","maximum_report_interval":"60","minimum_report_interval":5,"reportable_change":10,"transaction":"ndw32-1"}'
Zigbee2MQTT:info  2022-03-16 18:28:11: Configured reporting for 'Wohnzimmer Rack Plug/1', 'haElectricalMeasurement.activePower'
Zigbee2MQTT:info  2022-03-16 18:28:11: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"activePower","cluster":"haElectricalMeasurement","id":"Wohnzimmer Rack Plug/1","maximum_report_interval":"60","minimum_report_interval":5,"reportable_change":10},"status":"ok","transaction":"ndw32-1"}'
Zigbee2MQTT:debug 2022-03-16 18:29:10: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:30:10: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:30:48: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:30:53: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:31:53: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":17}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:32:53: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:33:53: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:34:53: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:35:16: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:35:21: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:36:21: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:37:21: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:38:21: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:39:21: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:39:45: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:39:50: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:40:50: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:41:50: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":17}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:42:50: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:43:53: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:44:13: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-16 18:44:21: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:debug 2022-03-16 18:53:10: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
```

For the moment it seems, every 6th value is reported as 0, I'll observe that a little longer and keep you posted.

Weak signal is unlikely, interference could be an issue. I have a vast Zigbee mesh and that plug is connected to many other routers:

[image]

LQI is currenlty 78.

However: None of my other plugs including 6 TS0121, of those three with a lower LQI show this particular behaviour.

nohn, 2022-03-17:
Other plugs have different loads like this one having connected two Laptops:

```
Zigbee2MQTT:debug 2022-03-17 12:54:04: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":35}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 12:55:56: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 12:56:01: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":31}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 12:58:56: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":42}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 12:59:20: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":31}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:00:23: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:00:28: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":27}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:01:16: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":40}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:01:21: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":29}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:01:44: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":44}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:01:49: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":31}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:02:12: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":52}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:02:17: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":34}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:04:30: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":46}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:04:35: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":33}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:04:51: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:04:56: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":28}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:05:59: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":43}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:06:17: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":31}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:07:43: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":53}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:07:48: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":89}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:07:53: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":72}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:07:58: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":91}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:09:21: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:09:26: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":92}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:10:08: Received Zigbee message from 'Schreibtisch Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":78}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:debug 2022-03-17 13:35:44: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":144}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:36:07: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-17 13:36:12: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":144}' from endpoint 1 with groupID 0
```

nohn, 2022-03-18:
Current and Power correlate:

```
Zigbee2MQTT:debug 2022-03-18 14:42:27: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:42:27: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":152}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:43:27: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:43:27: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":144}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:43:51: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:43:51: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:43:56: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":144}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:43:56: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:44:56: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":152}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-03-18 14:44:56: Received Zigbee message from 'Wohnzimmer Rack Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":18}' from endpoint 1 with groupID 0
```

Stable German electricity, directly from the grid. 3 phases in house, happens on all phases, but only with the _TZ3000_amdymr7l devices. No "funny" devices in my in house grid. Also happens, if I connect the plugs to electric circuits in other houses. 

Happens with _TZ3000_amdymr7l devices from three batches (two ordered @ Banggood, one ordered @ AliExpress).

Does not happen with lumi.plug.mmeu01 and not with  _TZ3000_g5xawfcq (that is sold as SHP13 as well). Also not Shelly Plugs (Wifi) and not with AVM plugs (DECT).

So yes: I'm 100% sure this is a serial fault in the _TZ3000_amdymr7l devices. I'd be happy to donate one to one of the Z2M core developers if they like to investigate deeper.

A ticket @ Blitzwolf is open, Blitzwolf is investigating with the supplier currently.

Mike19700, 2022-03-23:
@nohn ,
I also have the _TZ3000_amdymr7l from Banggood.
i tried one and it does not work. I even tried the .js file from Koen but also no right device.
Can you provide me the ticket from Blitzwolf so I can follow it?

nohn, 2022-03-23:
@Mike19700 although those devices are supported by Zigbee2MQTT (#11703), you currently have to run the dev version of it, as the changes haven't made it yet to the stable version.

Mike19700, 2022-03-24:
@nohn,
I installed a dev version and I got the same problem as you have.
{
    "child_lock": "UNLOCK",
    "energy": 0,
    "indicator_mode": "off/on",
    "linkquality": 65,
    "state": "ON",
    "current": null,
    "power": null,
    "power_outage_memory": null,
    "voltage": null
}

nohn, 2022-03-24:
Thanks for confirming.

@Koenkk how about putting a warning in the devices database that _TZ3000_amdymr7l report power, voltage and current unreliably?

Koenkk, 2022-03-24:
@nohn you can edit the docs by clicking the "✏ Help to make the docu better and edit this page on Github ✌ " below here: https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html

nohn, 2022-04-05:
Over at https://github.com/Koenkk/zigbee2mqtt/issues/11703#issuecomment-1088399144, it seems like it worked when working with a custom extension, when there was no official support for these devices in z2m, so there seems to be a solution for this problem.

GSzabados, 2022-04-06:
Probably the problem is with the automatic reporting of the device. The aformentioned converter polls it. Probably the polling always reports the right values.

nohn, 2022-04-07:
No problem. #12097 says "... in a few hours from now". Is it already available on ghcr.io/koenkk/zigbee2mqtt:latest-dev or should I wait a few hours before updating and testing?

Koenkk, 2022-04-07:
@nohn once https://github.com/Koenkk/zigbee2mqtt/runs/5871839033?check_suite_focus=true is finished it is in latest-dev (should be in a few mins from now)

nohn, 2022-04-07:
First results are looking good:

```
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:33: Device '0xa4c1380b4132cd1e' joined
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380b4132cd1e","ieee_address":"0xa4c1380b4132cd1e"},"type":"device_joined"}'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:34: MQTT publish: topic 'zigbee2mqtt/0xa4c1380b4132cd1e/availability', payload 'online'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:34: Starting interview of '0xa4c1380b4132cd1e'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380b4132cd1e","ieee_address":"0xa4c1380b4132cd1e","status":"started"},"type":"device_interview"}'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1380b4132cd1e","ieee_address":"0xa4c1380b4132cd1e"},"type":"device_announce"}'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:36: Successfully interviewed '0xa4c1380b4132cd1e', device has successfully been paired
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:36: Device '0xa4c1380b4132cd1e' is supported, identified as: TuYa Smart plug (with power monitoring by polling) (TS011F_plug_3)
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart plug (with power monitoring by polling)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":3,"description":"Recover state after power outage","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":7,"description":"Plug LED indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["off","off/on","on/off","on"]},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS011F_plug_3","options":[{"access":2,"description":"This device does not support reporting electric measurements so it is polled instead. The default poll interval is 60 seconds, set to -1 to disable.","name":"measurement_poll_interval","property":"measurement_poll_interval","type":"numeric","value_min":-1},{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"}],"supports_ota":true,"vendor":"TuYa"},"friendly_name":"0xa4c1380b4132cd1e","ieee_address":"0xa4c1380b4132cd1e","status":"successful","supported":true},"type":"device_interview"}'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:36: Configuring '0xa4c1380b4132cd1e'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:36: MQTT publish: topic 'homeassistant/switch/0xa4c1380b4132cd1e/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c1380b4132cd1e/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c1380b4132cd1e/set","device":{"configuration_url":"https://zigbee2mqtt.home.nohn.net/#/device/0xa4c1380b4132cd1e/info","identifiers":["zigbee2mqtt_0xa4c1380b4132cd1e"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring by polling) (TS011F_plug_3)","name":"0xa4c1380b4132cd1e"},"name":"0xa4c1380b4132cd1e","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0xa4c1380b4132cd1e","unique_id":"0xa4c1380b4132cd1e_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:11:36: MQTT publish: topic 'homeassistant/sensor/0xa4c1380b4132cd1e/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c1380b4132cd1e/availability"}],"availability_mode":"all","device":{"configuration_url":"https://zigbee2mqtt.home.nohn.net/#/device/0xa4c1380b4132cd1e/info","id [... cut ...]
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:42:46: MQTT publish: topic 'zigbee2mqtt/0xa4c1380b4132cd1e', payload '{"child_lock":"UNLOCK","current":0.22,"energy":0.01,"indicator_mode":"off/on","linkquality":84,"power":29,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'
zigbee2mqtt                | Zigbee2MQTT:info  2022-04-07 18:43:36: MQTT publish: topic 'zigbee2mqtt/0xa4c1380b4132cd1e', payload '{"child_lock":"UNLOCK","current":0.22,"energy":0.01,"indicator_mode":"off/on","linkquality":84,"power":29,"power_outage_memory":null,"state":"ON","update":{"state":null},"update_available":null,"voltage":228}'

```

nohn, 2022-04-15:
Over a longer time it seems more reliable, but metrics still drop to 0 every few measurements.

closed 2022-05-24 by github-actions[bot]

jonathh21, 2023-09-11:
FWIW.

I have a stack of these devices and they have this issue.

Initially I thought the device was legitimately dropping load but moved to a constant load device saw it again.

Then retrospectively looked at load on all others (10 or so) they all demonstrate this 'issue'.

Not noticed until now as all my automation require load for short periods to trigger flags and then 0 loads.