Koenkk/zigbee2mqtt #11803: No GET converter for WZ5 LED controller

lukejagger, 2022-03-11:
### What happened?

I'm trying to get the current colour of the LED strip by sending the command documented here: https://www.zigbee2mqtt.io/devices/WZ5_rgbcct.html
which reads:

> To read the hue/saturation send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"color":{"hue":"","saturation":""}} 

But it always results in this log:

> Error No converter available for 'zigbee2mqtt/Skydance LED controller 1/get' ("{\"color\":{\"hue\":\"\",\"saturation\":\"\"}}")

I also notice that the colour never updates in Home Assistant when changing the colour using the RT10 2.4ghz controller supplied with the WZ5.  It looks like the colour change is never sent back to Z2M, and is not Get'able either.

### What did you expect to happen?

Z2M should update the colour value as it changes, and allow the GET color to work, if indeed it's readable from the WZ5.

### How to reproduce it (minimal and precise)

send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"color":{"hue":"","saturation":""}} 

### Zigbee2MQTT version

1.24.0-dev commit: 9592147

### Adapter firmware version

20200805

### Adapter

Electrolama Zig-a-zig-ah

Koenkk, 2022-03-11:
@kirovilya could this be added?

lukejagger, 2022-03-11:
Just to add, that if this can be added, it would mean the RT10 colour wheel remote could be mapped to also change other zigbee light's colours! :)

lukejagger, 2022-03-11:
I think there needs to be a "color_temp" converter as well, when in RGB_CCT mode.

lukejagger, 2022-03-11:
Interestingly, when I look at state in the frontend while spinning the colour wheel on the remote, it updates instantly for color temperature, but if I am in RGB mode spinning the wheel, the state for color hue does not move.

lukejagger, 2022-03-16:
Any progress on this?  If I can help (now that I've got to grips with adding converters!), let me know what to try.

closed 2022-04-24 by github-actions[bot]