Koenkk/zigbee2mqtt #11805: ZLinky measurement_poll_interval parameter not working

capi92, 2022-03-11:
### What happened?

Lixee Linky running firmware date 20220217 version 4000-0005
Linky in standard mode, single phase, no production.

Despite measurement_poll_interval set to 300s, pooling remains every 60s.

[image]

### What did you expect to happen?

Setting 300s pooling interval, I expect sensor value update very 5 minutes.

### How to reproduce it (minimal and precise)

Set anything else than 60s default value, pooling remains every 60s.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20210120

### Adapter

SonOff ZIGBEE 3.0 USB DONGLE PLUS

Koenkk, 2022-03-12:
@vk496 could you check this?

vk496, 2022-03-12:
@capi92 the attributes that are not reportable, use the time you set. Since the device is very aggressive (as you can see), I forced to all reportable attributes use 60s minimum too. You can check that under reporting tab and adjust your needs

capi92, 2022-03-12:
[image]
 @vk496 I guess you are referring to this ?
I just changed everything to 300 except ActiveEnerfyOutDP1 that does not support 300.

BTW, I guess there is a typo, should be Activer Ener**g**youtDP1...

Will let you know in 1 hour or so.

Pofilo, 2022-03-12:
@renaiku I tag you as asked in #11674 :)

capi92, 2022-03-12:
Workaround for the time being:

sensor.yaml
```
- platform: filter
  entity_id: sensor.linky_easd02
  name: Filtered hchp
  filters:
    - filter: time_throttle
      window_size: 00:05

- platform: filter
  entity_id: sensor.linky_easd01
  name: Filtered hchc
  filters:
    - filter: time_throttle
      window_size: 00:05
```

capi92, 2022-03-12:
I restarted both Z2M and HA to see if it's better now (did not force restart Z2M initially , Z2M being a container I was thinking it would be restated together with HA when asking for HA to restart).

capi92, 2022-03-12:
I confirm it's now working.

Configuration:

[image]

Results

SELECT
*
FROM states
where
 entity_id = "sensor.linky_easd02" or  entity_id = "sensor.linky_easd01" 
ORDER BY created DESC LIMIT 8;

<div class="row" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none; box-sizing: border-box; display: flex; flex-wrap: wrap; margin-right: -15px; margin-left: -15px;"><div class="col-9" id="content" style="box-sizing: border-box; position: relative; width: 1488px; min-height: 1px; padding-right: 15px; padding-left: 15px; -webkit-box-flex: 0; flex: 0 0 75%; max-width: 75%;"><div style="box-sizing: border-box; margin-top: 20px;">

state_id | domain | entity_id | state | attributes | event_id | last_changed | last_updated | created | old_state_id
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
36821321 | sensor | sensor.linky_easd02 | 16184.525 | {"state_class":"total_increasing","unit_of_measurement":"kWh","device_class":"energy","friendly_name":"Compteur Heures Pleines","last_reset":"1970-01-01T00:00:00+00:00"} | 37889624 | 2022-03-12 15:24:32.991378 | 2022-03-12 15:24:32.991378 | 2022-03-12 15:24:32.991378 | 36821220
36821220 | sensor | sensor.linky_easd02 | 16184.39 | {"state_class":"total_increasing","unit_of_measurement":"kWh","device_class":"energy","friendly_name":"Compteur Heures Pleines","last_reset":"1970-01-01T00:00:00+00:00"} | 37889481 | 2022-03-12 15:19:32.978293 | 2022-03-12 15:19:32.978293 | 2022-03-12 15:19:32.978293 | 36821110
36821110 | sensor | sensor.linky_easd02 | 16184.248 | {"state_class":"total_increasing","unit_of_measurement":"kWh","device_class":"energy","friendly_name":"Compteur Heures Pleines","last_reset":"1970-01-01T00:00:00+00:00"} | 37889333 | 2022-03-12 15:14:33.012252 | 2022-03-12 15:14:33.012252 | 2022-03-12 15:14:33.012252 | 36820995
36820995 | sensor | sensor.linky_easd02 | 16184.098 | {"state_class":"total_increasing","unit_of_measurement":"kWh","device_class":"energy","friendly_name":"Compteur Heures Pleines","last_reset":"1970-01-01T00:00:00+00:00"} | 37889171 | 2022-03-12 15:09:32.990833 | 2022-03-12 15:09:32.990833 | 2022-03-12 15:09:32.990833 | 36820838
36820838 | sensor | sensor.linky_easd02 | 16183.952 | {"state_class":"total_increasing","unit_of_measurement":"kWh","device_class":"energy","friendly_name":"Compteur Heures Pleines","last_reset":"1970-01-01T00:00:00+00:00"} | 37888989 | 2022-03-12 15:07:56.870008 | 2022-03-12 15:07:56.870008 | 2022-03-12 15:07:56.870008 | 36820520
36820837 | sensor | sensor.linky_easd01 | 7874.693 | {"state_class":"total_increasing","unit_of_measurement":"kWh","device_class":"energy","friendly_name":"Compteur Heures Creuses","last_reset":"1970-01-01T00:00:00+00:00"} | 37888988 | 2022-03-12 15:07:56.869308 | 2022-03-12 15:07:56.869308 | 2022-03-12 15:07:56.869308 | 36820519
36820520 | sensor | sensor.linky_easd02 | unknown | {"state_class":"total_increasing","unit_of_measurement":"kWh","device_class":"energy","friendly_name":"Compteur Heures Pleines","last_reset":"1970-01-01T00:00:00+00:00"} | 37888660 | 2022-03-12 15:07:40.724272 | 2022-03-12 15:07:40.724272 | 2022-03-12 15:07:40.724272 | NULL
36820519 | sensor | sensor.linky_easd01 | unknown | {"state_class":"total_increasing","unit_of_measurement":"kWh","device_class":"energy","friendly_name":"Compteur Heures Creuses","last_reset":"1970-01-01T00:00:00+00:00"} | 37888659 | 2022-03-12 15:07:40.724072 | 2022-03-12 15:07:40.724072 | 2022-03-12 15:07:40.724072 | NULL

</div></div></div><div class="row" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none; box-sizing: border-box; display: flex; flex-wrap: wrap; margin-right: -15px; margin-left: -15px;"><div class="col-9 offset-3" style="box-sizing: border-box; position: relative; width: 1488px; min-height: 1px; padding-right: 15px; padding-left: 15px; -webkit-box-flex: 0; flex: 0 0 75%; max-width: 75%; margin-left: 496px; caret-color: rgb(136, 136, 136); color: rgb(136, 136, 136); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, &quot;Helvetica Neue&quot;, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;; font-size: 16px; font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none;"><footer style="box-sizing: border-box; display: block;"><hr style="box-sizing: content-box; height: 0px; overflow: visible; margin-top: 1rem; margin-bottom: 1rem; border-width: 1px 0px 0px; border-top-style: solid; border-top-color: rgba(0, 0, 0, 0.1);"><p></p></footer></div><br class="Apple-interchange-newline"></div><br class="Apple-interchange-newline">

closed 2022-03-12 by capi92