Koenkk/zigbee2mqtt #11810: Zig-a-zig-ah stopped working but works different machine

onlymejosh, 2022-03-11:
### What happened?

Ive been upgrading my ubunutu packages, after a few restarts I am receiving 

```
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:16:33: Error while starting zigbee-herdsman
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:16:33: Failed to start zigbee
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:16:33: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:16:33: Exiting...
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:16:33: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
zigbee2mqtt        |     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
zigbee2mqtt        |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
zigbee2mqtt        |     at Zigbee.start (/app/lib/zigbee.ts:58:27)
zigbee2mqtt        |     at Controller.start (/app/lib/controller.ts:100:27)
zigbee2mqtt        |     at start (/app/index.js:101:5)
```

My configuration hasn't changed

`docker-compose.yaml`

```
zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest
    volumes:
      - ./zigbee2mqtt:/app/data
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    restart: unless-stopped
    environment:
      - TZ=Europe/London
    ports:
      - 8080:8080
```

`configuration.yaml`
homeassistant: true
permit_join: true
frontend:
  port: 8080
advanced:
  availability:
    active:
      timeout: 5
    passive:
      timeout: 1500
  log_level: debug
  pan_id: 6765
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://haus.local:1883'
serial:
  port: /dev/ttyUSB0
devices:
  '0x00178801083cc7ef':
    friendly_name: garden_downlight_1
  '0x00178801083cae7c':
.....
```

```
josh@haus:~$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Mar 11 22:08 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0
```

If I remove my database.db and let zigbe2mqtt create a new one, I receive no errors. 

### Zigbee2MQTT version

1.22.1-dev commit: 53003b3

### Adapter firmware version

20200805

### Adapter

zig-a-zig-a

### Debug log

josh@haus:~/haus$ docker-compose up zigbee2mqtt
Starting zigbee2mqtt ... done
Attaching to zigbee2mqtt
zigbee2mqtt        | Using '/app/data' as data directory
zigbee2mqtt        | 2022-03-11T22:28:10.059Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6765,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
zigbee2mqtt        | 2022-03-11T22:28:10.060Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
zigbee2mqtt        | 2022-03-11T22:28:10.062Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
zigbee2mqtt        | 2022-03-11T22:28:10.063Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
zigbee2mqtt        | 2022-03-11T22:28:10.064Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
zigbee2mqtt        | 2022-03-11T22:28:10.318Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
zigbee2mqtt        | 2022-03-11T22:28:10.319Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
zigbee2mqtt        | 2022-03-11T22:28:11.321Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
zigbee2mqtt        | 2022-03-11T22:28:11.322Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
zigbee2mqtt        | 2022-03-11T22:28:11.572Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352
zigbee2mqtt        | 2022-03-11T22:28:12.030Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
zigbee2mqtt        | 2022-03-11T22:28:12.030Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
zigbee2mqtt        | 2022-03-11T22:28:18.031Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
zigbee2mqtt        | 2022-03-11T22:28:18.031Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
zigbee2mqtt        | 2022-03-11T22:28:24.033Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
zigbee2mqtt        | 2022-03-11T22:28:24.034Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:28:30: Error while starting zigbee-herdsman
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:28:30: Failed to start zigbee
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:28:30: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:28:30: Exiting...
zigbee2mqtt        | Zigbee2MQTT:error 2022-03-11 22:28:30: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
zigbee2mqtt        |     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
zigbee2mqtt        |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
zigbee2mqtt        |     at Zigbee.start (/app/lib/zigbee.ts:58:27)
zigbee2mqtt        |     at Controller.start (/app/lib/controller.ts:100:27)
zigbee2mqtt        |     at start (/app/index.js:97:5)
josh@haus:~/haus$

renaiku, 2022-03-13:
I think there was issues with the zzh firmware that has been fixed by koenk a few month ago. Have you still the issue after performing an update ?

I'm also using a zzh and had a lot of these errors before updating the firmware.

onlymejosh, 2022-03-13:
I haven’t updated zig-a-zig-ah but I did pull the latest zigbee2mqtt. 

I will run the latest firmware and  report back.
 Thanks!

closed 2022-04-20 by github-actions[bot]