Koenkk/zigbee2mqtt #11812: color_temp_startup Error for https://www.zigbee2mqtt.io/devices/TS0505B.html#tuya-ts0505b

acaonweb, 2022-03-12:
### What happened?

In settings, changing color_temp_startup I got an error
[image]

### What did you expect to happen?

Non error at all

### How to reproduce it (minimal and precise)

Just changing settimg

### Zigbee2MQTT version

 1.24.0 commit: 7a2ddf24

### Adapter firmware version

20210120

### Adapter

ZZH MULTIPROTOCOL RF STICK (CC2652R1 - EXTERNAL ANTENNA)

### Debug log

[image]

acaonweb, 2022-03-21:
No ideas?

Koenkk, 2022-03-22:
`UNSUPPORTED_ATTRIBUTE` -> this device doesn't support it.

closed 2022-03-29 by acaonweb