Koenkk/zigbee2mqtt #11827: Sengled E21-N1EA bulbs interview fails

jared-hess, 2022-03-13:
### What happened?

Trying to pair a couple of E21-N1EA bulbs to Z2M but consistently get the following error:
`Failed to interview '0xb0ce181400184384', device has not successfully been paired`
I've tried flashing the latest firmware to the Sonoff, factory resetting the bulbs, deleting and re-adding them to Z2M, switching to the dev branch, moving the bulb closer to the adapter. But always get the "failed to interview" error. I do have a few other devices added to Z2M, these bulbs are the only devices I've had trouble with so far.

After turning on the herdsman debugging, I have noticed the underlying error seems to be somewhat inconsistent. Sometimes it's this:
`zigbee-herdsman:controller:device:log Interview - failed for device '0xb0ce181400184384' with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms`
But sometimes it's this:
`zigbee-herdsman:controller:device:log Interview - failed for device '0xb0ce181400184384' with error 'Error: Interview failed because can not get active endpoints ('0xb0ce181400184384'`

### What did you expect to happen?

The bulbs are listed as supported, so I'd expect them to pair after being factory reset.

### How to reproduce it (minimal and precise)

Factory reset bulbs by turning on/off 10 times rapidly with Z2M set to register new devices. 

### Zigbee2MQTT version

1.24.0-dev commit: 3cf3366

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Here's one example:
```
zigbee2mqtt      | 2022-03-13T04:58:29.268Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,233,189,1,13,0,178,0,44,203,29,0,0,12,1,5,1,0,96,17,204,128,38,0,0,0,233,189,29,183]
zigbee2mqtt      | 2022-03-13T04:58:29.268Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,233,189,1,13,0,178,0,44,203,29,0,0,12,1,5,1,0,96,17,204,128,38,0,0,0,233,189,29,183]
zigbee2mqtt      | 2022-03-13T04:58:29.268Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,233,189,1,13,0,178,0,44,203,29,0,0,12,1,5,1,0,96,17,204,128,38,0,0,0,233,189,29] - 183
zigbee2mqtt      | 2022-03-13T04:58:29.268Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":48617,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":178,"securityuse":0,"timestamp":1952556,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,5,1,0,96,17,204,128,38,0,0,0]}}
zigbee2mqtt      | 2022-03-13T04:58:29.272Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4448,"imageType":32972,"fileVersion":38},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":48617,"endpoint":1,"linkquality":178,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-03-12 20:58:29: Received Zigbee message from '0xb0ce181400184384', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":38,"imageType":32972,"manufacturerCode":4448}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:warn  2022-03-12 20:58:29: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
zigbee2mqtt      | Zigbee2MQTT:warn  2022-03-12 20:58:29: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
zigbee2mqtt      | 2022-03-13T04:58:29.279Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | 2022-03-13T04:58:32.305Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,233,189,1,13,0,174,0,177,176,32,0,0,12,1,6,1,0,96,17,204,128,38,0,0,0,233,189,29,115]
zigbee2mqtt      | 2022-03-13T04:58:32.305Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,233,189,1,13,0,174,0,177,176,32,0,0,12,1,6,1,0,96,17,204,128,38,0,0,0,233,189,29,115]
zigbee2mqtt      | 2022-03-13T04:58:32.305Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,233,189,1,13,0,174,0,177,176,32,0,0,12,1,6,1,0,96,17,204,128,38,0,0,0,233,189,29] - 115
zigbee2mqtt      | 2022-03-13T04:58:32.305Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":48617,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":174,"securityuse":0,"timestamp":2142385,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,6,1,0,96,17,204,128,38,0,0,0]}}
zigbee2mqtt      | 2022-03-13T04:58:32.306Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4448,"imageType":32972,"fileVersion":38},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":48617,"endpoint":1,"linkquality":174,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-03-12 20:58:32: Received Zigbee message from '0xb0ce181400184384', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":38,"imageType":32972,"manufacturerCode":4448}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:warn  2022-03-12 20:58:32: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
zigbee2mqtt      | Zigbee2MQTT:warn  2022-03-12 20:58:32: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
zigbee2mqtt      | 2022-03-13T04:58:32.309Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | 2022-03-13T04:58:35.339Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,233,189,1,13,0,178,0,75,149,35,0,0,12,1,9,1,0,96,17,204,128,38,0,0,0,233,189,29,188]
zigbee2mqtt      | 2022-03-13T04:58:35.339Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,233,189,1,13,0,178,0,75,149,35,0,0,12,1,9,1,0,96,17,204,128,38,0,0,0,233,189,29,188]
zigbee2mqtt      | 2022-03-13T04:58:35.339Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,233,189,1,13,0,178,0,75,149,35,0,0,12,1,9,1,0,96,17,204,128,38,0,0,0,233,189,29] - 188
zigbee2mqtt      | 2022-03-13T04:58:35.339Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":48617,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":178,"securityuse":0,"timestamp":2331979,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,9,1,0,96,17,204,128,38,0,0,0]}}
zigbee2mqtt      | 2022-03-13T04:58:35.340Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":9,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4448,"imageType":32972,"fileVersion":38},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":48617,"endpoint":1,"linkquality":178,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-03-12 20:58:35: Received Zigbee message from '0xb0ce181400184384', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":38,"imageType":32972,"manufacturerCode":4448}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:warn  2022-03-12 20:58:35: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
[... cut ...] ,"imageType":32972,"manufacturerCode":4448}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:warn  2022-03-12 22:39:37: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
zigbee2mqtt      | Zigbee2MQTT:warn  2022-03-12 22:39:37: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
zigbee2mqtt      | 2022-03-13T06:39:37.677Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

closed 2022-04-20 by github-actions[bot]

jared-hess, 2022-04-20:
This is still an issue.

Emiller88, 2023-03-29:
@Koenkk just in case you weren't aware. This has also been reported in https://github.com/yllibed/Zigbee2MqttAssistant/issues/396