Koenkk/zigbee2mqtt #11828: Centralite 3400 Keypad

hvddrift, 2022-03-13:
### Link

https://github.com/zigpy/zha-device-handlers/issues/838

### Database entry

{"id":124,"type":"EndDevice","ieeeAddr":"0x000d6f000cfb7225","nwkAddr":50284,"manufId":4174,"manufName":"CentraLite","powerSource":"Battery","modelId":"3400","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1025,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25,1281],"clusters":{"genBasic":{"attributes":{"modelId":"3400","manufacturerName":"CentraLite","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00258f1f87","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":30}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2905}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b00258f1f87","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00258f1f87","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00258f1f87","endpointID":1},{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x00124b00258f1f87","endpointID":1},{"cluster":1281,"type":"endpoint","deviceIeeeAddress":"0x00124b00258f1f87","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"stackVersion":2,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1647159301441,"defaultSendRequestWhen":"active"}

### Comments

I have the Centralite 3400. When I paired the device it is shown as the [Centralite 3400-D](https://www.zigbee2mqtt.io/devices/3400-D.html)

It sort of works as I can get on way traffic from the device. But an unable to send messages to the unit.

I have linked the ZHA support page for the device. I am unsure (beyond my skill level) on the best way to move forward. 

I assume I would need to unpair the keypad. And update my database to commumicate.

Any help would be appreciated.

Hans

hvddrift, 2022-03-18:
@Koenkk just seeing if there is any more information I can gather to add this device?

Koenkk, 2022-03-18:
> But an unable to send messages to the unit.

what is logged when you try?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

hvddrift, 2022-03-19:
I am using zigbee2mqtt via docker. On the web interface I get this result

```
Received MQTT message on 'zigbee2mqtt/centralite_ground/set' with data '{"arm_mode":{"mode":"disarm"}}'
Publishing 'set' 'arm_mode' to 'centralite_ground'
```

or

```
Received MQTT message on 'zigbee2mqtt/centralite_ground/set' with data '{"arm_mode":{"mode":"arm_all_zones"}}'
Publishing 'set' 'arm_mode' to 'centralite_ground'
```

But the keypad panel does not register any messages. 

and this is me entering an alarm mode from the keypad.

```
Received Zigbee message from 'centralite_ground', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"6321","zoneid":23}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/centralite_ground/set' with data '{"arm_mode": {"transaction": "112","mode": "arm_day_zones"}}'
Publishing 'set' 'arm_mode' to 'centralite_ground'
Received Zigbee message from 'centralite_ground', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
```

Koenkk, 2022-03-19:
After `Publishing 'set' 'arm_mode' to 'centralite_ground'` is there anything more logged like an error?

hvddrift, 2022-03-19:
No, nothing. But just so we are on the same page, the device is listed as the 3400-D in zigbee2mqtt. The actual model number is 3400 on my device.

There seems to be some subtle differences

hvddrift, 2022-03-19:
Here is what publishing looks like.

> After `Publishing 'set' 'arm_mode' to 'centralite_ground'` is there anything more logged like an error?

Here are the results.
[image]

`Received MQTT message on 'zigbee2mqtt/centralite_ground/set' with data '{ "arm_mode": { "mode": "arm_all_zones" } }'
Publishing 'set' 'arm_mode' to 'centralite_ground'`

hvddrift, 2022-03-19:
and testing another command

`Received MQTT message on 'zigbee2mqtt/centralite_ground/set' with data '{ "arm_mode": { "mode": "arm_day_zones" } }'
Publishing 'set' 'arm_mode' to 'centralite_ground'`

Koenkk, 2022-03-20:
@[kevincaradant](https://github.com/kevincaradant) do you know how to control this device?

hvddrift, 2022-03-28:
@Koenkk @kevincaradantany other suggestions?

Mikhail5555, 2022-04-18:
I am not sure how relevant this is, but i saw this implementation in SmartThingsPublic: 
https://github.com/mitchpond/SmartThingsPublic/blob/master/devicetypes/mitchpond/centralite-keypad.src/centralite-keypad.groovy
It looked like part of the protocol

Mikhail5555, 2022-04-18:
```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const constants = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const arm_mode = {
    key: ['arm_mode'],
    convertSet: async (entity, key, value, meta) => {
        meta.logger.info(`[AlarmKeypad] Converter called for ${JSON.stringify(key)} and ${JSON.stringify(value)}`);
        const isNotification = value.hasOwnProperty('transaction');
        const modeSrc = isNotification ? constants.armNotification : constants.armMode;
        const mode = utils.getKey(modeSrc, value.mode, undefined, Number);
        if (mode === undefined) {
            throw new Error(`Unsupported mode: '${value.mode}', should be one of: ${Object.values(modeSrc)}`);
        }

        if (isNotification) {
            entity.commandResponse('ssIasAce', 'armRsp', {armnotification: mode}, {}, value.transaction);

            // Do not update PanelStatus after confirming transaction.
            // Instead the server should send an arm_mode command with the necessary state.
            // e.g. exit_delay as a result of arm_all_zones
            return;
        }

        globalStore.putValue(entity, 'panelStatus', mode);
        const payload = {panelstatus: mode, secondsremain: 0, audiblenotif: 0, alarmstatus: 0};
        
        meta.logger.info(`[AlarmKeypad] Sending message ${JSON.stringify(payload)}`);
        entity.commandResponse('ssIasAce', 'panelStatusChanged', payload);
    },
};

const definition = {
    zigbeeModel: ['3400-G'],
    model: '3400-G',
    vendor: 'Centralite',
    description: '3-Series security keypad',
    meta: {battery: {voltageToPercentage: '3V_2100'}},
    fromZigbee: [fz.command_arm_with_transaction, fz.temperature, fz.battery, fz.ias_ace_occupancy_with_timeout],
    toZigbee: [arm_mode],
    exposes: [e.battery(), e.temperature(), e.occupancy(),
        exposes.numeric('action_code', ea.STATE).withDescription('Pin code introduced.'),
        exposes.numeric('action_transaction', ea.STATE).withDescription('Last action transaction number.'),
        exposes.text('action_zone', ea.STATE).withDescription('Alarm zone. Default value 0'),
        e.action(['disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'exit_delay', 'emergency'])],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const clusters = ['msTemperatureMeasurement', 'genPowerCfg', 'ssIasZone', 'ssIasAce'];
        await reporting.bind(endpoint, coordinatorEndpoint, clusters);
        await reporting.temperature(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
    onEvent: async (type, data, device) => {
        if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce' && globalStore.hasValue(device.getEndpoint(1), 'panelStatus')) {
            const payload = {
                panelstatus: globalStore.getValue(device.getEndpoint(1), 'panelStatus'),
                secondsremain: 0x00, audiblenotif: 0x00, alarmstatus: 0x00,
            };
            await device.getEndpoint(1).commandResponse(
                'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
            );
        }
    },

};

module.exports = definition;
```

This seems to work mostly for my 3400-G, even though it doesnt seem to accept commands like `exit_delay`, in the templates above there seems to be some custom hack to make the device beep during exit delay.

closed 2022-05-26 by github-actions[bot]

abenedet78, 2023-01-16:
@Mikhail5555 -  you mentioned this worked for your 3400-G:

> This seems to work mostly for my 3400-G, even though it doesnt seem to accept commands like `exit_delay`, in the templates above there seems to be some custom hack to make the device beep during exit delay.

For the life of me, I cannot get this to work. I'm using Z2M through HomeAssistant. I have 20 other devices connected and working properly. Conbee II stick flashed with recent firmware and I'm using the latest version of Z2M. 

I used the code above to create a custom external converter. The device is recognized, but I only get values for OCCUPANCY, LQI, and sometimes ACTION (but it says empty string "" when I press some random pattern of buttons). 

Are you please able to help me navigate this? You're the only one I've found that has the "G" version working.  Thanks in advance!

[image]

abenedet78, 2023-01-16:
Thx for the response. Did you get messages coming through regarding armed states? I'm not and I believe I need these messages in order to process the HA automations you are referencing. I've tried to pair it at least 10x now, but maybe I'll keep trying.