Koenkk/zigbee2mqtt #11835: Xiaomi DJT11LM "vibration_timeout" does not work

lux73, 2022-03-13:
### What happened?

Setting `vibration_timeout` in devices.yaml does not work. There's no reset. I had to do via OpenHAB expire command

https://www.zigbee2mqtt.io/devices/DJT11LM.html#options

snip from my devices.yaml:

```
'0x00xxxxxxxxxxxxxx':
  friendly_name: sensor-vibration
  debounce: 1
  vibration_timeout: 60
  filtered_attributes:
    - angle_x_absolute
    - angle_y_absolute
```

### What did you expect to happen?

after setting `vibration_timeout` in devices.yaml that the Sensor is resetting after my specified 60s from z2m

### How to reproduce it (minimal and precise)

set `vibration_timeout` & confirm it does not work

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Slaesh CC2652RB

Koenkk, 2022-03-13:
- Make sure you are running the latest z2m first before creating an issue (1.24.0)
- If it still appear there, please provide the debug log when triggering vibration on the device + 3 minutes after that.

lux73, 2022-03-13:
ups, sorry. Will update tomorrow & provide the Logs then

lux73, 2022-03-15:
sorry for the Delay - after updating to 1.24.0 i had some issues with Xioami RTCGQ13LM Motion Detection...

this is the Debug Log:
[zigbee_debug.log](https://github.com/Koenkk/zigbee2mqtt/files/8257212/zigbee_debug.log)

Vibration triggert @22:09:01 - sensor-vibration

hope this helps to find the Bug :-)

Koenkk, 2022-03-16:
I don't see the issue:

vibration triggered (= vibration: true published)
```
[32mZigbee2MQTT:info [39m 2022-03-15 22:09:02: MQTT publish: topic 'zigbee2mqtt/sensor-vibration', payload '{"action":"tilt","angle":19,"angle_x":19,"angle_y":69,"angle_z":8,"battery":99,"last_seen":1647378541944,"linkquality":120,"sensitivity":"high","strength":7,"vibration":true,"voltage":2985}'
```

1 minute after this vibration false is published:
```
[32mZigbee2MQTT:info [39m 2022-03-15 22:10:01: MQTT publish: topic 'zigbee2mqtt/sensor-vibration', payload '{"angle":19,"angle_x":19,"angle_y":69,"angle_z":8,"battery":99,"last_seen":1647378541944,"linkquality":120,"sensitivity":"high","strength":7,"vibration":false,"voltage":2985}'
```

lux73, 2022-03-17:
i just realised that i'm using the `action` Trigger in my OpenHAB System not the `vibration` - i'm feeling stupid now :-/

thx for your Help

I will close this Thread

closed 2022-03-17 by lux73