Koenkk/zigbee2mqtt #11837: New Blitzwolf SHP-13 unsupported (Probably changed to TS011F)

wilqq-the, 2022-03-13:
### What happened?

Recently I bought additional plugs from Blitzwolf (SHP13) - I have already like 5 of them and they are working perfectly fine. New ones are appearing as unsupported - probably something changed inside (firmware).

Old ones reported as TS0121_plug.
[image]

News ones reporting as TS011F.
[image]

### What did you expect to happen?

Expected to add BW-SHP13 without problems to Z2M.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20210708

### Adapter

 zStack3x0 (Sonoff Zigbee USB dongle)

Koenkk, 2022-03-13:
Please provide the data/database.db entry of this device.

nohn, 2022-03-14:
Looks like #11703 to me.

closed 2022-04-21 by github-actions[bot]