Koenkk/zigbee2mqtt #11838: [New device support]: please can you add the ledvance PAR16S TW

dj9p, 2022-03-13:
### Link

https://www.ledvance.fr/particuliers/produits/smart-home/lampes-smart-home/smart-zigbee/spot-smart-spot-avec-technologie-zigbee/ampoules-led-reflecteur-avec-technologie-smart-home--temperature-de-couleur-reglable--par16-c6270

### Database entry

Adresse IEEE     0xf0d1b800001c0bd7 Adresse réseau     0x465B Version Firmware     02136550

### Comments

Sorry but i'm french and i don't understand how to use converter can you help me?
I have test to add the new device but doesn't work...
[image]

### Supported color modes

color temperature 2700-6500 K

Koenkk, 2022-03-13:
Please provide the correct color temp range as explained on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

dj9p, 2022-03-13:
sorry but i have nothing in the consol dev see,
[image]

dj9p, 2022-03-13:
[image]

dj9p, 2022-03-13:
i can't read the color temp range. Please see : 
[image]

Koenkk, 2022-03-14:
Added the device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-14 by Koenkk