Koenkk/zigbee2mqtt #11842: Sonoff ZBMINI - quite often device will not respond - error log included.

bednim, 2022-03-13:
### What happened?

sonoff zbmini - got few of these devices and quite often some of them will not respond to commands (on/off).

### What did you expect to happen?

have normal reliable control over the device.

### How to reproduce it (minimal and precise)

just try and toggle the relay few times in a row - when you see that the toggle doesn't react to last change (goes back to previous state) - this is when it breaks.

### Zigbee2MQTT version

1.24.0-1

### Adapter firmware version

26720700

### Adapter

conbee II stick

### Debug log

Zigbee2MQTT:error 2022-03-13 21:09:03: Publish 'set' 'state' to 'sonoff zbmini terrace garden corner' failed: 'Error: Command 0x847127fffe92f39d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:error 2022-03-13 21:09:10: Publish 'set' 'state' to 'sonoff zbmini terrace garden corner' failed: 'Error: Command 0x847127fffe92f39d/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

willmackey2nd, 2022-03-14:
Not meaning to highjack your issue but thought to join instead of creating another one.
I've spent hours reading several old discussion about "zigbee becoming unresponsive after x time" but none of them seem to give any answer and it's becoming really frustrating to have zero reliability with these things.

I have slaeh's CC2652RB stick that I use with HA (that I run as a VM on Proxmox).
Running the network on channel 1. My own WiFi APs (two Unifi Wifi6 Lites) are not even partly overlapping. Neighbor has 2.4Ghz Wifi channel 11, but that's 100 meters away. 
At the moment I have very little network consisting of only few mains powered nodes:
- 4x Sonoff ZBMINI
- 5x Silvercrest smart plugs (from Lidl)

Coordinator is in a quite central location.

Everything works nicely when I setup the network but after one night, it just becomes really unreliable. One of the smart plugs might function fine when I toggle them, but they might fail to report the status and also fail to response to any following commands. They'll just stay dead for some time and might then self heal or in some cases need rebooting (unplug and replug).

I've gone through deleting devices DB and configuration backup json, reflashing the coordinator, reinstalling z2m etc. Nothing seems to help.

Like said, it's becoming really frustrating and even more frustrating is to notice that all the discussion about the same issue just end up in stale issues and dead forum threads.

bednim, 2022-03-15:
@Docteur-No I have them very close to coordinator and also routers but still this error carries on too often.

Its pretty annoying to find out that external light didn't come off after motion was cleared.

Thanks!

Docteur-No, 2022-03-15:
Try to buy GIRIER JR-ZDS01 switches (Aliexpress.com) Which seems to be of a good quality

willmackey2nd, 2022-03-20:
@Docteur-No can't be the poor radio in my case either. All devices have good LQI and max distance between routering nodes is only 3m or so.

It's something in the traffic... like said my nodes might work fine on first try but after that it just blacks out.
For example:
1. All nodes have been idling few hours. No commands has been sent.
2. I toggle one of the smart plugs on or off and it responds instantly and reports changed status just fine.
3. I try to toggle that same plug again few seconds later and it's not responding at all. I need to try few times to make it work and the node's status will update really slowly back to HA even after I finally manage to toggle it.

To me it seems more like something fundamentally wrong in the network or z2m, especially because this behavior starts some time after setting up completely new network and everything works really well during first minutes.

aZa1905, 2022-03-21:
I can only add on to this.
Not a zbmini (mine works fine) but sometimes my powerplugs from IKEA just dont respond anymore, which adds on to the theory that something in z2m is wrong or maybe with the conbee stick?

I would be interested to find a solution as well!

viljamimaki, 2022-03-23:
Here's MQTT traffic from an example situation. I'm controlling _lattiavalo_oh_ here and first it responds to ON request nicely. Then few seconds later I request OFF but only get status reply from a different node. Then after second try I get timeout but then it also finally responds to the request. Notice how the link quality drops dramatically in the reply messages. Not seen here but quickly after that it begins to report ~140 LQI again.

`zigbee2mqtt/lattiavalo_oh/set ON
zigbee2mqtt/lattiavalo_oh {"linkquality":138,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":138,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":134,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":134,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":138,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":134,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":138,"state":"ON"}
zigbee2mqtt/lattiavalo_oh/set OFF
zigbee2mqtt/ulkovalot_u {"linkquality":123,"state":"OFF"}
zigbee2mqtt/lattiavalo_oh/set OFF
zigbee2mqtt/bridge/logging {"level":"error","message":"Publish 'set' 'state' to 'lattiavalo_oh' failed: 'Error: Command 0x842e14fffecd0ddb/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 85 - 1 - 55 - 6 - 11 after 10000ms)'"}
zigbee2mqtt/lattiavalo_oh {"linkquality":25,"state":"OFF"}
zigbee2mqtt/lattiavalo_oh {"linkquality":25,"state":"OFF"}
zigbee2mqtt/ulkovalot_w1 {"linkquality":25,"state":"ON"}`

willmackey2nd, 2022-03-23:
Here's MQTT traffic from an example situation. I'm controlling a node called _"lattiavalo_oh"_ here and first it responds to ON request nicely. Although I don't know why it's receiving so many status replies.

Then few seconds later I request OFF but only get status reply from a different node. Then after second try I get timeout but then it also finally responds to the request. 

Notice how the link quality drops dramatically in the reply messages. Not seen here but quickly after that it begins to report ~140 LQI again.

```
zigbee2mqtt/lattiavalo_oh/set ON
zigbee2mqtt/lattiavalo_oh {"linkquality":138,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":138,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":134,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":134,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":138,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":134,"state":"ON"}
zigbee2mqtt/lattiavalo_oh {"linkquality":138,"state":"ON"}
zigbee2mqtt/lattiavalo_oh/set OFF
zigbee2mqtt/ulkovalot_u {"linkquality":123,"state":"OFF"}
zigbee2mqtt/lattiavalo_oh/set OFF
zigbee2mqtt/bridge/logging {"level":"error","message":"Publish 'set' 'state' to 'lattiavalo_oh' failed: 'Error: Command 0x842e14fffecd0ddb/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 85 - 1 - 55 - 6 - 11 after 10000ms)'"}
zigbee2mqtt/lattiavalo_oh {"linkquality":25,"state":"OFF"}
zigbee2mqtt/lattiavalo_oh {"linkquality":25,"state":"OFF"}
zigbee2mqtt/ulkovalot_w1 {"linkquality":25,"state":"ON"}
```

bednim, 2022-03-27:
Added two zbminis yesterday - behaviour is pretty terrible at the moment, power cycling devices doesn't even bring control back 😢

I just hope someone could solve this issue!

willmackey2nd, 2022-04-14:
I paired everything again once more but switched from channel 11 to 25. Even though neighbors overlapping signal was really weak, it seems it affected ZB network enough to make is unusable. Did not move any node and only other thing I changed was removed one long stairway wiring that caused false triggering of one ZBMINI. I doubt that has any effect to ZB radio but then again who knows how well these boards are designed against interference.

New setup has been up 2 days now without issues so not that long but earlier it crapped out after one night every time.

bednim, 2022-04-14:
Please let us know in few days how is it performing.

Thank you!

willmackey2nd, 2022-04-18:
Still all fine. Not a single hiccup after the mentioned changes. I guess the ch11 was just so noisy even with the low signals from far away that it crippled the network after it had fully formed itself (all my devices are serving as routers too).

bednim, 2022-04-18:
@willmackey2nd

I have too performed channel change (from 11 to 25) and on WiFi between 1 to 6 (3 AP) - so far it has been surprisingly amazing. All zbminis perform without any hiccup, reporting is also fine in Home Assistant, its instant, even using external switch on sonoff devices.

I just can't believe how difference channel change has made.

I'm very pleased with my system now.

Even Ikea bulbs are behaving properly.

Hope it will stay this way.

@aZa1905 @Docteur-No you may want to try the same with your networks.

closed 2022-05-26 by github-actions[bot]