Koenkk/zigbee2mqtt #11843: Xiaomi RTCGQ13LM occupancy resets to false even there is occupancy in Front of Sensor after Update to z2m 1.24.0

lux73, 2022-03-14:
### What happened?

as the Title described after Update z2m from 1.22.2 to 1.24.0 the Occupancy resets after the desired "detection_interval" 

https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html#detection-interval-numeric

### What did you expect to happen?

with v1.22.2 the Occupancy does NOT resets to false when there is occupancy all the time in front of Sensor - it stays in true

**this should be the correct behavior!**

i have extra bought two of these for Corridor & Bath while the cheap RTCGQ11LM do this: they reset even there is much occupancy...

there was a change: https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html#detection-interval-numeric

before this Change this was **occupancy_timeout** and worked lika a charme

could this be the Problem? How to revert to the old behavior?

now i have to get out from the Sensor Coverage & wait few Seconds before the Sensor is discover new occupancy... :-(

if i stay in the Coverage there is no new Occupancy Detection

### How to reproduce it (minimal and precise)

update z2m from 1.22.2 to 1.24.0

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

Slaesh CC2652RB

Otnow, 2022-03-14:
In the mode when the sensor considers that it is communicating with the native hub, it stops transmitting the event of the end of motion detection on its own, and for it this event has to be generated by the converter by a timeout that is equal to the set detection period (`detection_interval`).

At the moment, that is why, when the timeout ends, the sensor reports the end of the movement even if in fact there is still movement.

To fix this problem, I made changes to the converter and suggest you test it.

In `Settings (specific)`, I added the `motion_wait_timeout` parameter (_value in **ms**_):
  
_If during the time specified in this parameter, after the end of the detection period, the sensor continues to detect movement, then the end of movement event is not generated and a new timeout is started._

You need to [add an external converter](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters) presented below to z2m and set a delay acceptable for you after the end of the detection interval.

In case of a positive effect, I will create a PR for inclusion in the z2m repository.

- **RTCGQ13LM_with_motion_wait_timeout.js**

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    RTCGQ13LM_occupancy: {
        // This is for occupancy sensor that only send a message when motion detected,
        // but do not send a motion stop.
        // Therefore we need to publish the no_motion detected by ourselves.
        cluster: 'msOccupancySensing',
        type: ['attributeReport', 'readResponse'],
        options: [exposes.numeric('motion_wait_timeout').withValueMin(0).withUnit('ms')
            .withDescription('If a new motion event occurs within the time specified in this option (value is in ms) ' +
            'after the end of "detection_interval", the "occupancy" state is not cleared')],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.occupancy !== 1) {
                // In case of 0 no occupancy is reported.
                // https://github.com/Koenkk/zigbee2mqtt/issues/467
                return;
            }

            // The occupancy sensor only sends a message when motion detected.
            // Therefore we need to publish the no_motion detected by ourselves.
            let timeout = meta && meta.state && meta.state.hasOwnProperty('detection_interval') ? meta.state.detection_interval : 60;
            const delay = options && options.hasOwnProperty('motion_wait_timeout') ? options.motion_wait_timeout : 0;
            if (delay > 0) timeout += delay/1000;

            // Stop existing timers because motion is detected and set a new one.
            globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
            globalStore.putValue(msg.endpoint, 'timers', []);

            if (timeout !== 0) {
                const timer = setTimeout(() => {
                    publish({occupancy: false});
                }, timeout * 1000);

                globalStore.getValue(msg.endpoint, 'timers').push(timer);
            }

            return {occupancy: true};
        },
    }
}

const definition = {
    zigbeeModel: ['lumi.motion.agl04'],
    model: 'RTCGQ13LM',
    vendor: 'Xiaomi',
    description: 'Aqara high precision motion sensor',
    fromZigbee: [fzLocal.RTCGQ13LM_occupancy, fz.aqara_opple, fz.battery],
    toZigbee: [tz.aqara_detection_interval, tz.RTCGQ13LM_motion_sensitivity],
    exposes: [e.occupancy(), exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']),
        exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
            .withDescription('Time interval for detecting actions'), e.battery()],
    meta: {battery: {voltageToPercentage: '3V_2850_3000_log'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genPowerCfg', ['batteryVoltage']);
        await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
        await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

lux73, 2022-03-14:
Thx for your quick response!

i have added your external converter and could see under Specific (settings) the new Value called "motion_wait_timeout"

but unfortunately it does nothing - i have added 2000ms but all what it does: it adds the 2000ms to my specified "detection_interval" (which was 5 Seconds) and then the Sensor shows "false" even there is motion...

Otnow, 2022-03-14:
In your example, in order for the sensor not go to false, it is necessary that it detects movement precisely during an additional period of 2000 ms.

Try increasing the sensitivity of the sensor or value of `motion_wait_timeout`.

lux73, 2022-03-14:
sensitivity is "high" as is before with 1.22.2

[image]

i have changed `motion_wait_timeout` to 10000ms - but it does not work...  after 15 Seconds is switches to Occupancy Clear in WebIf from z2m & thats what i could see in my OpenHAB Logging - i'm heavy jumping&moving in Front of the Sensor ;-)

it is expected that there a two concurrent settings exposed? `occupancy_timeout` & `detection_interval` ?

[image]

Otnow, 2022-03-14:
> it is expected that there a two concurrent settings exposed? `occupancy_timeout` & `detection_interval` ?

`occupancy_timeout` remained after switching to `detection_interval`.

> i have changed `motion_wait_timeout` to 10000ms - but it does not work... after 15 Seconds is switches to Occupancy Clear in WebIf from z2m & thats what i could see in my OpenHAB Logging - i'm heavy jumping&moving in Front of the Sensor ;-)

Very strange :-)

Do you currently have Zigbee2MQTT version 1.24.0 installed?

If so, try re-pairing the sensor.

ruicai83, 2022-03-15:
I have  a same problem , i think the program mis-understand  "occupancy_timeout" and "detection_interval".
now the detection_interval setting exactly means occupancy_timeout ,but not detect interval.

Zigbee2MQTT version: 1.23
Coordinator revision:20211217

ruicai83, 2022-03-15:
> In the mode when the sensor considers that it is communicating with the native hub, it stops transmitting the event of the end of motion detection on its own, and for it this event has to be generated by the converter by a timeout that is equal to the set detection period (`detection_interval`).
> 
> At the moment, that is why, when the timeout ends, the sensor reports the end of the movement even if in fact there is still movement.
> 
> To fix this problem, I made changes to the converter and suggest you test it.
> 
> In `Settings (specific)`, I added the `motion_wait_timeout` parameter (_value in **ms**_):
> 
> _If during the time specified in this parameter, after the end of the detection period, the sensor continues to detect movement, then the end of movement event is not generated and a new timeout is started._
> 
> You need to [add an external converter](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters) presented below to z2m and set a delay acceptable for you after the end of the detection interval.
> 
> In case of a positive effect, I will create a PR for inclusion in the z2m repository.
> 
> * **RTCGQ13LM_with_motion_wait_timeout.js**
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const fzLocal = {
>     RTCGQ13LM_occupancy: {
>         // This is for occupancy sensor that only send a message when motion detected,
>         // but do not send a motion stop.
>         // Therefore we need to publish the no_motion detected by ourselves.
>         cluster: 'msOccupancySensing',
>         type: ['attributeReport', 'readResponse'],
>         options: [exposes.numeric('motion_wait_timeout').withValueMin(0).withUnit('ms')
>             .withDescription('If a new motion event occurs within the time specified in this option (value is in ms) ' +
>             'after the end of "detection_interval", the "occupancy" state is not cleared')],
>         convert: (model, msg, publish, options, meta) => {
>             if (msg.data.occupancy !== 1) {
>                 // In case of 0 no occupancy is reported.
>                 // https://github.com/Koenkk/zigbee2mqtt/issues/467
>                 return;
>             }
> 
>             // The occupancy sensor only sends a message when motion detected.
>             // Therefore we need to publish the no_motion detected by ourselves.
>             let timeout = meta && meta.state && meta.state.hasOwnProperty('detection_interval') ? meta.state.detection_interval : 60;
>             const delay = options && options.hasOwnProperty('motion_wait_timeout') ? options.motion_wait_timeout : 0;
>             if (delay > 0) timeout += delay/1000;
> 
>             // Stop existing timers because motion is detected and set a new one.
>             globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
>             globalStore.putValue(msg.endpoint, 'timers', []);
> 
>             if (timeout !== 0) {
>                 const timer = setTimeout(() => {
>                     publish({occupancy: false});
>                 }, timeout * 1000);
> 
>                 globalStore.getValue(msg.endpoint, 'timers').push(timer);
>             }
> 
>             return {occupancy: true};
>         },
>     }
> }
> 
> const definition = {
>     zigbeeModel: ['lumi.motion.agl04'],
>     model: 'RTCGQ13LM',
>     vendor: 'Xiaomi',
>     description: 'Aqara high precision motion sensor',
>     fromZigbee: [fzLocal.RTCGQ13LM_occupancy, fz.aqara_opple, fz.battery],
>     toZigbee: [tz.aqara_detection_interval, tz.RTCGQ13LM_motion_sensitivity],
>     exposes: [e.occupancy(), exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']),
>         exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
>             .withDescription('Time interval for detecting actions'), e.battery()],
>     meta: {battery: {voltageToPercentage: '3V_2850_3000_log'}},
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await endpoint.read('genPowerCfg', ['batteryVoltage']);
>         await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
>         await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
>     },
>     ota: ota.zigbeeOTA,
> };
> 
> module.exports = definition;
> ```

thanks for solution, it works at z2m v1.24.
and the "motion_wait_timeout" setting find at settings(specific) tab,but not  exposes tab.

lux73, 2022-03-15:
@Otnow 

after **re-pairing** i could either confirmed it worked! :-)

BUT i had to re-pair the Sensor - re-configure via WebIf does not work...

i have `detection_interval` lowered to 2 Sec. & set `motion_set_timeout` to 3000ms - this works like a Charme for me

pls i would like to know why we had to Change this Setting from `occupancy_timeout` to `detection_interval` ?

z-x-f, 2022-03-15:
> @Otnow
>
>
>
> after **re-pairing** i could either confirmed it worked! :-)
>
>
>
> BUT i had to re-pair the Sensor - re-configure via WebIf does not work...
>
>
>
> i have `detection_interval` lowered to 2 Sec. [...]

https://github.com/Koenkk/zigbee-herdsman-converters/pull/3631#issuecomment-1007649464

These are two different properties.

z-x-f, 2022-03-15:
Ah, this should be similar to the original occupancy_timeout.

lux73, 2022-03-15:
> 
> [Koenkk/zigbee-herdsman-converters#3631 (comment)](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3631#issuecomment-1007649464)
> 
> These are two different properties.

@z-x-f 
thanks for the Link! understood now :-)

@Otnow
thx again for your quick help!

closed 2022-03-16 by lux73

Otnow, 2022-03-19:
During [code review of PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4006#pullrequestreview-910441042), it was decided to use the `occupancy_timeout` option instead of `motion_wait_timeout`, but this does not change the essence, because `occupancy_timeout` is nothing more than `detection_interval` + `motion_wait_timeout`.

As a result, [modified PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4006) is merged into the repository and is now available in the _[dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)_, and will appear in the _master branch_ at the beginning of next month.

lux73, 2022-03-23:
@Otnow 

Thx for the Info - does this mean we roll back to the behavior before??

Otnow, 2022-03-23:
> Thx for the Info - does this mean we roll back to the behavior before??

No, the problem is solved, but using `occupancy_timeout`.

> i have `detection_interval` lowered to 2 Sec. & set `motion_set_timeout` to 3000ms - this works like a Charme for me

For your example, the settings would look like this: `detection_interval` = 2s and `occupancy_timeout` = 5s.

PaulMcMillan, 2022-03-24:
I also discovered weird behavior until I re-paired my own device after this change. Is there a standard format for noting this sort of changes in the docs?

Otnow, 2022-03-24:
> Is there a standard format for noting this sort of changes in the docs?

Documentation for **RTCGQ13LM** sensor is published [here](https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html).

Information about `occupancy_timeout` option and other changes from [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4006) will be automatically added to [documentation](https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html) at beginning of next month along with next zigbee2mqtt release.