Koenkk/zigbee2mqtt #11847: [New device support]: Philips Hue White Ambiance 1600

gregjwinter, 2022-03-15:
### Link

https://www.philips-hue.com/en-us/p/hue-white-1-pack-a21-e26/046677557805#overview

### Database entry

Could not find a directory called data under home assistant

### Comments

I loaded new external converter and device was paired.  I got stuck at "Once finished, restart Zigbee2MQTT and trigger some actions on the device. You will see messages like:..."
I cant see any controls in dashboard to trigger.  Also exposed to Home Assistant and did not see the entity added so I could trigger any actions on the device.

This configuration is probably over my head - I just purchased this lamp from Amazon and saw it listed on your compatible devices.  [(https://www.zigbee2mqtt.io/devices/9290023351.html)]
Mine is a model LTA012 - I see a unit just like it called a LTA011 in your database.  I am stumped.  Should I just return it to Amazon?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LTA012'],
    model: '9290024720',
    vendor: 'myVendor',
    description: 'Hue White Ambiance 1600',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

### Supported color modes

Says on package 2200-6500K

Koenkk, 2022-03-15:
Without any changes/external converter, can you grab the color temp range so I can add it?

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

gregjwinter, 2022-03-15:
TempPhysicalMin is 153
colorTempPhysicalMax is 454

Thank you! -Greg.

Koenkk, 2022-03-15:
added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-15 by Koenkk

gregjwinter, 2022-03-15:
It works!  Thank you very much. -Greg.